FreeCAD/FreeCAD #11442: [Problem] 50642457 Окна перекрывают рабочее поле.

Ltybc1980, 2023-11-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

  FreeCAD 0.21.1 использовал  минимальный размер шрифтов и все равно - не в одном окне не возможно работать. Окно боковой панели занимает 1/4 экрана, панель приборов 1/2 экрана . Содержимое что бы редактировать (левой панели) надо растягивать окно на весь экран И ЭТО при минимально возможном шрифте. При максимальных настройках шрифтов - "рабочего поля" вообще нет 1/8 экрана.

### Full version info

```shell
[code]
OS: Linux Mint 20.2 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * Manipulator 1.5.0
[/code]
```

### Subproject(s) affected?

None

### Anything else?

нет

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2023-11-18:
Приложил бы скриншот. Откуда разработчикам знать настройки твоих шрифтов?

Ltybc1980, 2023-11-18:
Это настройки программы, ( я только пытаюсь в ней учиться) но это реально баг, К
[image]
как здесь приложить срин?

Ltybc1980, 2023-11-18:
[image]
[image]

Ltybc1980, 2023-11-18:
[image]

Ltybc1980, 2023-11-18:
[image]
при минимальных настройках шрифтов

Ltybc1980, 2023-11-18:
Шрифты  в нижней левой панели вообще не поддаются настройки  и панель приходится растягивать для редактирования до максимума. Это при настройке шрифтов "малый 16 рх"
[image]
как то так

Ltybc1980, 2023-11-18:
[image]
1 Как миниму надо сократить в 5 раз шрифт
2 Как минимум  сократить шрифт в 3 раза
3 Рабочее поле очень мало Сократиф шрифты  в 1 и 2 поле будет удовлетворительным для работы
В данном исполнение не возможно сделать

hasecilu, 2023-11-18:
If you press `Alt + F11` you can gain a few pixels from the title bar and the Linux Mint Panel.

Ltybc1980, 2023-11-18:
ALt + F11 не решает проблемы: маленькой рабочей области. Ко всему прочему  на скрин показал не возможность изменять  -размер  шрифта ( выделил 1) Данная консоль, для внесения в неё параметров, требут постоянно растягивать и закрывать рабочую область, что очень не удобно. Настройками параметров невозможно решить проблемы.

hasecilu, 2023-11-18:
I know it doesn't solve the problem but I think I could help a little bit, in this cases every pixel count.

luzpaz, 2023-11-19:
Please use english in the bugtracker.