FreeCAD/FreeCAD #11444: "Anti-aliasing - Line Smoothing" causes XYZ Plane selection graphic artifacts in Sketcher

JohnOCFII, 2023-11-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When you first enter the screen to select a plane, everything looks as expected, with the three axes shown in a semi-transparent way. Once I clcck into the graphics area to select a plane, all three planes turn a single solid shade.

I had previously posted this in the Forum in July - but hadn't narrowed it down to `anti-aliasing` and `Line Smoothing` at that time.
[image]

[image]

### Full version info

```shell
OS: macOS 14.1.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35024 (Git)
Build type: Release
Branch: main
Hash: 4ff00ef99869e35295a5a17b8f486901504c655b
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
```
```

### Subproject(s) affected?

Sketcher

### Anything else?

Previous discussion in the forum: https://forum.freecad.org/viewtopic.php?p=719742#p719742

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

GAZ082, 2023-11-18:
Same issue in Linux!

[code]
OS: Arch Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: makepkg
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.11.6, Qt 5.15.11, Coin 4.0.1, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Plot 2022.4.17
  * Curves 0.6.14
  * ExtremeProDark 2.9.0
  * DynamicData 2.53.0
  * Assembly4 0.50.4
  * Curves.backup1693059780.2922277 0.6.13 (Disabled)
  * Assembly4.backup1696800837.713216 0.50.2 (Disabled)
  * ThreadProfile 1.89.0
  * ProDarkThemePreferencePack 1.0.0
  * CfdOF.backup1699129791.898633 1.24.7 (Disabled)
  * Manipulator 1.5.0
  * CfdOF 1.24.8
[/code]

GAZ082, 2024-04-28:
Still applies to DEV:

OS: Arch Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36996 (Git) AppImage
Build type: Release
Branch: main
Hash: d92f74ef019fcedbdcdc32055ab2437c950ac23b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * 3DfindIT 1.2.0
  * Curves.backup1713713759.0032098 0.6.29 (Disabled)
  * Plot 2022.4.17
  * Defeaturing 1.2.2
  * QuickMeasure 2022.10.28
  * Curves 0.6.35
  * ExtremeProDark 2.9.0
  * Curves.backup1693059780.2922277 0.6.13 (Disabled)
  * CfdOF.backup1710694906.6805208 1.24.9 (Disabled)
  * Assembly4.backup1696800837.713216 0.50.2 (Disabled)
  * Curves.backup1714093860.7275996 0.6.34 (Disabled)
  * ThreadProfile 1.89.0
  * OpenTheme 2024.4.20
  * ProDarkThemePreferencePack 1.0.0
  * CfdOF.backup1699129791.898633 1.24.7 (Disabled)
  * Assembly4.backup1710902627.637761 0.50.9 (Disabled)
  * Defeaturing.backup1710696629.0288966 1.2.2 (Disabled)
  * Manipulator 1.5.7

## Hardware Information:
- **Hardware Model:**                              Gigabyte Technology Co., Ltd. P55-US3L
- **Memory:**                                      16,0 GiB
- **Processor:**                                   Intel® Core™ i5 760  × 4
- **Graphics:**                                    AMD Radeon™ RX 550 / 550 Series
- **Disk Capacity:**                               240,1 GB

JohnOCFII, 2024-11-16:
Still an issue (but slightly not as bad, perhaps due to OpenTheme, as you can read the axis name) in current dev.  Easy workaround is to select any anti-alias option `other than Line Smoothing.`

```
OS: macOS 15.1
Architecture: arm64
Version: 1.1.0dev.39176 (Git) Conda
Build type: Release
Branch: main
Hash: 45d3864942c6928cec7dc18f95b1f222485756e5
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * Curves 0.6.51
```
[image]

JohnOCFII, 2025-03-27:
With the change to plane selection, I was curious to see if this problem still existed.  It does, but it does present differently, and not as severe an issue. 

Display with `Line Smoothing` and `One Pass` options:

[image]

Display with `MSAA 4x` and `Backface Pass` options:

[image]

Perhaps, given the reduced display depth, that is all we can expect, and if so, I'll close the issue.

luzpaz, 2025-04-08:
> Perhaps, given the reduced display depth, that is all we can expect, and if so, I'll close the issue.

Is there more to be improved here ?

JohnOCFII, 2025-04-10:
> Is there more to be improved here ?

I don't know.  It would be good if someone familiar with the expected behavior could review and respond if this is expected behavior or not.

FEA-eng, 2025-07-17:
There's also this glitch in the Sketcher:

[image]