FreeCAD/FreeCAD #11445: [Problem] Path  3D  surface operation does not properly use G1, other issues.

blazini36, 2023-11-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I know this is an "experimental feature" but worth noting the issues anyway. I detailed most of them in this post on the FreeCAD forum.
[https://forum.freecad.org/viewtopic.php?p=719714#p719714](url)

These issues were noticed running multipass, drop cutter=Y

1) This is probably an issue with the postprocessor but the A-Axis output is from -180 to 180 degrees. "Wrapped rotary" is the Gcode standard, this should at least output 0-360. Was able to fix this with the LinuxCNC PP

2) Rotary axis moves are always expressed in degrees. Using an "inch" postptocessor argument wrongly converts the A-axis values along with the others. I was able to get the LinuxCNC PP to re-convert the A values.

3) The feedrates output don't make great sense, they don't follow anything obvious. I assume it is trying to calculate linear feedrate to apply to the rotary axis but the same feedrate is applied to linear axes. The standard for G1 is to always use linear feedrate (the tools feedrate settings) unless the rotary is used alone, in which case the output should represent degrees/min. What you wind up with is linear feedrates that are way too fast and rotary feedrates that are way too slow. This problem is exagerated using multipass. I output a part that started with a 2" diameter and had sections milled down to about 15mm. The first pass started with a FR of 12, the last pass had a FR of 212. A FR of 12deg/min is way too slow for anything but a very large diameter. A FR of 212in/min is way, way too fast for any linear move

4) The first pass in multipass is started (apparently) at safe height. It should be entering Z at safe height and running the first pass at start height. This is an assumption because the first pass I ran was well above the part.

5) there is alot of odd, not useful movement. It starts out with the Z going up and down a few times. towards the end of the program it makes a ton of movements just along Z and X without a feedrate. With the feedrate missing it assumes 212in/min. I cut all of these movements out when I tested it so I didn't actually see it bouncing up and down or zooming back and forth but that's obviously what it would have done.

If possible this operation should print details in the Gcode output referencing how it came up with the feedrates it used so it can be checked.

[Microscope camera arm adapter OP2-orig.txt](https://github.com/FreeCAD/FreeCAD/files/13401660/Microscope.camera.arm.adapter.OP2-orig.txt)

### Full version info

```shell
OS: Manjaro Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35039 (Git)
Build type: None
Branch: main
Hash: e937cc1efc3e210655575a408425098fafef39e7
Python 3.11.5, Qt 5.15.11, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.60k
  * freecad.gears 1.0.0
  * kicadStepUpMod 10.14.5
  * dxf-library
  * fasteners 0.4.54
  * Assembly4 0.12.6
```
```

### Subproject(s) affected?

Path

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-11-19:
Are any of these issues duplicates of [open Path tickets](https://github.com/FreeCAD/FreeCAD/issues?q=is%3Aopen+is%3Aissue+label%3A%22WB+Path%22) ?

blazini36, 2023-11-19:
Not that I can find as of the past year

luzpaz, 2023-11-19:
@mlampert @LarryWoestman would y'all mind weighing in ?

LarryWoestman, 2023-11-23:
I think we have multiple issues in this report.

The first issue is that the linuxcnc_post.py file (for the linuxcnc postprocessor implementation) treats the A parameter the same as it does the X parameter, trying to change it when in "--inches" mode.  The A, B, and C parameters are expected to be in degrees, not distance, and should not be changing when switching from metric to inches mode.

The second issue is that the linuxcnc_post.py file has been recently modified to replace "for c in pathobj.Path.Commands:" with "for c in PathUtils.getPathWithPlacement(pathobj).Commands:", which breaks my expectations of how the A, B, and C parameters are being handled.  It appears to me that the placement code is using the A, B, and C parameters in a manner that conflicts with the manner that the PATH code expects, particularly the code that attempts to use the A, B, or C parameters to drive rotary one or more axis on CNC machines.

There are likely more issues as well, including feed rates that I have not tried to investigate.

At the moment the master build is broken.  When it is working again I will create a PR with a modified linuxcnc_post.py file that has a fix for the A, B, and C parameter handling and reverts the pathobj line change describes above.  The PR will also contain a revised version of the TestLinuxCNCPost.py file that contains some tests that focus on the A, B, and C parameters.

@sliptonic @Russ4262 Would you be willing to chime in on how the A, B, and C parameters are expected to be used in the PATH code and what is the valid range of values for those parameters?  I can't find any information on this, and it is needed to understand whether there is a conflict between the placement code's and the PATH code's use of A, B, and C.

LarryWoestman, 2023-11-28:
I have added a draft Pull Request #11575 that contains a modified linuxcnc_post.py file along with tests for the A, B, and C parameter handling in TestLinuxCNCPost.py.

LarryWoestman, 2024-04-29:
I ended up closing draft Pull Request #11575 by my error, so I have created a new draft Pull Request #13702 with up-to-date code.

I have also posted the question in the forum at [https://github.com/FreeCAD/FreeCAD/pull/url](url)

LarryWoestman, 2024-05-06:
Pull Request #13702 has been merged into main, to my surprise because it was still labeled as Draft.  I am happy that it is merged and it should work well, but I am still trying to get answers about the possible conflict between how Path/CAM uses the A, B, and C parameters and how the rest of FreeCAD appears to be using them.

sliptonic, 2024-05-06:
We can revert if you want but the PR was not marked draft.  It was merged during the monday merge meeting and we work from [a canned view that specifically excludes draft PRs](https://github.com/FreeCAD/FreeCAD/pulls?q=is%3Apr+is%3Aopen+draft%3Afalse+sort%3Aupdated-asc).

LarryWoestman, 2024-05-06:
As far as I can tell, I don't have the permissions to mark a PR as 
"draft".  I had "Draft" in the title, though.

No harm done, unless the open questions end up meaning that I 
misunderstood something important.  If so, I suspect someone will
eventually notice :-)

LarryWoestman, 2025-01-30:
I have fixed #1 and #2,  I believe.  I haven't seen any feedback otherwise, at least :-)

I have a pending PR that should fix at least some of the feed rate issues described in #3.  It would be useful if the original poster could try things after that PR has been merged.

#4 and #5 are in areas that I don't feel I know enough to address at this time.

sliptonic, 2025-07-13:
> [@sliptonic](https://github.com/sliptonic) Per [#11445 (comment)](https://github.com/FreeCAD/FreeCAD/issues/11445#issuecomment-2623287439) there are 2 remaining issues. Should I split them into their own tickets ?

That would be great.  Thanks

FEA-eng, 2025-09-14:
@sliptonic Still relevant?

luzpaz, 2025-11-15:
Yea, i need to split 4 and 5 out of this ticket. I'll do that now.