FreeCAD/FreeCAD #11446: Spacecoast does not work on 0.21.1

strangeluck, 2023-11-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I recently upgrade both my Spacemouse driver and Freecad and discovered my Spacemouse didn't work in Freecad any longer. I experimented with Freecad/Spacemouse and the results are below

Spacemouse    FreeCAD 0.20.2     FreeCAD 0.21.1
10.7.0                             YES                        NO
10.7.4                              NO                        NO
10.8.2                             NO                        NO

Spacemouse is amazing and I hope support never dies. Hope this gets resolved. Sticking with what works... 
Love your software!

### Full version info

```shell
Does work:

[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
[/code]

Doesn't work:

[code]
OS: macOS 13.3.1
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2023-11-20:
See https://github.com/FreeCAD/FreeCAD/issues/10926#issuecomment-1762203236 it seems relevant, it would seem that 0.21.2 should be packaged with the fix

numo68, 2023-12-18:
#11753 restores the functionality with 0.8.2

luzpaz, 2023-12-20:
@numo68 in other words, this ticket can be closed ?

numo68, 2023-12-20:
> in other words, this ticket can be closed ?

@luzpaz The PR was merged to main - if that is enough to close the ticket, then yes. The fix is not present in 0.21 (at least not yet; the backport label is there).

luzpaz, 2023-12-20:
Closing in favor of #11753 which is slated to backported.

closed 2023-12-20 by luzpaz