FreeCAD/FreeCAD #11448: [Problem] cannot stop calculix

aavogt, 2023-11-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In SolverCcxTools task view, the label on the button that starts the analysis changes to "Stop CalculiX". The timer rests but the process keeps running. I attached a video, and the relevant parts of the console are:
```
01:11:54  run CalculiX at: /tmp/.mount_freecaMDDZk5/usr/bin/ccx with: /tmp/fcfem_fys8rs9z/FEMMeshNetgen.inp
01:11:54  QProcess::start: Process is already running
01:14:32  Error() 
```

### Full version info

```shell
0.21.1
33694
2023/08/31
Ubuntu 22.04.3 LTS
64-bit
```

### Subproject(s) affected?

None

### Anything else?

[freecad.webm](https://github.com/FreeCAD/FreeCAD/assets/93006/4fdcdbd1-c704-493e-ae2c-4df2213c0464)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-11-19:
@aavogt please edit the OP and add your full [About](https://wiki.freecad.org) info

CC @FEA-eng (do you mind confirming?)

FEA-eng, 2023-11-19:
> CC @FEA-eng (do you mind confirming?)

I can confirm this, the Stop CalculiX button is not working properly.

marioalexis84, 2023-11-24:
I will have a look

commit c7b70713 (2024-02-12): Fem: Enable "Stop CalculiX" button - fixes #11448
closed 2024-02-12 by chennes