FreeCAD/FreeCAD #11449: [Problem] Fails to build against VTK 9.3.0

MehdiChinoune, 2023-11-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCAD fails to build against latest VTK 9.3.0
```
  [19/4686] Building CXX object src/3rdParty/salomesmesh/CMakeFiles/SMDS.dir/src/SMDS/SMDS_FaceOfNodes.cpp.obj
  FAILED: src/3rdParty/salomesmesh/CMakeFiles/SMDS.dir/src/SMDS/SMDS_FaceOfNodes.cpp.obj 
  D:\M\msys64\clang64\bin\clang++.exe -DBOOST_PP_VARIADICS=1 -DCSFDB -DDIY_NO_THREADS -DH5_BUILT_AS_DYNAMIC_LIB -DHAVE_CONFIG_H -DHAVE_D3D -DHAVE_FFMPEG -DHAVE_FREEIMAGE -DHAVE_FREETYPE -DHAVE_OPENGL_EXT -DHAVE_OPENVR -DHAVE_RAPIDJSON -DHAVE_SNPRINTF -DHAVE_TBB -DHAVE_TK -DHAVE_VTK -DKISSFFT_DLL_IMPORT=1 -DOCC_CONVERT_SIGNALS -DPYCXX_6_2_COMPATIBILITY -DQT_NO_KEYWORDS -DSMDS_EXPORTS -DUNICODE -DVTK_OPENGL2_BACKEND -DWIN32 -DWNT -D_OCC64 -D_UNICODE -D_WINDOWS -Dkiss_fft_scalar=double -IC:/_/B/src/build-CLANG64/src/3rdParty/salomesmesh/SMDS_autogen/include -IC:/_/B/src/build-CLANG64 -IC:/_/B/src/build-CLANG64/src -IC:/_/B/src/FreeCAD-0.21.2/src -IC:/_/B/src/FreeCAD-0.21.2/src/3rdParty/salomesmesh/src/SMDS -IC:/_/B/src/FreeCAD-0.21.2/src/3rdParty/salomesmesh/src/Driver -IC:/_/B/src/FreeCAD-0.21.2/src/3rdParty/salomesmesh/src/DriverUNV -IC:/_/B/src/FreeCAD-0.21.2/src/3rdParty/salomesmesh/src/DriverDAT -IC:/_/B/src/FreeCAD-0.21.2/src/3rdParty/salomesmesh/src/DriverSTL -IC:/_/B/src/FreeCAD-0.21.2/src/3rdParty/salomesmesh/src/StdMeshers -IC:/_/B/src/FreeCAD-0.21.2/src/3rdParty/salomesmesh/inc -IC:/_/B/src/build-CLANG64/src/3rdParty/salomesmesh -isystem D:/M/msys64/clang64/include/opencascade -isystem D:/M/msys64/clang64/include/vtk -isystem D:/M/msys64/clang64/include/freetype2 -Wall -Wextra -Wpedantic -Wno-write-strings -march=nocona -msahf -mtune=generic -O2 -pipe -Wno-attributes -Wno-self-assign -Wno-sign-compare -Wno-logical-op-parentheses -Wno-reorder -Wno-switch -Wno-switch-enum -Wno-unknown-pragmas -Wno-unused-variable -Wno-unused-private-field -Wno-unused-function -Wno-sometimes-uninitialized -Wno-overloaded-virtual -Wno-dynamic-class-memaccess -Wno-comment -Wno-unused-parameter -Wno-extra-semi -Wno-deprecated-copy -Wno-missing-field-initializers -Wno-c++17-extensions -O3 -DNDEBUG -std=gnu++17 -DSMDS_EXPORTS -DUTILS_EXPORTS -pthread -fopenmp=libomp -MD -MT src/3rdParty/salomesmesh/CMakeFiles/SMDS.dir/src/SMDS/SMDS_FaceOfNodes.cpp.obj -MF src\3rdParty\salomesmesh\CMakeFiles\SMDS.dir\src\SMDS\SMDS_FaceOfNodes.cpp.obj.d -o src/3rdParty/salomesmesh/CMakeFiles/SMDS.dir/src/SMDS/SMDS_FaceOfNodes.cpp.obj -c C:/_/B/src/FreeCAD-0.21.2/src/3rdParty/salomesmesh/src/SMDS/SMDS_FaceOfNodes.cpp
  In file included from C:/_/B/src/FreeCAD-0.21.2/src/3rdParty/salomesmesh/src/SMDS/SMDS_FaceOfNodes.cpp:33:
  In file included from C:/_/B/src/FreeCAD-0.21.2/src/3rdParty/salomesmesh/inc/SMDS_Mesh.hxx:45:
  In file included from C:/_/B/src/FreeCAD-0.21.2/src/3rdParty/salomesmesh/inc/SMDS_VtkVolume.hxx:26:
  C:/_/B/src/FreeCAD-0.21.2/src/3rdParty/salomesmesh/inc/SMDS_UnstructuredGrid.hxx:71:26: error: virtual function 'GetMTime' has a different return type ('unsigned long') than the function it overrides (which has return type 'vtkMTimeType' (aka 'unsigned long long'))
     71 |   virtual VTK_MTIME_TYPE GetMTime();
        |           ~~~~~~~~~~~~~~ ^
  D:/M/msys64/clang64/include/vtk/vtkPointSet.h:208:16: note: overridden virtual function is here
    208 |   vtkMTimeType GetMTime() override;
        |   ~~~~~~~~~~~~ ^
  In file included from C:/_/B/src/FreeCAD-0.21.2/src/3rdParty/salomesmesh/src/SMDS/SMDS_FaceOfNodes.cpp:33:
  In file included from C:/_/B/src/FreeCAD-0.21.2/src/3rdParty/salomesmesh/inc/SMDS_Mesh.hxx:45:
  In file included from C:/_/B/src/FreeCAD-0.21.2/src/3rdParty/salomesmesh/inc/SMDS_VtkVolume.hxx:26:
  C:/_/B/src/FreeCAD-0.21.2/src/3rdParty/salomesmesh/inc/SMDS_UnstructuredGrid.hxx:99:39: warning: 'GetCellLinks' is deprecated: Use GetLinks() instead. [-Wdeprecated-declarations]
     99 |     return static_cast<vtkCellLinks*>(GetCellLinks());
        |                                       ^
  D:/M/msys64/clang64/include/vtk/vtkUnstructuredGrid.h:289:3: note: 'GetCellLinks' has been explicitly marked deprecated here
    289 |   VTK_DEPRECATED_IN_9_3_0("Use GetLinks() instead.")
        |   ^
  D:/M/msys64/clang64/include/vtk/vtkDeprecation.h:99:41: note: expanded from macro 'VTK_DEPRECATED_IN_9_3_0'
     99 | #define VTK_DEPRECATED_IN_9_3_0(reason) VTK_DEPRECATION(reason)
  
        |                                         ^
  
  D:/M/msys64/clang64/include/vtk/vtkDeprecation.h:77:44: note: expanded from macro 'VTK_DEPRECATION'
  
     77 | #define VTK_DEPRECATION(reason) __declspec(deprecated(reason))
  
        |                                            ^
  
  1 warning and 1 error generated.
```

### Full version info

```shell
0.21.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FlachyJoe, 2023-11-19:
Could we consider adding SalomeSMESH as a submodule in order to upgrade it easily?
https://git.salome-platform.org/gitweb/?p=modules/smesh.git;a=summary

MehdiChinoune, 2023-11-19:
It doesn't look the failure is due new version of VTK, because https://gitlab.kitware.com/vtk/vtk/-/blob/v9.3.0/Common/DataModel/vtkPointSet.h?ref_type=tags#L208 wasn't changed since 6 years.
I think the latest clang-17 starts rejecting the code.

Pesc0, 2023-11-21:
> Could we consider adding SalomeSMESH as a submodule in order to upgrade it easily?

CC @adrianinsaval

kadet1090, 2023-11-21:
> I think the latest clang-17 starts rejecting the code.

Don't think so, downgrading VTK to 9.2.6 fixes the issue.

adrianinsaval, 2023-11-21:
it's because vtk 9.3, the arch package applies a patch for it https://gitlab.archlinux.org/archlinux/packaging/packages/freecad/-/blob/c2e2e082a68f957382e2cf3ecc403f07becf2c8e/freecad-vtk9.3.patch

adrianinsaval, 2023-11-21:
> > Could we consider adding SalomeSMESH as a submodule in order to upgrade it easily?
> 
> CC @adrianinsaval

@wwmayer do you think we should try this?

wwmayer, 2023-11-22:
This is the relevant part of the vtk change:
```
  virtual void BuildLinks(vtkDataSet* data) = 0;
  VTK_DEPRECATED_IN_9_3_0("Use SetDataSet() and BuildLinks() instead.")
  void BuildLinks(vtkDataSet* data);

  /**
   * Build the link list array from the input dataset.
   */
  virtual void BuildLinks() = 0;
```
kadet1090 is right that it is not clear why the build fails because the base class still defines the overloaded version that accepts a single argument.

wwmayer, 2023-11-22:
> @wwmayer do you think we should try this?

What exactly would this solve? The past has shown that upstream (i.e. [Salome](https://git.salome-platform.org)) doesn't worry much about compatibility over several vtk versions. As an example just look at this example [SMDS_UnstructuredGrid::BuildLinks()](https://git.salome-platform.org/gitweb/?p=modules/smesh.git;a=blob;f=src/SMDS/SMDS_UnstructuredGrid.cxx;h=cf820b9c876531eb816a93127d8c8e9d498ab52f;hb=HEAD#l1019)

wwmayer, 2023-11-22:
> it's because vtk 9.3, the arch package applies a patch for it

Btw, the patch is not complete. According to the above mentioned change in vtk the calls must be:
```
SetDataSet(this);
BuildLinks();
```

adrianinsaval, 2023-11-22:
@xyproto this might be of interest to you:

> > it's because vtk 9.3, the arch package applies a patch for it
> 
> Btw, the patch is not complete. According to the above mentioned change in vtk the calls must be:
> 
> ```
> SetDataSet(this);
> BuildLinks();
> ```

I also reported on archlinux bugtracker https://bugs.archlinux.org/task/80321

adrianinsaval, 2023-11-22:
would that method work on older vtk versions? what's the minimum version we aim to support?

wwmayer, 2023-11-22:
> would that method work on older vtk versions?

No, I don't think so.

> what's the minimum version we aim to support?

If anyhow possible then we should still support vtk7 because that's the version of Ubuntu 20.04.

greyltc, 2023-11-22:
Meshing in the FEM workbench with Arch's freecad package is broken for me as of very recently. SIGSEV crashes out the whole program with the trace below. I've tested packages 0.21.1-7 and 0.21.1-7. Same on both. If anyone could propose a change to Arch's vtk9.3 patch, I'd be happy to test it (sorry, I'm not sure I quite understand how the patch needs to change from what's written here).

```
$ freecad
FreeCAD 0.21.1, Libs: 0.21.1R33694 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

** (freecad:410897): WARNING **: 22:17:37.315: Error loading Implements key from /usr/share/applications/dbtool.desktop: Key file does not have group “Desktop Entry”
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3e710) [0x7f921945c710]
#1  0x7f916459ddce in vtkCellLinks::BuildLinks() from /usr/lib/libvtkCommonDataModel.so.1+0x2e
#2  0x7f91c7bcace8 in SMDS_UnstructuredGrid::BuildLinks() from /usr/lib/freecad/lib/libSMDS.so+0xf8
#3  0x7f91c7badb87 in SMDS_Mesh::SMDS_Mesh() from /usr/lib/freecad/lib/libSMDS.so+0x4a7
#4  0x7f91fb38fd80 in SMESHDS_Mesh::SMESHDS_Mesh(int, bool) from /usr/lib/freecad/lib/libSMESHDS.so+0x20
#5  0x7f91fb387e4a in SMESHDS_Document::NewMesh(bool, int) from /usr/lib/freecad/lib/libSMESHDS.so+0x9a
#6  0x7f91655b88f2 in SMESH_Mesh::SMESH_Mesh(int, int, SMESH_Gen*, bool, SMESHDS_Document*) from /usr/lib/freecad/lib/libSMESH.so+0x152
#7  0x7f91655a6f4f in SMESH_Gen::CreateMesh(int, bool) from /usr/lib/freecad/lib/libSMESH.so+0x4f
#8  0x7f91664e573d in Fem::FemMesh::FemMesh() from /usr/lib/freecad/lib/Fem.so+0x8d
#9  0x7f9166518457 in Fem::PropertyFemMesh::PropertyFemMesh() from /usr/lib/freecad/lib/Fem.so+0x37
#10  0x7f91664d5c5b in Fem::FemMeshObject::FemMeshObject() from /usr/lib/freecad/lib/Fem.so+0x3b
#11  0x7f91664d5e39 in App::FeaturePythonT<Fem::FemMeshObject>::create() from /usr/lib/freecad/lib/Fem.so+0x39
#12  0x7f921bc69d09 in App::Document::addObject(char const*, char const*, bool, char const*, bool) from /usr/lib/freecad/lib/libFreeCADApp.so+0xd9
#13  0x7f921bd53a41 in App::DocumentPy::addObject(_object*, _object*) from /usr/lib/freecad/lib/libFreeCADApp.so+0x101
#14  0x7f921bd54b48 in App::DocumentPy::staticCallback_addObject(_object*, _object*, _object*) from /usr/lib/freecad/lib/libFreeCADApp.so+0x28
#15  /usr/lib/libpython3.11.so.1.0(+0x1fb2e1) [0x7f921b1fb2e1]
#16  /usr/lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x2bc) [0x7f921b1da8ec]
#17  /usr/lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6d3) [0x7f921b1e4c23]
#18  /usr/lib/libpython3.11.so.1.0(+0x29c484) [0x7f921b29c484]
#19  /usr/lib/libpython3.11.so.1.0(PyEval_EvalCode+0xac) [0x7f921b29be6c]
#20  /usr/lib/libpython3.11.so.1.0(+0x2b9fc3) [0x7f921b2b9fc3]
#21  /usr/lib/libpython3.11.so.1.0(+0x2b63ea) [0x7f921b2b63ea]
#22  /usr/lib/libpython3.11.so.1.0(PyRun_StringFlags+0x6d) [0x7f921b2aaa3d]
#23  0x7f921c73e94d in Gui::Application::sDoCommand(_object*, _object*) from /usr/lib/freecad/lib/libFreeCADGui.so+0xbd
#24  /usr/lib/libpython3.11.so.1.0(+0x1fb308) [0x7f921b1fb308]
#25  /usr/lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x2bc) [0x7f921b1da8ec]
#26  /usr/lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6d3) [0x7f921b1e4c23]
#27  /usr/lib/libpython3.11.so.1.0(+0x22e223) [0x7f921b22e223]
#28  /usr/lib/libpython3.11.so.1.0(+0x22d540) [0x7f921b22d540]
#29  /usr/lib/freecad/lib/libFreeCADBase.so(PP_Run_Method+0xe7) [0x7f921b81e707]
#30  0x7f921b7527ff in Base::InterpreterSingleton::runMethodVoid(_object*, char const*) from /usr/lib/freecad/lib/libFreeCADBase.so+0x2f
#31  0x7f921c857477 in Gui::PythonCommand::activated(int) from /usr/lib/freecad/lib/libFreeCADGui.so+0x57
#32  0x7f921c857c97 in Gui::Command::_invoke(int, bool) from /usr/lib/freecad/lib/libFreeCADGui.so+0x247
#33  0x7f921c8580c7 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /usr/lib/freecad/lib/libFreeCADGui.so+0x137
#34  /usr/lib/libQt5Core.so.5(+0x2d1097) [0x7f9219cd1097]
#35  0x7f921a96bbb7 in QAction::triggered(bool) from /usr/lib/libQt5Widgets.so.5+0x47
#36  0x7f921a97160b in QAction::activate(QAction::ActionEvent) from /usr/lib/libQt5Widgets.so.5+0xbb
#37  /usr/lib/libQt5Widgets.so.5(+0x2f42db) [0x7f921aaf42db]
#38  /usr/lib/libQt5Widgets.so.5(+0x2f43fd) [0x7f921aaf43fd]
#39  0x7f921a9af1a1 in QWidget::event(QEvent*) from /usr/lib/libQt5Widgets.so.5+0x8d1
#40  0x7f921a9788ff in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt5Widgets.so.5+0x8f
#41  0x7f921a97ddaf in QApplication::notify(QObject*, QEvent*) from /usr/lib/libQt5Widgets.so.5+0x10ff
#42  0x7f921c7fc429 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x99
#43  0x7f9219c9c168 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt5Core.so.5+0x128
#44  0x7f921a97c0ea in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /usr/lib/libQt5Widgets.so.5+0x1ca
[... cut ...]
#57  0x7f9219c9ae74 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt5Core.so.5+0x134
#58  0x7f9219c9c313 in QCoreApplication::exec() from /usr/lib/libQt5Core.so.5+0x93
#59  0x7f921c71d0ad in Gui::Application::runApplication() from /usr/lib/freecad/lib/libFreeCADGui.so+0x1ecd
#60  freecad(+0x7a1d) [0x562aa1667a1d]
#61  /usr/lib/libc.so.6(+0x27cd0) [0x7f9219445cd0]
#62  /usr/lib/libc.so.6(__libc_start_main+0x8a) [0x7f9219445d8a]
#63  freecad(+0x7e15) [0x562aa1667e15]
```

adrianinsaval, 2023-11-22:
try 
```
diff --git a/src/3rdParty/salomesmesh/src/SMDS/SMDS_UnstructuredGrid.cpp b/src/3rdParty/salomesmesh/src/SMDS/SMDS_UnstructuredGrid.cpp
index 7e3de0070a..dc5249158d 100644
--- a/src/3rdParty/salomesmesh/src/SMDS/SMDS_UnstructuredGrid.cpp
+++ b/src/3rdParty/salomesmesh/src/SMDS/SMDS_UnstructuredGrid.cpp
@@ -1026,7 +1026,8 @@ void SMDS_UnstructuredGrid::BuildLinks()
   GetLinks()->Allocate(this->GetNumberOfPoints());
   GetLinks()->Register(this);
 //FIXME: vtk9
-  GetLinks()->BuildLinks(this);
+  GetLinks()->SetDataSet(this);
+  GetLinks()->BuildLinks();
   GetLinks()->Delete();
 #else
   this->Links = SMDS_CellLinks::New();
```

greyltc, 2023-11-22:
Fixed. Thank you @adrianinsaval!

adrianinsaval, 2023-11-22:
> > would that method work on older vtk versions?
> 
> No, I don't think so.

is there something like VTK_VERSION we can use to put it inside an if else block?

MehdiChinoune, 2023-11-22:
The ArchLinux patch doesn't fix building with clang-17

adrianinsaval, 2023-11-22:
what's the error?

MehdiChinoune, 2023-11-22:
> what's the error?

above.

adrianinsaval, 2023-11-22:
try this patch then:
```
diff --git a/src/3rdParty/salomesmesh/inc/SMESH_SMDS.hxx b/src/3rdParty/salomesmesh/inc/SMESH_SMDS.hxx
index 7f89573efd..47a52fe585 100644
--- a/src/3rdParty/salomesmesh/inc/SMESH_SMDS.hxx
+++ b/src/3rdParty/salomesmesh/inc/SMESH_SMDS.hxx
@@ -39,10 +39,6 @@
  #define SMDS_EXPORT
 #endif
 
-#ifdef VTK_HAS_MTIME_TYPE
 #define VTK_MTIME_TYPE vtkMTimeType
-#else
-#define VTK_MTIME_TYPE unsigned long
-#endif
 
 #endif
diff --git a/src/3rdParty/salomesmesh/src/SMDS/SMDS_UnstructuredGrid.cpp b/src/3rdParty/salomesmesh/src/SMDS/SMDS_UnstructuredGrid.cpp
index 7e3de0070a..e9895d6157 100644
--- a/src/3rdParty/salomesmesh/src/SMDS/SMDS_UnstructuredGrid.cpp
+++ b/src/3rdParty/salomesmesh/src/SMDS/SMDS_UnstructuredGrid.cpp
@@ -1026,7 +1026,12 @@ void SMDS_UnstructuredGrid::BuildLinks()
   GetLinks()->Allocate(this->GetNumberOfPoints());
   GetLinks()->Register(this);
 //FIXME: vtk9
+  #if VTK_VERSION_NUMBER < VTK_VERSION_CHECK(9,3,0)
   GetLinks()->BuildLinks(this);
+  #else
+  GetLinks()->SetDataSet(this);
+  GetLinks()->BuildLinks();
+  #endif
   GetLinks()->Delete();
 #else
   this->Links = SMDS_CellLinks::New();
```

MehdiChinoune, 2023-11-23:
@adrianinsaval your patch works.

adrianinsaval, 2023-11-23:
#11496

closed 2023-11-24 by adrianinsaval

wsteffe, 2023-11-24:
Hi, MehdiChinoune. 

I would like to ask you if yoy could make a MSYS2/mingw64  package also for the RealThunder fork of FreeCAD that you may find at https://github.com/realthunder/FreeCAD/tree/LinkMerge.
I need it to for my layouDD package (https://github.com/wsteffe/layoutDD).

Sorry if my post is a little bit out of place but I haven't found any another way to contact you.

Thanks
Walter

MehdiChinoune, 2023-11-24:
@wsteffe Please open/fill an issue on msys2/MINGW-package repo.

wsteffe, 2023-11-24:
Done.