FreeCAD/FreeCAD #11452: [Problem] [TechDraw] Some fonts render clipped in dimensions

alexneufeld, 2023-11-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When using certain fonts in dimensions, the text renders higher than normal above the dimension arrow, and the top of the glyphs is clipped.

Example, using [https://www.fontsquirrel.com/fonts/Orbitron#](https://www.fontsquirrel.com/fonts/Orbitron#)
[image]

The issue also occurs with this font: [http://legionfonts.com/fonts/dinengschriftstd](http://legionfonts.com/fonts/dinengschriftstd)

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35074 (Git)
Build type: Unknown
Branch: main
Hash: f30e643c828cbc901377184e752619253308b4f1
Python 3.11.5, Qt 5.15.11, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: English/Canada (en_CA)
Installed mods: 
  * test_module
  * Curves 0.6.13
  * python_site_packages
  * Pyramids-and-Polyhedrons
  * FreeCAD-Flat-Icons
  * Assembly3 0.12.0
  * OpenDark 2023.10.30
  * Assembly4.backup1696795971.8344212 (Disabled)
  * robotpy (Disabled)
  * brlcad_wb
  * fasteners 0.4.62
  * FreeCAD-PreferencePacks.backup1669700355.378573
  * Behave-Dark-Colors 0.1.1
  * freecad.gears 1.0.0
  * Assembly4 0.50.4
  * FreeCAD_SheetMetal 0.2.56
  * Silk
```
```

### Subproject(s) affected?

Techdraw

### Anything else?

[font_error.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/13404729/font_error.FCStd.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-11-20:
Are we supposed to make a list of fonts that are acceptable or make specific alterations for their idiosyncrasies?

WandererFan, 2023-11-25:
The vertical compression of dimension and tolerance text comes from these PRs: #4082,  #4103.   The problem is that these PR's assume that there is always significant empty space above the glyph in the bounding rectangle.   In the case of the Orbitron font, the glyph is very close to the upper boundary of the bounding rectangle,  so when the bounding rectangle is reduced for tighter spacing, it removes part of the glyph.

I don't know of any way to know the position of the glyph within the bounding rectangle,  without parsing the actual font file.  

I'm thinking that a boolean "UseTightSpacing" property might be the only way to provide the tight spacing for "well behaved" fonts and still allow "odd" fonts to be used without clipping.

[image]
[image]

aapo-aapo, 2023-11-26:
Yes, these font things are very difficult, mainly because some fonts "reserve" very large empty areas around the glyphs, whereas other fonts don't.  Also, the Qt routines for calculating very tight bounding boxes does not seem to work very reliably, probably due to bugs in font files.  That said, I wonder if it would be possible to use the bounding box just for positioning, and just draw the offending glyphs outside the box if they happen to reach there.  I don't see any good reason to render the text within the clip groups; except that they seem to be automatically added by Qt.

luzpaz, 2023-11-27:
What's the way forward here ?

sslupsky, 2024-11-05:
This is still a problem on macOS with the Helvetica font.

OS: macOS 14.6.1
Architecture: x86_64
Version: 1.1.0dev.39100 (Git)
Build type: Release
Branch: main
Hash: 8865450a3e14220925e0e449c0f1f79056b4fb89
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default

[image]

luzpaz, 2024-11-06:
cc @WandererFan

commit e9c30cf3 (2025-01-13): [TD]Dimension font clipping (fix #11452) (#18771)

* [TD]separate alignment and rendering bounding rectangles

* [TD]use alignment rectangle (fix #11452)
commit e605d73d (2025-02-17): [TD]Dimension font clipping (fix #11452) (#18771)

* [TD]separate alignment and rendering bounding rectangles

* [TD]use alignment rectangle (fix #11452)
closed 2025-01-13 by chennes