FreeCAD/FreeCAD #11453: File Dialogue Can't See Contents of Crypttab Mapped & Fstab mounted Drive

DocTrucker, 2023-11-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The video shows me trying to view the contents of a symlinked folder in the stable and realthunder branches of Freecad and seeing what appears as a blank directory. I then browse to the same location with file browser and see contents. I have checked and symlinks to other directories not on this encrypted (but unlocked) drive are fine.

https://youtu.be/qNXWmJldjqE

This version of freecad was installed using "sudo snap install freecad"

A work around has been suggested that works (link below) but thought this issue should be raised as the issue presents with the standard dialogue window which is used by default by FreeCAD.

https://forum.freecad.org/viewtopic.php?p=719975

### Full version info

```shell
[code]
OS: Ubuntu Core 20 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git) Snap 759
Build type: Release
Branch: tag: 0.21.1
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.7, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
[/code]
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-11-20:
See #7596

DocTrucker, 2023-11-20:
That seems similar but I can't even see files in the browser after following the symlink to the unlocked drive or going through the /mnt/xxx mount point. I have tried following a symlink from my user directory to my documents folder also within my user directory and that worked fine.

DocTrucker, 2024-01-04:
This is a bigger problem than I first thought. With the native file browser I can see the folder, but it won't save files to there. oddly it will export stl files fine, but not the freecad files. It creates a temporary name for "BasePlaseHelper" that on my system became ".xdp-BasePlaseHelper.FCStd.99b869eb-caae-49ff-a144-6cb070033a0c-wY7pf1" but didn't make the "BasePlaseHelper.FCStd".

Yeah, it was supposed to be "BasePlateHelper", but typos struck...!

DocTrucker, 2024-01-16:
Found the problem. It's a snap settings issue.

Run the following in the command prompt:

snap connect gimp:removable-media :removable-media

closed 2024-01-16 by DocTrucker