FreeCAD/FreeCAD #11459: [Problem] Multiple usability issues with Points Editor (VectorListEditor) e.g. for Draft Wire

thkoch2001, 2023-11-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The Points property of a [Draft Wire](https://wiki.freecad.org/Draft_Wire) is internally represented by a VectorList property. The editor for this property is defined in `src/Gui/VectorListEditor.{cpp|h|ui}`.

This editor could use some improvements and this is probably a good project for somebody starting with FreeCAD development.

1. There is a mouse pointer button that is hardcoded to always be disabled. Thus there was an idea of what this button should do, but it was never implemented.
2. It is not clear for the user, that changes only get applied when clicking the checkmark button. Even after I learned this, I forget it often. I'm not sure yet about the best way to change the UI here.
3. It is not possible to use expressions like in other property fields.
4. There is also no Unit visible in the UI.
5. It would be very helpful, if the currently selected point would be highlighted in the 3D view.
6. It would also be helpful, if the currently selected point could be centered in the 3D view by clicking a button.
7. The 3D view can not be navigated while the editor is open.
8. It would be nice if changes could be applied while the editor is still open.
9. It would be nice if one could also use absolute values and not only relative ones.

Some observations not necessarily related to the editor but to Draft Wire:

10. Once the first point gets changed from (0, 0, 0), it gets harder to calculate the relative position of other points. So it would be nice to have a way to replace the wire to the new start point and thus make the first point (0, 0, 0) again.
11. It would be nice to have a way to switch the direction of a wire.
12. There is no documentation for the editor in the wiki. (And I'm waiting for getting a wiki account. :-) )

### Full version info

```shell
[code]
OS: KDE Flatpak runtime
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at f670854)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-11-20:
Keep in mind  that a VectorList might be used for many purposes. Some of what you say only makes sense for a Draft Wire.

Roy-043, 2023-11-20:
Note that the points of a Draft_Wire can already be edited in the 3D view. See https://wiki.freecad.org/Draft_Edit.

Roy-043, 2023-11-20:
2n note; wires can also be reversed with Draft_Edit.

maxwxyz, 2025-01-27:
I stumbled upon that issue because editing 2D Draft/BIM elements could definitely need an improvement.
When double clicking on a 2D Draft elemen (e.g. a line) you can edit the points. 
A blank task dialog opens with just the close button. There is no feedback for the user that he needs to click on a point or how the user could edit the line.
Once clicked on a point, you can edit that point.

[image]