FreeCAD/FreeCAD #11460: [Feature Request] zoom widget in footer

thkoch2001, 2023-11-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Other graphical programs or e.g. libreoffice have a zoom widget in the footer bar that allows zooming with + and - buttons, a slider, selectbox or text input and show the current zoom level. This would help when using FreeCAD without a mouse and thus without a scrolling wheel.

I just noticed that CTRL++ and CTRL+- are actually available. But they only work when the 3D view has focus and thus I initially thought they would not work at all. So this issue is less important now.

### Full version info

```shell
[code]
OS: KDE Flatpak runtime
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at f670854)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-12-14:
Since I have made few PRs concerning zoom, it is quite possible that I can do something in this case too. However, I just can't imagine usage of _FreeCAD_ without a mouse. How can this be implemented in the 3D view case? If your mouse does not work, you just can't use almost anything in _FreeCAD_. _Ctrl + Plus (+)_ and _Ctrl + Minus (-)_ work if the 3D view has the focus. What else one would need?

Is there a program that has similar feature? I don't see slider and zoom buttons in _MeshLab_ and _Blender_.

xtemp09, 2025-01-28:
> Other programs with zoom widgets: gimp, gwenview, inkscape, ...

They are not 3D viewers/editors.

> When you're working on a laptop you often don't have a mouse.

Does _Ctrl_ + _Plus_ (+) and _Ctrl_ + _Minus_ (-) work? Does these gestures work on your laptop?

[image]