FreeCAD/FreeCAD #11465: [Problem] [Part Design] Revolution has multiple issues in new modes

AjinkyaDahale, 2023-11-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Originally observed in [this comment in PR#7193](https://github.com/FreeCAD/FreeCAD/pull/7193#issuecomment-1795191219).

A non-exclusive list of issues:
1. In two dimensions mode the first angle cannot be set 0, but this is a perfectly valid value in that mode.
2. When starting the task on an existing revolution set to "two dimensions", it just launches with the options for (single) "dimension".
3. A default angle of 360° may be detrimental for the two dimensions mode and that the behavior when there is overlap between the angles is undesirable/confusing.

### Full version info

```shell
OS: Manjaro Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35083 +8 (Git)
Build type: Debug
Branch: pd-more-revol-options
Hash: 28fc763360f1758be28f52b92a83b064af962080
Python 3.11.5, Qt 5.15.11, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Curves 0.5.12
  * Curves.backup1662220878.4365516 0.4.4
  * ExtremeProDark 2.5.3
  * ProDarkThemePreferencePack 1.0.0
  * ProDarkThemePreferencePack.backup1662220881.5889003 1.0.0
```
```

### Subproject(s) affected?

None

### Anything else?

The feature was merged with the known issues to avoid further complications possible when rebasing and fixing conflicts.

Version declared is from the PR branch just before the it got merged. TODO: fill with main version when these features are

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

AjinkyaDahale, 2023-11-21:
@adrianinsaval anything to add?

adrianinsaval, 2023-11-21:
I'll repeat my suggestions for fixing 3.

- Get rid of "Dimension" mode as it doesn't provide anything more than "Two Dimensions", present "Two Dimensions" as default mode with Angle=Angle2=180° as default values. This means a full revolution is still the default but we don't get an unreasonable default for two dimensions mode. It's also a more useful mode even when you want to make a full revolution as you can play around with the values to control where the seam line will be. Symmetric option may still be provided which would disable the second angle, in light of that we could rename "Two Dimensions" to just "Dimensions".

- Make it such that whenever Angle + Angle2 >= 360° it's a full revolution, not a partial revolution like this:
[image]
Since these values might get used to control where the seam line will be the Angle value should be respected and Angle2 ignored and use whatever value to make it a full revolution. If possible we could even restrict Angle2 to have only values up to 360°-Angle

AjinkyaDahale, 2023-12-01:
>     2. When starting the task on an existing revolution set to "two dimensions", it just launches with the options for (single) "dimension".

This should be fixed with #11475.

> 1. In two dimensions mode the first angle cannot be set 0, but this is a perfectly valid value in that mode.
> 3. A default angle of 360° may be detrimental for the two dimensions mode and that the behavior when there is overlap between the angles is undesirable/confusing.

I don't see the issue with the default: this is basically same as "Dimension" mode default.

> * Get rid of "Dimension" mode as it doesn't provide anything more than "Two Dimensions",

We can do somewhat better: replace dimension 2 with "base offset" (which is what is happening internally anyway). This way, we can also have, say, a case where we have an angle on one side, and an up-to-face on the other. The only thing that doesn't work here is if we want to create a revolution from one face to another. Let me know if this is still unclear and I'll try creating some drawings.

> present "Two Dimensions" as default mode with Angle=Angle2=180° as default values.

This puts the "seam" opposite to where the profile is. I don't think there's any use in this.

> * Make it such that whenever Angle + Angle2 >= 360° it's a full revolution, not a partial revolution...
> Since these values might get used to control where the seam line will be the Angle value should be respected and Angle2 ignored and use whatever value to make it a full revolution. If possible we could even restrict Angle2 to have only values up to 360°-Angle

To avoid overlap, if we are using the "base offset", I suggest to remove limit on angle2 (now called offset), and limit angle1 to 0-360 deg, with the lower limit not allowed (so you can ask for as small an angle as you like, so long as it isn't 0).

AjinkyaDahale, 2023-12-01:
@adrianinsaval opinions?

69Piero, 2023-12-18:
"two dimensions option" should have also up to first/last/face in its sub-setting not only the angles value.

eg: i need 30°from a side and up to a face to the other side

what do you think?

AjinkyaDahale, 2023-12-26:
> eg: i need 30°from a side and up to a face to the other side
> 
> what do you think?

@69Piero what do you think of the idea I suggested in https://github.com/FreeCAD/FreeCAD/issues/11465#issuecomment-1835851386? It does have a weakness that it doesn't support a "from face to face" revolution but will support your example.

closed 2025-02-19 by github-actions[bot]