FreeCAD/FreeCAD #11472: [Problem] Sketcher does a poor job of indicating when a wire is closed.

obelisk79, 2023-11-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Part Design relies heavily on the sketcher. As such, it is heavily reliant on 'closed wires'. While the sketcher does an excellent job indicating whether or not all elements are fully constrained, it falls short letting a user know if a wire is closed or not. This is evident from a high frequency of issues reported (mostly by newer users) that are caused by open wires.

### Full version info

```shell
OS: EndeavourOS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35142 (Git)
Build type: None
Branch: main
Hash: 90dc4aeb136d30c5a107610f64e87fa307a761fa
Python 3.11.5, Qt 5.15.11, Coin 4.0.1, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * IconThemes
  * OpenDark 2023.10.30
  * freecad.gears 1.0.0
  * Curves 0.6.14
  * TabBar
  * SelectorToolbar
  * Dracula 0.0.5
```
```

### Subproject(s) affected?

Sketcher

### Anything else?

I propose that shapes drawn in sketcher would fill with a high transparency color when a wire is closed. This would provide valuable visual cues while not notably obscuring  other elements in the MainView.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2023-11-21:
Showing a surface whenever a wire is closed would help. Picture by jetjaguar on discord : 
[image]

PaddleStroke, 2023-11-21:
Also the idea of changing point colors based on coincidence status. See discussion : https://forum.freecad.org/index.php?style=8

obelisk79, 2023-11-21:
I don't think changing point colors in this context is a good idea. Sketcher is already too reliant on color for visual cues.

PaddleStroke, 2023-11-21:
Well this is true but the colors would not be additional colors if we use construction mode color and constraint color. But yes it adds one element switching of color.

Pesc0, 2023-11-21:
the image is wrong though, the second large circle, the small circles and the semicircles near the triangles should be shaded as well. they are all closed areas

koopdi, 2023-11-21:
@Pesc0 , I created a separate issue describing the problem of negative/positive space, which the image was meant to illustrate.

Pesc0, 2023-11-21:
Ah! Thanks for clarifying that. I think the end user shouldnt have to worry about positive and negative space. The areas should be equally shaded imo.

maxwxyz, 2024-01-05:
Related to https://github.com/FreeCAD/FreeCAD/issues/11799

adrianinsaval, 2024-01-16:
I would say #11799 is a duplicate, seems to be more active so I think it's better to have that one supersede this IMO

closed 2024-01-16 by adrianinsaval