FreeCAD/FreeCAD #11474: [Sketcher] Deactivated block constraints are not deactivated

PaddleStroke, 2023-11-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Deactivated block constraints are not deactivated
[image]

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

flaviut, 2023-11-23:
I was easily able to reproduce this.

[constraint-11474.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/13445331/constraint-11474.FCStd.zip)

```
OS: Arch Linux (sway)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: makepkg
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.11.6, Qt 5.15.11, Coin 4.0.1, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * kicadStepUpMod 10.20.7
  * Glass
  * PieMenu 1.2.7
```

maxwxyz, 2024-01-14:
Not sure if Constrain Block is working generally.
If a line has a block constraint I cannot set the line lenght but can dimension the horizontal distance and change it without a problem:
[image]

PaddleStroke, 2025-01-29:
still relevant

luzpaz, 2025-01-29:
Is this a straightforward fix or is there hidden complexity that needs to be taken into consideration ?