FreeCAD/FreeCAD #11479: [Problem] [path]

snelweg, 2023-11-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The path_sanity command stopped working again in the dev version of FC

### Full version info

```shell
OS: Linux Mint 21.2 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35081 (Git) AppImage
Build type: Release
Branch: main
Hash: 32db044caa92a2adc448850179c524b793a03c3e
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods: 
  * Assembly4 0.50.4
  * btl 0.9.9
```
```

### Subproject(s) affected?

None

### Anything else?

Running the Python command 'Path_Sanity' failed:
Traceback (most recent call last):
  File "/tmp/.mount_FreeCAOnAqWP/usr/Mod/Path/Path/Main/Gui/Sanity.py", line 156, in Activated
    html = self.__report(data, obj)
  File "/tmp/.mount_FreeCAOnAqWP/usr/Mod/Path/Path/Main/Gui/Sanity.py", line 615, in __report
    reportHtmlTemplate += i["cycleTime"]

can only concatenate str (not "float") to str

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2023-11-21:
So any file with any Post Processor doesn't work, Sanity hasn't changed for 2 months?

snelweg, 2023-11-22:
I have not used the Sanity functions for a while, so maybe the fault was there already for a long time.

Previous problems I could solve by not using the substitutions in the output file name of a job (sanity does not handle the substitutions). This time that doesn't help.

Syres916, 2023-11-22:
Attached at the end of this post  is a file for you to test, it has a substitution in the job info as per:

[image]

Tested using:

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35144 (Git)
Build type: Release
Branch: main
Hash: d7927279bd6242b17784edecd7181e49b1927815
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Silk 0.1.3
  * Defeaturing 1.2.0
  * CurvedShapes 1.0.4
  * freecad.gears 1.0.0
  * Curves 0.6.13
  * Help 1.0.3
  * fasteners 0.4.56
  * sheetmetal 0.3.0
```

Cycle time output as expected:

[image]

[PathSanitySpecial.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/13440312/PathSanitySpecial.FCStd.zip)

This should be tested on your locale so it can be determined if the formatting of the time field is different to English and so creating an unwanted float. If that is the case then I'm going to be struggling to replicate it without some debugging on your side.

snelweg, 2023-11-23:
Changed locale to en_US

No change at all. There are some images rendered in the job directory, but not the Asciidoc files.

> xxx@xxx-HP-ProDesk-600-G1-SFF$ which asciidoc
/usr/bin/asciidoc
xxx@xxxHP-ProDesk-600-G1-SFF$ asciidoc --version
asciidoc 10.1.2

`03:15:57  Running the Python command 'Path_Sanity' failed:
Traceback (most recent call last):
  File "/tmp/.mount_FreeCAQN8sUt/usr/Mod/Path/Path/Main/Gui/Sanity.py", line 156, in Activated
    html = self.__report(data, obj)
  File "/tmp/.mount_FreeCAQN8sUt/usr/Mod/Path/Path/Main/Gui/Sanity.py", line 822, in __report
    for o in value["ops"]:

('ops',)`

```````
OS: Linux Mint 21.2 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35081 (Git) AppImage
Build type: Release
Branch: main
Hash: 32db044caa92a2adc448850179c524b793a03c3e
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.4
  * btl 0.9.9

```````

[image]
[image]

Syres916, 2023-11-23:
As you can see I submitted a possible fix for the Cycle time error but the fact that you've opened a perfectly good file, changed the Job settings and lost all of the Ops seems like there's something seriously wrong on your setup. By the way you won't see any Asciidoc files as the whole point of the change was to eliminate this unnecessary dependency. The Html is generated on the fly and then opened on completion.

All I can advise is to use the 0.21.1 stable version until there's a way to replicate your setup.

snelweg, 2023-11-23:
> As you can see I submitted a possible fix for the Cycle time error 

Nice that you have been able to fix that error

>but the fact that you've opened a perfectly good file, changed the Job settings and lost all of the Ops seems like there's something seriously wrong on your setup. 

The ops are not lost, they are still there. It's just that Sanity does not recognize them

>By the way you won't see any Asciidoc files as the whole point of the change was to eliminate this unnecessary dependency. The Html is generated on the fly and then opened on completion.

Aha, that's nice! Would love to see it in action.

> All I can advise is to use the 0.21.1 stable version until there's a way to replicate your setup.

I will do without the Sanity extra's then. But again, if you think the ops are missing: they are not.

[image]

closed 2023-12-04 by sliptonic