FreeCAD/FreeCAD #11482: [Problem] Diffucult to Discern Positive/Negative Space in Sketcher

koopdi, 2023-11-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When creating nested islands of closed geometry in sketcher, it can be difficult to tell apart the negative and positive space.
In the image below, it is not easy to tell which of the small circles will be filled (left).
[image]
Issue: #11472 is related but distinct. It is possible that a common solution could be found.

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * 3DfindIT 1.2.0
  * A2plus 0.4.60n
  * Assembly4 0.50.4
  * Curves 0.6.14
  * Defeaturing 1.2.1
  * DynamicData 2.53.0
  * ExplodedAssembly
  * fasteners 0.4.70
  * freecad.gears 1.0.0
  * Help 1.0.3
  * Manipulator 1.5.0
  * Nodes 0.1.36
[/code]
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2023-11-22:
A toggle or setting would be great to enable a slightly transparent overlay of positive closed loops.

pierreporte, 2023-11-28:
This is a great feature. It already exists in some CAD programs. However it is going to need #7334 resolved because the new color will need to go well with construction lines and constraints.

In my opinion, the fill color should be very close to the background color, just opaque enough to be distinguished and understood, but discreet enough to allow working comfortably with the sketch.

maxwxyz, 2023-12-21:
Additionally a slightly colored overlay would be great for the islands:

- islands not fully constrained: default color
- island defined by fully constrained geometry: green
- islands defined by only construction lines: no color or blue

I guess overall the islands or spaces should only be drawn between normal sketch geometry, not construction lines, internal geometry or external references.

pierreporte, 2024-01-03:
@maxwxyz Aren’t line colors sufficient to indicate how the shapes are constrained?

maxwxyz, 2024-01-04:
I think there are already too many colors 😂

pierreporte, 2024-01-04:
@maxwxyz Yes this is exactly my point.

pierreporte, 2024-04-21:
This feature would still be useful and a first step to determine if the profile is closed, thus allowed for most features. But there should be an option in the sketches to disable it so it doesn’t get in the way when the sketch is not going to be used for this. For example, master sketches don’t need to be closed and showing it doesn’t make sense when they happen to be.