FreeCAD/FreeCAD #11488: [Problem] Segfault when making part intersection

rezahousseini, 2023-11-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Having two bodys and only selecting one then clicking on difference in the Part menu a segfault occurs with the following backtrace:

```
Program received signal SIGSEGV, Segmentation fault.
#0  /gnu/store/ln6hxqjvz6m9gdd9s97pivlqck7hzs99-glibc-2.35/lib/libc.so.6(+0x3d2a0) [0x7f08d7c412a0]
#1  0x7f080558e4ea in CmdPartCommon::activated(int) from /gnu/store/31zvvmh38wv9w2l6px02asm9msq566l7-freecad-0.21.1/lib/PartGui.so+0x56a
#2  0x7f08dbc6d77e in Gui::Command::_invoke(int, bool) from /gnu/store/31zvvmh38wv9w2l6px02asm9msq566l7-freecad-0.21.1/lib/libFreeCADGui.so+0x1fe
#3  0x7f08dbc6db48 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /gnu/store/31zvvmh38wv9w2l6px02asm9msq566l7-freecad-0.21.1/lib/libFreeCADGui.so+0x108
#4  /gnu/store/pgwj7lh1fr6l6h2hsz1zv404sjkqgi3r-qtbase-5.15.10/lib/libQt5Core.so.5(+0x2df5fd) [0x7f08d84df5fd]
#5  0x7f08d9174a92 in QAction::triggered(bool) from /gnu/store/pgwj7lh1fr6l6h2hsz1zv404sjkqgi3r-qtbase-5.15.10/lib/libQt5Widgets.so.5+0x32
#6  0x7f08d9177444 in QAction::activate(QAction::ActionEvent) from /gnu/store/pgwj7lh1fr6l6h2hsz1zv404sjkqgi3r-qtbase-5.15.10/lib/libQt5Widgets.so.5+0xa4
#7  /gnu/store/pgwj7lh1fr6l6h2hsz1zv404sjkqgi3r-qtbase-5.15.10/lib/libQt5Widgets.so.5(+0x26939a) [0x7f08d926939a]
#8  0x7f08d92694e3 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /gnu/store/pgwj7lh1fr6l6h2hsz1zv404sjkqgi3r-qtbase-5.15.10/lib/libQt5Widgets.so.5+0xe3
#9  0x7f08d935342a in QToolButton::mouseReleaseEvent(QMouseEvent*) from /gnu/store/pgwj7lh1fr6l6h2hsz1zv404sjkqgi3r-qtbase-5.15.10/lib/libQt5Widgets.so.5+0xa
#10  0x7f08d91b987e in QWidget::event(QEvent*) from /gnu/store/pgwj7lh1fr6l6h2hsz1zv404sjkqgi3r-qtbase-5.15.10/lib/libQt5Widgets.so.5+0x1be
#11  0x7f08d917aaaf in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /gnu/store/pgwj7lh1fr6l6h2hsz1zv404sjkqgi3r-qtbase-5.15.10/lib/libQt5Widgets.so.5+0x7f
#12  0x7f08d9181f69 in QApplication::notify(QObject*, QEvent*) from /gnu/store/pgwj7lh1fr6l6h2hsz1zv404sjkqgi3r-qtbase-5.15.10/lib/libQt5Widgets.so.5+0xa49
#13  0x7f08dbc229c8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /gnu/store/31zvvmh38wv9w2l6px02asm9msq566l7-freecad-0.21.1/lib/libFreeCADGui.so+0x68
#14  0x7f08d84ab43a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /gnu/store/pgwj7lh1fr6l6h2hsz1zv404sjkqgi3r-qtbase-5.15.10/lib/libQt5Core.so.5+0x11a
#15  0x7f08d9180b00 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /gnu/store/pgwj7lh1fr6l6h2hsz1zv404sjkqgi3r-qtbase-5.15.10/lib/libQt5Widgets.so.5+0x1a0
#16  /gnu/store/pgwj7lh1fr6l6h2hsz1zv404sjkqgi3r-qtbase-5.15.10/lib/libQt5Widgets.so.5(+0x1d1d53) [0x7f08d91d1d53]
#17  /gnu/store/pgwj7lh1fr6l6h2hsz1zv404sjkqgi3r-qtbase-5.15.10/lib/libQt5Widgets.so.5(+0x1d4e15) [0x7f08d91d4e15]
#18  0x7f08d917aaaf in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /gnu/store/pgwj7lh1fr6l6h2hsz1zv404sjkqgi3r-qtbase-5.15.10/lib/libQt5Widgets.so.5+0x7f
#19  0x7f08dbc229c8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /gnu/store/31zvvmh38wv9w2l6px02asm9msq566l7-freecad-0.21.1/lib/libFreeCADGui.so+0x68
#20  0x7f08d84ab43a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /gnu/store/pgwj7lh1fr6l6h2hsz1zv404sjkqgi3r-qtbase-5.15.10/lib/libQt5Core.so.5+0x11a
#21  0x7f08d8951517 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /gnu/store/pgwj7lh1fr6l6h2hsz1zv404sjkqgi3r-qtbase-5.15.10/lib/libQt5Gui.so.5+0x677
#22  0x7f08d89288fc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /gnu/store/pgwj7lh1fr6l6h2hsz1zv404sjkqgi3r-qtbase-5.15.10/lib/libQt5Gui.so.5+0xac
#23  /gnu/store/pgwj7lh1fr6l6h2hsz1zv404sjkqgi3r-qtbase-5.15.10/lib/libQt5XcbQpa.so.5(+0x68c8a) [0x7f08d3fa2c8a]
#24  /gnu/store/7vmjnwqjp22jhs93rarxf5qfxm610jhm-glib-2.72.3/lib/libglib-2.0.so.0(g_main_context_dispatch+0x24b) [0x7f08d561f98b]
#25  /gnu/store/7vmjnwqjp22jhs93rarxf5qfxm610jhm-glib-2.72.3/lib/libglib-2.0.so.0(+0x55bf8) [0x7f08d561fbf8]
#26  /gnu/store/7vmjnwqjp22jhs93rarxf5qfxm610jhm-glib-2.72.3/lib/libglib-2.0.so.0(g_main_context_iteration+0x2f) [0x7f08d561fc9f]
#27  0x7f08d84fef34 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /gnu/store/pgwj7lh1fr6l6h2hsz1zv404sjkqgi3r-qtbase-5.15.10/lib/libQt5Core.so.5+0x64
#28  0x7f08d84a9fbb in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /gnu/store/pgwj7lh1fr6l6h2hsz1zv404sjkqgi3r-qtbase-5.15.10/lib/libQt5Core.so.5+0x11b
#29  0x7f08d84b1c0c in QCoreApplication::exec() from /gnu/store/pgwj7lh1fr6l6h2hsz1zv404sjkqgi3r-qtbase-5.15.10/lib/libQt5Core.so.5+0x7c
#30  0x7f08dbb5e6a8 in Gui::Application::runApplication() from /gnu/store/31zvvmh38wv9w2l6px02asm9msq566l7-freecad-0.21.1/lib/libFreeCADGui.so+0x2528
#31  /gnu/store/xwl62nbba217i5qs9i9s4bg82gjw6gv0-profile/bin/FreeCAD() [0x4089b5]
#32  /gnu/store/ln6hxqjvz6m9gdd9s97pivlqck7hzs99-glibc-2.35/lib/libc.so.6(+0x291f7) [0x7f08d7c2d1f7]
#33  /gnu/store/ln6hxqjvz6m9gdd9s97pivlqck7hzs99-glibc-2.35/lib/libc.so.6(__libc_start_main+0x7c) [0x7f08d7c2d2ac]
#34  /gnu/store/xwl62nbba217i5qs9i9s4bg82gjw6gv0-profile/bin/FreeCAD() [0x408c51]
```

### Full version info

```shell
[code]
OS: Guix System (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.1.Unknown
Build type: Release
Python 3.10.7, Qt 5.15.10, Coin 4.0.0, Vtk 9.2.2, OCC 7.6.2
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.4.56
  * Assembly4 0.12.7
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2023-11-23:
> Version: 0.21.1.Unknown

How is this possible?

rezahousseini, 2023-11-23:
I tried to reproduce but I couldn't. It is really hard to try to reproduce this kind of bugs where you click in the GUI for several minutes until it suddenly crashes. Is there no log with all the keyboard and mouse events to reproduce this kind of bugs?

xtemp09, 2023-11-23:
There is [debugging log](https://wiki.freecad.org/Debugging) that can be turn on with `-l` or `--write-log` option

@rezahousseini, can you reproduce the same crash with [_AppImage_](https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)?

-----

_P.S. If the crash cannot be reproduced, then I just don't know what to do with it._

luzpaz, 2023-11-27:
> How is this possible?

@xtemp09 IIRC some downstream repos package in a way where they don't pull directly from FreeCAD/FreeCAD and therefore don't run certain build scripts in the process. This creates this anomaly. The hash though can let us know: f6708547a9bb3f71a4aaade12109f511a72c207c

maxwxyz, 2024-06-13:
I'll close this as there is no further feedback and the issue cannot be reprocduced.

closed 2024-06-13 by maxwxyz