FreeCAD/FreeCAD #11493: (261): Null geometry in isBsplinePole - please report

Figgypudding, 2023-11-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

14:03:23  OpenGL version is: 4.6 (4.6 (Compatibility Profile) Mesa 22.3.6)
14:03:23  Run Gui init script
14:03:24  Init: Running FreeCADGuiInit.py start script...
14:03:24  Init:   Searching modules...
14:03:25  Init:      Initializing /usr/share/freecad/Mod/Plot(InitGui.py not found)... ignore
14:03:28  Init:      Initializing /usr/share/freecad/Mod/Tux... done
14:03:28  Init:      Initializing /usr/share/freecad/Mod/AddonManager... done
14:03:28  Init:      Initializing /usr/share/freecad/Mod/Sketcher... done
14:03:28  Init:      Initializing /usr/share/freecad/Mod/Test... done
14:03:28  Init:      Initializing /usr/share/freecad/Mod/Path... done
14:03:28  Init:      Initializing /usr/share/freecad/Mod/Measure(InitGui.py not found)... ignore
14:03:28  Init:      Initializing /usr/share/freecad/Mod/ReverseEngineering... done
14:03:28  Init:      Initializing /usr/share/freecad/Mod/PartDesign... done
14:03:28  Init:      Initializing /usr/share/freecad/Mod/Part... done
14:03:28  Init:      Initializing /usr/share/freecad/Mod/OpenSCAD... done
14:03:28  Init:      Initializing /usr/share/freecad/Mod/Start... done
14:03:28  Init:      Initializing /usr/share/freecad/Mod/Show(InitGui.py not found)... ignore
14:03:29  Init:      Initializing /usr/share/freecad/Mod/Inspection... done
14:03:29  Init:      Initializing /usr/share/freecad/Mod/Draft... done
14:03:29  Init:      Initializing /usr/share/freecad/Mod/Web... done
14:03:29  Init:      Initializing /usr/share/freecad/Mod/TechDraw... done
14:03:30  Init:      Initializing /usr/share/freecad/Mod/Material... done
14:03:30  Init:      Initializing /usr/share/freecad/Mod/Fem... done
14:03:30  Init:      Initializing /usr/share/freecad/Mod/Mesh... done
14:03:30  Init:      Initializing /usr/share/freecad/Mod/Idf(InitGui.py not found)... ignore
14:03:30  Init:      Initializing /usr/share/freecad/Mod/Surface... done
14:03:30  Init:      Initializing /usr/share/freecad/Mod/Image... done
14:03:30  Init:      Initializing /usr/share/freecad/Mod/Import... done
14:03:30  Init:      Initializing /usr/share/freecad/Mod/Drawing... done
14:03:30  Init:      Initializing /usr/share/freecad/Mod/Spreadsheet... done
14:03:30  Init:      Initializing /usr/share/freecad/Mod/MeshPart... done
14:03:30  Init:      Initializing /usr/share/freecad/Mod/Raytracing... done
14:03:30  Init:      Initializing /usr/share/freecad/Mod/Robot... done
14:03:31  Init:      Initializing /usr/share/freecad/Mod/Points... done
14:03:32  Init:      Initializing /usr/share/freecad/Mod/Arch... done
14:03:32  Init:      Initializing /home/frnk01/.local/share/FreeCAD/Mod/lattice2/./... done
14:03:32  All modules with GUIs using InitGui.py are now initialized
14:03:32  All modules with GUIs initialized using pkgutil are now initialized
14:03:32  Init: Loading FreeCAD GUI
14:03:32  Init: Running FreeCADGuiInit.py start script... done
14:03:45  Init: Activating default workbench StartWorkbench
14:03:45  Module: Part
14:03:45  Loading Part module... done
14:03:45  Loading Sketcher module... done
14:03:45  Loading PartDesign module... done
14:03:45  Loading GUI of Part module... done
14:03:45  Loading GUI of Sketcher module... done
14:03:45  Loading GUI of PartDesign module... done
14:03:45  Init: Showing main window
14:03:45  Main window restored
14:03:45  Show main window
14:03:45  Toolbars restored
14:03:45  Couldn't connect to spacenav daemon. Please ignore if you don't have a spacemouse.
14:03:45  Loading Mesh module... done
14:03:45  Loading GUI of Mesh module... done
14:03:45  Loading GUI of MeshPart module... done
14:03:46  Init: Entering event loop
14:03:46  Init: Executing event loop...
14:03:46  Init: Processing command line files
14:03:46  Init:     Processing file: -
14:03:46  Init:     Processing file: /home/frnk01/3D Objects/tri-laser-sight/3-laser-sight.FCStd
14:14:49  Sketcher::RedundantSolving-DogLeg-
14:14:49  Sketcher Redundant solving: 1 redundants
14:14:49  Sketcher::setUpSketch()-T:0.127
[... 1195 lines cut ...]

```shell
[code]
OS: Debian GNU/Linux 12 (bookworm) (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.20.2.
Build type: Release
Python 3.11.1, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * lattice2 1.0.0
[/code]
```

### Subproject(s) affected?

None

### Anything else?

never done this before appologies if it is wrong

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

flaviut, 2023-11-23:
> the file I was working on - will not paste into this box

If you compress your FCStd file into a zip it'll allow you to upload it here.

luzpaz, 2023-11-23:
Report is for v0.20 which is old. V0.21.2 is latest stable. Please test and report back. Also share example file.

luzpaz, 2023-11-27:
@Figgypudding bump

luzpaz, 2023-11-28:
@Figgypudding ping

Figgypudding, 2023-11-28:
Hi sorry for not coming back to you.  This is my first issue reported.  I don't know your text short cut yet.

I do not have the file Owas working on.  As a beginner with FreeCAD I'm used to abandoning my files and starting over till I complete the file I want.  Unfortunately I didn't isolate this one and managed to over write it.  Next time I'll know the protocol and be more cautious.

Whilst doing the drawing :-
1, I started in the previous version of FreeCAD and completed it by drawing everything.
2, As I wanted to adjust some dimensions I started a new drawing (having gone the long way round to install it on Debian) in the new version.
3, I tried to do it with the main loop and a polar array for the 3 small horse shoes.  This did not work as the solver would not join all the end points and continued not to during the 5 or so other attempts at trying different thing to get it to close the wire.
4, I decided that I would try to do it by using the 3 section polar array but instead of doing it just for the small bits, to cut the large circle as well with 1 of the small one's in the centre of the large circle's the 120 degree section.

This finally closed the section to enable me to complete the drawing, mesh it and 3d print it.  However in doing so I over write the drawing you wanted.  Apologies I won't do it again, I didn't think, I'm very used to rubbing items out of my drawing tree.

If you want me to forward the new drawing let me know.

Thanks for all the hard work you do.
Regards
Frank

DCSven, 2024-02-02:
Hi,
I got a fault message with the Headline "Exception" and the content "Null geometry in isBsplinePole - Please report.". I'm using FreeCad 0.21.2 (Portable version). It occurred while I was changing a value of an angle  in a sketch001. I Uploaded the model together with two print screens (saying about the same as mentioned in the beginning of my note).
[isBsplinePole_2024-02-02.zip](https://github.com/FreeCAD/FreeCAD/files/14139319/isBsplinePole_2024-02-02.zip)
After I got the fault message, I continued to add/edit value of a similar constraint, but this time there was no issue and it was possible to continue designing.

luzpaz, 2024-03-15:
@DCSven can you reproduce on latest development version ? https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Figgypudding, 2024-03-15:
Hi luzpaz, I have given up on Debian as it took so long to get anything done.  I replaced it with Mint deb version, without using FreeCAD on it.

I went back to windows for FreeCAD use.

I will try to emulate the fault but I am on the current windows version, it's 4-5 months since I did the drawing with details a little hazy and I wiped the drawing with the issue.  I don't hold out much hope that I can replicate it, but I will try.

luzpaz, 2024-03-15:
Thanks @Figgypudding, appreciate the perseverance. Everything we can reproduce is potentially fixable and that improves FC for everyone. Cheers

luzpaz, 2024-07-24:
Any progress here ?

luzpaz, 2024-08-27:
@Figgypudding @DCSven bump

AjinkyaDahale, 2025-03-14:
For future reference, this error is in `src/Mod/Sketcher/Gui/Utils.cpp`, in `SketcherGui::isBsplinePole(const Part::Geometry* geo)`. I don't know how this happens, however.

luzpaz, 2025-06-08:
> For future reference, this error is in `src/Mod/Sketcher/Gui/Utils.cpp`, in `SketcherGui::isBsplinePole(const Part::Geometry* geo)`. I don't know how this happens, however.

@AjinkyaDahale can you trigger the problem still? I couldn't (using @DCSven's example).

```
OS: Manjaro Linux (KDE/plasmawayland/wayland)
Architecture: x86_64
Version: 1.1.0dev.42095 (Git) AppImage
Build date: 2025/06/06 16:36:48
Build type: Release
Branch: main
Hash: 9c017b0dbf538c2570df7c3316355cdcdc3163f5
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/95.065/1
Installed mods: 
  * fcVM
  * FreeCAD-Ribbon 1.9.1
  * SearchBar 1.4.2
  * Silk 0.1.8
  * Telemetry 1.0.1
```

DCSven, 2025-08-25:
OK, with me.