FreeCAD/FreeCAD #11498: [Problem] Sketcher gets laggy when dragging complex constraint systems

flaviut, 2023-11-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have this sketch:

[image]

When I click and drag the selected point, FreeCAD uses 100% of one of my CPU cores, behaves sluggishly (0.1fps), and takes about 30 seconds to recover.

[box.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/13445211/box.FCStd.zip)

### Full version info

```shell
OS: Arch Linux (sway)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: makepkg
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.11.6, Qt 5.15.11, Coin 4.0.1, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * kicadStepUpMod 10.20.7
  * Glass
  * PieMenu 1.2.7
```

### Subproject(s) affected?

None

### Anything else?

I suspect the issue here is that the MouseMove event is being called too frequently. I tested this by attaching to FreeCAD with GDB:

```
(gdb) b SIM::Coin3D::Quarter::EventFilter::eventFilter
Breakpoint 1 at 0x7fb6b538f640
(gdb) ignore 1 1000000
Will ignore next 1000000 crossings of breakpoint 1.
(gdb) info breakpoints
Num     Type           Disp Enb Address            What
1       breakpoint     keep y   0x00007fb6b538f640 <SIM::Coin3D::Quarter::EventFilter::eventFilter(QObject*, QEvent*)>
	ignore next 1000000 hits
```

1. I started a timer and moved the point for 10s. I then moved my cursor out of freecad, and paused the process once CPU had gone back to 0%. This lead to `eventFilter` being called 2200 times.
2. I started a timer and just moved my cursor into freecad, then after 10s, moved my cursor out of freecad. This lead to `eventFilter` being called 373 times.
3. I moved the point for about 1s. I then moved my cursor out of freecad, and paused the process once CPU had gone back to 0%. This lead to `eventFilter` being called 763 times.
4. I moved my cursor into freecad and waited about 1s. I then moved my cursor out of freecad. This lead to `eventFilter` being called 142 times.

This shows that the mouse move event is being handled about 621 times in 1 second. This is more than necessary--we don't need to repeat `SketcherGui::EditModeConstraintCoinManager::processConstraints` that often. 60Hz ought to be the absolute maximum, and I think 20-30Hz is plenty.

[I've attached the profiler data (for speedscope.app)](https://github.com/FreeCAD/FreeCAD/files/13445256/profile.zip)

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-11-24:
Side note: Your sketch has 300+ constraints. This is considered bad practice.

flaviut, 2023-11-24:
> Side note: Your sketch has 300+ constraints. This is considered bad practice.

no argument here. but my opinion is that if it is a number I (a human) can count to, then software should not struggle with it.

Roy-043, 2023-11-24:
The number of equations the solver has to process increases exponentially with the number of constraints.

flaviut, 2023-11-24:
I don't want to argue, but I want to really understand how the constraints system works. I understand a bit of linear algebra, but am not familiar with non-linear systems of equations.

My intuition is that each constraint adds a fixed number of equations to the problem, i.e the number of equations is linear with the number of constraints. For example, the vertical constraint adds one equation: $x_1 = x_2$. The coincident constraint adds two equations: $x_1 = x_2; y_1 = y_2$. I'm out of my depth when it comes to solving this system, but it looks like it should just be quadratic or or $n^3$

---

Anyway, I've done some experimentation here. This code works fine for this situation... where the constraint solving takes 100-200ms on my computer. But it obviously is not frequent enough for faster solutions, and decreasing the interval ends up not improving the lag.

```patch
diff --git a/src/Mod/Sketcher/Gui/ViewProviderSketch.cpp b/src/Mod/Sketcher/Gui/ViewProviderSketch.cpp
index f9716bc12b..6091fc7dfc 100644
--- a/src/Mod/Sketcher/Gui/ViewProviderSketch.cpp
+++ b/src/Mod/Sketcher/Gui/ViewProviderSketch.cpp
@@ -453,6 +453,8 @@ SbVec2s ViewProviderSketch::DoubleClick::prvClickPos;// used by double-click-det
 SbVec2s ViewProviderSketch::DoubleClick::prvCursorPos;
 SbVec2s ViewProviderSketch::DoubleClick::newCursorPos;
 
+SbTime ViewProviderSketch::MouseDrag::prvDragPassTime;
+
 //**************************************************************************
 // Construction/Destruction
 
@@ -1374,6 +1376,8 @@ bool ViewProviderSketch::mouseMove(const SbVec2s& cursorPos, Gui::View3DInventor
 {
     // maximum radius for mouse moves when selecting a geometry before switching to drag mode
     const int dragIgnoredDistance = 3;
+    // minimum time (s) between constraint re-solves when dragging points
+    const double dragThrottleTime = 0.2;
 
     static bool selectableConstraints = true;
 
@@ -1388,8 +1392,8 @@ bool ViewProviderSketch::mouseMove(const SbVec2s& cursorPos, Gui::View3DInventor
     if (!isInEditMode())
         return false;
 
-    // ignore small moves after selection
     switch (Mode) {
+        // ignore small moves after selection
         case STATUS_SELECT_Point:
         case STATUS_SELECT_Edge:
         case STATUS_SELECT_Constraint:
@@ -1398,6 +1402,16 @@ bool ViewProviderSketch::mouseMove(const SbVec2s& cursorPos, Gui::View3DInventor
             (cursorPos - DoubleClick::prvCursorPos).getValue(dx, dy);
             if (std::abs(dx) < dragIgnoredDistance && std::abs(dy) < dragIgnoredDistance)
                 return false;
+            break;
+        // throttle constraint solves to avoid lagging (default event interval is 200Hz+)
+        case STATUS_SKETCH_DragPoint:
+        case STATUS_SKETCH_DragConstraint:
+        case STATUS_SKETCH_DragCurve:
+            if((SbTime::getTimeOfDay() - MouseDrag::prvDragPassTime).getValue() < dragThrottleTime) {
+                return false;
+            }
+            MouseDrag::prvDragPassTime = SbTime::getTimeOfDay();
+            break;
         default:
             break;
     }
diff --git a/src/Mod/Sketcher/Gui/ViewProviderSketch.h b/src/Mod/Sketcher/Gui/ViewProviderSketch.h
index 792a81d356..a9eca67a8a 100644
--- a/src/Mod/Sketcher/Gui/ViewProviderSketch.h
+++ b/src/Mod/Sketcher/Gui/ViewProviderSketch.h
@@ -449,6 +449,12 @@ private:
         static SbVec2s newCursorPos;
     };
 
+    struct MouseDrag
+    {
+        /** the last time when a mouse drag event was handled (and not rejected due to throttling) */
+        static SbTime prvDragPassTime;
+    };
+
     /** @brief Private struct grouping ViewProvider parameters and internal variables
      */
     struct ViewProviderParameters
```

The problem here is really a queuing problem. The solution here would be:

1. build up a queue of mouse drag events
2. a worker should read off the queue, and each time it looks for work, it should do according to the following algorithm, since the intermediate drag events are not important:
   ```python
   drag_events = Queue()
   def worker():
     while True:
       event = drag_events.get(block=True)
       while not drag_events.empty():
         event = drag_events.get(block=False)
       process(event)
   ```

I'm not at all familiar with the FreeCAD code, but it seems like this kind of code would be very difficult to implement in the current architecture.

howie-j, 2023-11-25:
@abdullahtahiriyo is the main developer and maintainer for sketcher, and has written this extensive document on how the solver/constrain system works: [fc-solver-latex.pdf](https://github.com/FreeCAD/FreeCAD/files/13462593/fc-solver-latex.pdf)

flaviut, 2023-11-25:
Investigating even further: it seems like the issue is not at all the constraints solver system. Solving the constraints takes 3% of the time here; the other 86% of the time is spent trying to lay out the constraints in the UI.

if I just replace the `EditModeConstraintCoinManager::seekConstraintPosition` method with `return norm * 0.5f * step;`, it's incredibly snappy. The layout, by coincidence I think, actually better in this case with fewer overlaps:

[image]

flaviut, 2023-11-30:
> Why there is a limit in constraints? It should not

there is no fixed limit on constraints, and agreed there should not be. my part is now 2200 constraints and it works fine with my patch. Slow… but tolerable. 17.82s to solve for 2250 constraints on my machine

I don't really understand linear algebra, but constraint matrices seem simple:

[image]

And [according to this paper](https://arxiv.org/pdf/1802.03773.pdf) it is possible that there is an algorithm with far better performance here[^1]:
[image]

> what should be limited its the number of constraints shown in screen

Also agreed--though [I suspect many folks will disagree with me](https://forum.freecad.org/viewtopic.php?t=82978), for good reason. Although I think it is possible to handle those use cases through careful design.

> and being recalculated, and all constraints should be calculated only as a final result

unfortunately this is not possible. the math involved requires all constraints be included in the evaluation.

[^1]: "SuiteSparse QR (SPQR) SuiteSparse implementation of multifrontal rank-revealing QR factorization [12] using BLAS [ 26 ] and LAPACK [ 2]. It does not assume any sparsity pattern in advance."

flaviut, 2025-01-29:
Freecad nightly works well with the 300 constraints in the OP.

The issue is still present for 1100 constraints, but FreeCAD is usable: ~.2fps while dragging, no lockup.

I see `EditModeConstraintCoinManager::seekConstraintPosition` has not been changed in the meantime.

jhaand, 2025-05-15:
> Side note: Your sketch has 300+ constraints. This is considered bad practice.

Then maybe it's a good benchmark and should be included in a test set. If the performance is measurably slow, we can profile the behaviour and benchmark different solutions.

AgCaliva, 2025-06-24:
How much you offer for solving this one? I have been out of the repo for almost 2 years because i find devs here pretty bad. Why we are still using XML when JSON its better?

closed 2026-01-02 by kadet1090