FreeCAD/FreeCAD #11511: [Problem] Preferences dialog to large and not resizable - dissapears from screen

maxwxyz, 2023-11-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The preferences dialog is too large on my screen, so it is not possible to click "OK" or "CANCEL" nor reach the "Reset..." group.
The dialog cannot be resized, when I try to scale it vertically, the dialog moves quickly out of the screen (into the bottom) and is unable to reach again. I can only close it with ESC.

Quick Demo:

https://github.com/FreeCAD/FreeCAD/assets/6246609/0d61ff61-bbc0-4821-b548-e3745183523d

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35176 (Git)
Build type: Release
Branch: main
Hash: dcd533af4bf0cb3dbc50c7d0e2fb579cad168819
Python 3.10.13, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.24.8
  * CurvedShapes 1.0.5
  * Curves 0.6.14
  * Defeaturing 1.2.1
  * fasteners 0.4.70
  * FEMbyGEN 2.0.0
  * freecad.gears 1.0.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.10.30
  * sheetmetal 0.3.12
  * woodworking 0.21.30486
```
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-11-24:
Duplicate #11004.

kadet1090, 2023-11-25:
This should be reopenm that are two different issues.

#11004 is about one dialog that is way to large. But this issue is about ability to resize the window and scrolling which was removed in #11018, we should allow that because addons can provide setting pages that can have arbitrary height. 

@Roy-043 Can you reopen that and assign to me?

PR #11540 (2023-11-27): Gui: Add ScrollArea to Preferences and auto expand groups
This adds QScrollArea widget to DlgPreferences which in turn allows preference pages to be scrollable. Stacked widget sizes are forced to selected widget so scroll area should appear when it is needed.

Second commit ensures that after selecting item the group is automatically expanded. User can force expansion by explcitly clicking expand button, groups expanded that way will stay expanded unless user collapses them. Otherwise non-active group will be collapsed automatically.

This PR also changes default size of the dialog to 1000x900 [... cut]
commit b147fe00 (2023-11-27): Gui: Add ScrollArea to Preferences

This adds QScrollArea widget to DlgPreferences which in turn allows
preference pages to be scrollable. Stacked widget sizes are forced to
selected widget so scroll area should appear when it is needed.

Fixes #11511
closed 2023-11-27 by chennes