FreeCAD/FreeCAD #11514: GEARS  cannot create a rack

G3PVW, 2023-11-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Development version bug.  
File "New" then select  "Gears" workbench.  Creating a rack, involute or cycloid, fails with :-   
<class 'TypeError'>: cannot pickle 'Part.Wire' object
14:34:57  InvoluteRack: cannot pickle 'Part.Wire' object.  
All other gear types work as expected.

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35176 (Git) AppImage
Build type: Release
Branch: main
Hash: dcd533af4bf0cb3dbc50c7d0e2fb579cad168819
Python 3.10.13, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Assembly4 0.50.4
  * Cfd
  * QuickMeasure 2022.10.28
  * fasteners 0.4.70
  * CfdOF.backup1691329306.5560899 1.24.2 (Disabled)
  * CfdOF.backup1689257654.2153373 1.23.1 (Disabled)
  * freecad.gears 1.0.0
  * Plot 2022.4.17
  * Help 1.0.3
  * ExtremeProLight 1.6.0
  * CfdOF 1.24.8
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-11-24:
Confirmed.

There are 2 issues:

Core:
`copy.deepcopy(wire)` fails in V0.22.

Gear:
Using copy.deepcopy() may not be the most logical solution.
Replacing:
`tooth = copy.deepcopy(tooth)`
with:
`tooth = Wire(tooth)`
seems more appropriate.

@looooo: FYI. Note also that there is a versioning issue. The Addon Manager does not download the latest Gear GitHub files.

Syres916, 2023-11-24:
I've just tested on 0.21.2 and it fails at the same place so 0.20.x is the latest version to work for me.

Syres916, 2023-11-25:
I'm going to _attempt_ to find the commit but that doesn't mean the core code should be changed to fix an addon, it might help @looooo achieve a solution. It's going to be hours as there's something in the region of 4000 commits.

Syres916, 2023-11-25:
@looooo It's all coming back now, I had a minor testing role in https://github.com/FreeCAD/FreeCAD/commit/fbe2fef66cfcee5685d435db57cf34a364315ae9 Basically Werner had to change all the `__getstate__` and `__setstate__` due to a change in Python 3.11 that created a null `__setstate__` for any new PropertyPythonObject which meant no data was collected.

So as you can see all the core Python files have had to change to `dumps` and `loads`. This was easier to handle as all the code changed in one commit so it was all in sync, I'll do some testing on both 0.21.2 and 0.20.2 and if I can get a solution I send you a PR.

Syres916, 2023-11-25:
Hopefully the simple fix above will be acceptable for all.

luzpaz, 2023-11-26:
@Syres916 can we close this ticket as the fix is downstream, correct ?

Syres916, 2023-11-27:
I think it can be closed as I see no reason for add-on devs to use `deepcopy` in the future. The commit in question was far more important to have Python 3.11 compatibility and so won't be backed out IMHO.

closed 2023-11-27 by luzpaz

CarloSartori, 2024-02-20:
> Confirmed.
> 
> There are 2 issues:
> 
> Part: `copy.deepcopy(wire)` fails in V0.22.
> 
> Gear: Using `copy.deepcopy()` may not be the most logical solution. Replacing: `tooth = copy.deepcopy(tooth)` with: `tooth = Wire(tooth)` seems more appropriate and fixes the Gear issue.
> 
> @looooo: FYI. Note also that there is a versioning issue. The Addon Manager does not download the latest Gear GitHub files.

This solutions doesn't work for me (or I faild to implement it)
I change the file features.py according to the suggested solution and here is my output:
`17:19:23  Traceback (most recent call last):
  File "C:\Users\Admin\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\features.py", line 106, in execute
    gear_shape = self.generate_gear_shape(fp)
  File "C:\Users\Admin\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\features.py", line 613, in generate_gear_shape
    tooth = Wire(tooth)
  File "C:\Program Files\FreeCAD 0.21\bin\lib\copy.py", line 161, in deepcopy
    rv = reductor(4)
<class 'TypeError'>: cannot pickle 'Part.Wire' object
17:19:23  Traceback (most recent call last):
  File "C:\Users\Admin\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\features.py", line 106, in execute
    gear_shape = self.generate_gear_shape(fp)
  File "C:\Users\Admin\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\features.py", line 613, in generate_gear_shape
    tooth = Wire(tooth)
  File "C:\Program Files\FreeCAD 0.21\bin\lib\copy.py", line 161, in deepcopy
    rv = reductor(4)
<class 'TypeError'>: cannot pickle 'Part.Wire' object
17:19:23  Traceback (most recent call last):
  File "C:\Users\Admin\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\features.py", line 106, in execute
    gear_shape = self.generate_gear_shape(fp)
  File "C:\Users\Admin\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\features.py", line 613, in generate_gear_shape
    tooth = Wire(tooth)
  File "C:\Program Files\FreeCAD 0.21\bin\lib\copy.py", line 161, in deepcopy
    rv = reductor(4)
<class 'TypeError'>: cannot pickle 'Part.Wire' object
17:19:23  Traceback (most recent call last):
  File "C:\Users\Admin\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\features.py", line 106, in execute
    gear_shape = self.generate_gear_shape(fp)
  File "C:\Users\Admin\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\features.py", line 613, in generate_gear_shape
    tooth = Wire(tooth)
  File "C:\Program Files\FreeCAD 0.21\bin\lib\copy.py", line 161, in deepcopy
    rv = reductor(4)
<class 'TypeError'>: cannot pickle 'Part.Wire' object
17:19:23  Recompute failed!`
I'm running Freecad version 0.21.2 33771(Git) on windows and gears versione 1.0.0

Roy-043, 2024-02-21:
There is no need to edit the code yourself. You need to update the Gears WB.
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods: 
  * freecad.gears 1.2.0
```

CarloSartori, 2024-02-21:
>   * freecad.gears 1.2.0
> ```

Thank you for your answer, but I can not update freegears through the addon manager. 
How can i do it manually in windows?
thank you
Carlo

Roy-043, 2024-02-21:
Try uninstalling and then installing again. That usually works for me when I have an update issue with an addon.