FreeCAD/FreeCAD #11516: [Sketcher] It is not possible to drag a group of geometries

PaddleStroke, 2023-11-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In sketcher you can drag a single curve. But you cannot drag a group of geometries.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2023-11-24:
This is very slightly disingenuous, with the minimal of constraints, the user can box select large quantities of geometry and drag them around. For a workbench that I use nearly everyday, you appear to be _really_ picky about the user interface. I really hope my muscle memory isn't going to be severely tested with all these UI changes.

PaddleStroke, 2023-11-24:
> you appear to be _really_ picky about the user interface.

Yes of course I am the very picky one. No one else ever complained about freecad UI.

How is this feature request going to hurt your muscle memory I wonder ?

Yes constrained geos will move together. But its not a reason not to allow dragging a group of geos. For example if we get text in sketcher, then it won't be constrained. Or if you import a logo as a DXF for example. Then its not possible to drag it.

luzpaz, 2023-11-26:
Duplicate of #11205

PaddleStroke, 2023-11-26:
sorry @luzpaz !

closed 2023-11-26 by PaddleStroke