FreeCAD/FreeCAD #11517: FreeCAD turning extremely laggy

Herrcookiekiller, 2023-11-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When turning on transparency in the appearance menu for any object, and then selecting any face (this is _only_ working with faces) and or keeping the face highlighted, then moving the 3D Viewport the FPS drop down to 4.
However when i _do not_ have any face of the transparent object selected and move the 3D Viewport everything is completely fine.
Having vertices or edges selected on the transparent object does not cause any issues either. _See video_

https://github.com/FreeCAD/FreeCAD/assets/40491077/0ffa4b3d-dc5c-4b9a-90dd-69a229c1219e

I have tried several of the graphics options already, disabled the double render pass for transparent objects and so on.
Different render styles (except for wireframe - because there are no faces) dont fix anything either.

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3D_Printing_Tools
  * A2plus 0.4.60n
  * Assembly4 0.50.2
  * Autoload
  * CubeMenu
  * Curves 0.6.12
  * Defeaturing 1.2.0
  * dxf-library
  * fasteners 0.4.56
  * fcgear 1.0.0
  * freecad.gears 1.0.0
  * InventorLoader 1.4.0
  * kicadStepUpMod 10.19.4
  * Lithophane
  * Manipulator 1.5.0
  * parts_library
  * POV-Ray-Rendering
  * Render 2023.2.6
  * sheetmetal 0.2.63
  * ToolbarStyle
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2023-11-27:
Can't reproduce it.

luzpaz, 2023-11-27:
@Herrcookiekiller yikes, just noticed you are using v0.20!  
Please update to latest stable v0.21.2 and try to reproduce.

Herrcookiekiller, 2023-11-27:
> @Herrcookiekiller can you provide an example file?
> 
> @wwmayer any idea ?

Here you go

[Piniongehäuse1.zip](https://github.com/FreeCAD/FreeCAD/files/13474405/Piniongehause1.zip)

wwmayer, 2023-11-27:
> Are you on Linux?

Yes. It's an Xubuntu 22.04.

luzpaz, 2023-11-27:
Cannot reproduce on 
```
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35209 (Git) AppImage
Build type: Release
Branch: main
Hash: 1c0456654880e58e5d9e839f27df177c68980185
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
```

Rexbas, 2023-11-27:
I can reproduce this with the flatpak 0.21.2 and 0.22.0.dev. When I start a drag while hovering above the non-transparent part I get 75 fps. When I start a drag while hovering above the transparent part, the fps goes down to about 15.

Rexbas, 2023-11-27:
I changed my Anti-Aliasing from MSAA 8x to None, restarted FreeCAD and then the problem disappeared. Now when I turn MSAA 8x back on I can no longer reproduce this....

Rexbas, 2023-11-27:
Oh its back. It probably has nothing to do with Anti-Aliasing. Here is what I did:
1. Download Piniongehause1.zip
2. Extract Piniongehäuse1.FCStd
3. Open Piniongehäuse1.FCStd in FreeCAD (0.22.0dev). Now it is reproducible
4. Open preferences and hit apply without changing anything
5. Close FreeCAD, a popup with an option to save or discard is shown, press discard
6. Open Piniongehäuse1.FCStd again. It is still reproducible
7. Repeat step 4 and 5 but this time press save
8. Open Piniongehäuse1.FCStd again. The problem disappeared

I also noticed a second file was created after saving with the name Piniongehäuse1.FCStd1, the new Piniongehäuse1.FCStd is smaller than the original unzipped Piniongehäuse1.FCStd. To reproduce the problem again I had to delete the files and unzip again.

Herrcookiekiller, 2023-11-28:
> Oh its back. It probably has nothing to do with Anti-Aliasing. Here is what I did:
> 
>     1. Download Piniongehause1.zip
> 
>     2. Extract Piniongehäuse1.FCStd
> 
>     3. Open Piniongehäuse1.FCStd in FreeCAD (0.21.2 or 0.22.0dev, does not matter). Now it is reproducible
> 
>     4. Open preferences and hit apply without changing anything
> 
>     5. Close FreeCAD, a popup with an option to save or discard is shown, press discard
> 
>     6. Open Piniongehäuse1.FCStd again. It is still reproducible
> 
>     7. Repeat step 4 and 5 but this time press save
> 
>     8. Open Piniongehäuse1.FCStd again. The problem disappeared
> 
> 
> I also noticed a second file was created after saving with the name Piniongehäuse1.FCStd1, the new
> Piniongehäuse1.FCStd is smaller than the original unzipped Piniongehäuse1.FCStd. To reproduce the problem again I
> had to delete the files and unzip again.

As far as i know .FCStd1 files are backup files automatically created when saving a project.

Also i followed your steps to reproduce, but mine was still laggy after step 8.

wwmayer, 2023-11-28:
With your test file I can confirm the fps drop, too. If nothing is selected I have 30 fps and after selecting a face it goes down to 8.

Syres916, 2023-11-28:
@Rexbas it wasn't too difficult to find the change in the file:

```
Angular Deflection 2.00 degrees
Deviation 0.4
```

changed to:

```
Angular Deviation 28.5 degrees
Deviation 0.5
```

hence the major difference in performance.

Personally I use:

```
Angular Deflection 10.00 degrees
Deviation 0.25
```

which results in decent responsiveness and good aesthetics on screen.

Rexbas, 2023-11-28:
It is however still strange that the fps only drops when the transparent part is selected.

pfn, 2023-11-30:
Couldn't repro this on linkdaily 1102 on Windows 11 as an additional data point.

Herrcookiekiller, 2023-11-30:
> Yes you are right. If I set the angular deflection to 2.0 degrees and deviation to 0.4 then the steps I described before don´t have any effect. So I guess @Herrcookiekiller should change these values (Preferences->Part/Part Design->Shape View).
> 
> I previously used angular deviation of 28.5 degrees and deviation of 0.5 which I believe are the defaults.

This is correct, if i change these values to the ones you said, my fps only drop to 12 instead of 6.

Herrcookiekiller, 2023-11-30:
> Couldn't repro this on linkdaily 1102 on Windows 11 as an additional data point.
> FreeCAD.Link.Daily.2023-11-30.10-43-18.mp4

You did not select any of the transparent faces on the bottom part.

And the top part turning transparent when you select it seems to be something completely different to me, thus it doesnt cause any issues.

pfn, 2023-11-30:
https://github.com/FreeCAD/FreeCAD/assets/512197/eb08c4c2-21d7-4d0b-b325-0b96acedcd13

Also works OK for me on ls3.

Rexbas, 2023-11-30:
I can confirm this is not reproducible on linkdaily 1102. I can see in the code that something about transparency sorting is different there.

AgCaliva, 2023-12-01:
I dont want to do people divert here, but i think that the fps drop can be something related to the changes made to src/Base/XMLReader.cpp lastly, more specifically in bool Base::XMLReader::readNextElement() implementation. I think it would be nice to test without last changes in reading xml, if  this issue its related to reading data, which im not sure this is the case.
If im wrong im sory to waste your time.

Rexbas, 2023-12-01:
It is a rendering problem and it is not something new. I also reproduced this problem in 0.19.3, probably also happens in older versions.

PR #11611 (2023-12-03): Part: Use materialIndex.setValuesPointer() only if the new values are different
This fixes #11517. See https://github.com/FreeCAD/FreeCAD/issues/11517#issuecomment-1837139082 for the cause of the problem.

Using `materialIndex.setValuesPointer()` only when the values change prevents the continuous cache invalidation and fixes the low fps.

To make this work I also had to remove [these](https://github.com/FreeCAD/FreeCAD/commit/6711e53d1681e86d9ea88783fa9353a8b84c13ce#diff-e4fb66281089c4211e8116c2a8bf4e05736a12e47518ca5786f3481ddb4f714fL646-L648) lines. I am not sure what the intention of these [... cut]