FreeCAD/FreeCAD #11521: Path - Crash after quitting FreeCAD after using ToolBit Library Editor

JohnOCFII, 2023-11-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Trying to follow the new Path tutorial by @concretedog posted in the blog: https://blog.freecad.org and experienced FreeCAD crashes approximately 70% of the time after saving file and quitting FreeCAD. Seems to only happen after spending time in the ToolBits Library.  This on macOS M1/arm.

### Full version info

```shell
OS: macOS 14.1.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35081 (Git)
Build type: Release
Branch: main
Hash: 32db044caa92a2adc448850179c524b793a03c3e
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
```
```

### Subproject(s) affected?

Path

### Anything else?

Crash report -- I capture 3 that were almost identical, other than addresses:

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [23134]
Path:                  /Applications/Other Apps/FreeCAD-35081.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        Exited process [23128]
User ID:               501

Date/Time:             2023-11-24 19:39:48.7930 -0600
OS Version:            macOS 14.1.1 (23B81)
Report Version:        12
Anonymous UUID:        E1A9B3E8-17C0-3635-633F-31D076EC9805

Sleep/Wake UUID:       499488E4-0E4A-4A4E-85C8-3D38854F7199

Time Awake Since Boot: 250000 seconds
Time Since Wake:       40390 seconds

System Integrity Protection: enabled

Crashed Thread:        0  CrBrowserMain  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000f41fd9c969a8 -> 0x0000741fd9c969a8 (possible pointer authentication failure)
Exception Codes:       0x0000000000000001, 0x0000f41fd9c969a8

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [23134]

VM Region Info: 0x741fd9c969a8 is not in any region.  Bytes after previous region: 22126493526441  
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      MALLOC_NANO              600000000000-600020000000 [512.0M] rw-/rwx SM=PRV  
--->  
      UNUSED SPACE AT END

Thread 0 Crashed:: CrBrowserMain Dispatch queue: com.apple.main-thread
0   libFreeCADGui.dylib           	       0x10629cc5c std::__1::__hash_iterator<std::__1::__hash_node<std::__1::__hash_value_type<App::DocumentObject*, std::__1::set<App::DocumentObject*, std::__1::less<App::DocumentObject*>, std::__1::allocator<App::DocumentObject*>>>, void*>*> std::__1::__hash_table<std::__1::__hash_value_type<App::DocumentObject*, std::__1::set<App::DocumentObject*, std::__1::less<App::DocumentObject*>, std::__1::allocator<App::DocumentObject*>>>, std::__1::__unordered_map_hasher<App::DocumentObject*, std::__1::__hash_value_type<App::DocumentObject*, std::__1::set<App::DocumentObject*, std::__1::less<App::DocumentObject*>, std::__1::allocator<App::DocumentObject*>>>, std::__1::hash<App::DocumentObject*>, std::__1::equal_to<App::DocumentObject*>, true>, std::__1::__unordered_map_equal<App::DocumentObject*, std::__1::__hash_value_type<App::DocumentObject*, std::__1::set<App::DocumentObject*, std::__1::less<App::DocumentObject*>, std::__1::allocator<App::DocumentObject*>>>, std::__1::equal_to<App::DocumentObject*>, std::__1::hash<App::DocumentObject*>, true>, std::__1::allocator<std::__1::__hash_value_type<App::DocumentObject*, std::__1::set<App::DocumentObject*, std::__1::less<App::DocumentObject*>, std::__1::allocator<App::DocumentObject*>>>>>::find<App::DocumentObject*>(App::DocumentObject* const&) + 180
1   libFreeCADGui.dylib           	       0x1067b4768 Gui::WrapperManager::destroyed(QObject*) + 64
2   libQt5Core.5.15.8.dylib       	       0x1088e683c 0x1087a0000 + 1337404
3   libFreeCADGui.dylib           	       0x10628bc94 Py::Object::~Object() + 40
4   libFreeCADGui.dylib           	       0x1067b442c Gui::WrapperManager::clear() + 212
5   libQt5Core.5.15.8.dylib       	       0x1088e683c 0x1087a0000 + 1337404
6   freecad                       	       0x10495f030 main + 4932
7   dyld                          	       0x1827990e0 start + 2360
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-11-26:
This tutorial was just published on the FreeCAD blog. Other users are probably encountering this in that case.

xtemp09, 2023-12-13:
@wwmayer, I could reproduce the crash on Ubuntu, so I dug into it. This issue was tagged with _WB Path_ label; however, it's merely reproducible with _Path_.

After having banged my head against the wall for the first days, I found the place where the segmentation fault occurs and a few ways to fix it. The crash was caused by concurrent access to the _wrappers_ container.

I reimplemented the code of _WrapperManager_ in a simpler, Qt way, so now it does not need any container at all. A wise guy has to rewrite _WrapperManager_ or, at least, find time to dive into this issue. @wwmayer, if you have time, you should deal with it. If you don't, perhaps, at the moment, could you take a look at my commit xtemp09/FreeCAD@d5de03b67726e38015e47ce8ff2f6f2c63e59940 and test it? Since the patch fixes the segmentation fault, maybe, the commit should be merged until later.

I also have made TSAN reports. Since the segmentation fault takes place, the _pre_-commit report does not contain valuable information except for pointing out at the concurrent access to the container, while the _post_-commit report cannot possibly contain anything interesting on this topic.

JohnOCFII, 2023-12-19:
I wasn't sure if the comments above implied this was fixed -- but the crash still occurs.

```
OS: macOS 14.2
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35414 (Git)
Build type: Release
Branch: main
Hash: 759930e3aab711355d2f0895f641a8657897af48
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
```

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [81364]
Path:                  /Applications/Other Apps/FreeCAD-35414.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        Exited process [81356]
User ID:               501

Date/Time:             2023-12-18 19:50:13.1023 -0600
OS Version:            macOS 14.2 (23C64)
Report Version:        12
Anonymous UUID:        E1A9B3E8-17C0-3635-633F-31D076EC9805

Sleep/Wake UUID:       0285CE66-F874-4E53-BD0C-ECC294282D80

Time Awake Since Boot: 440000 seconds
Time Since Wake:       7238 seconds

System Integrity Protection: enabled

Crashed Thread:        0  CrBrowserMain  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x00003bd7f3f23de8
Exception Codes:       0x0000000000000001, 0x00003bd7f3f23de8

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [81364]

VM Region Info: 0x3bd7f3f23de8 is not in any region.  Bytes after previous region: 65317660409321  Bytes before following region: 39754419520024
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      commpage (reserved)        1000000000-7000000000   [384.0G] ---/--- SM=NUL  ...(unallocated)
--->  GAP OF 0x5f9000000000 BYTES
      MALLOC_NANO              600000000000-600020000000 [512.0M] rw-/rwx SM=PRV  

Kernel Triage:
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter

Thread 0 Crashed:: CrBrowserMain Dispatch queue: com.apple.main-thread
0   libFreeCADGui.dylib           	       0x101c90cd0 std::__1::__hash_iterator<std::__1::__hash_node<std::__1::__hash_value_type<App::DocumentObject*, std::__1::set<App::DocumentObject*, std::__1::less<App::DocumentObject*>, std::__1::allocator<App::DocumentObject*>>>, void*>*> std::__1::__hash_table<std::__1::__hash_value_type<App::DocumentObject*, std::__1::set<App::DocumentObject*, std::__1::less<App::DocumentObject*>, std::__1::allocator<App::DocumentObject*>>>, std::__1::__unordered_map_hasher<App::DocumentObject*, std::__1::__hash_value_type<App::DocumentObject*, std::__1::set<App::DocumentObject*, std::__1::less<App::DocumentObject*>, std::__1::allocator<App::DocumentObject*>>>, std::__1::hash<App::DocumentObject*>, std::__1::equal_to<App::DocumentObject*>, true>, std::__1::__unordered_map_equal<App::DocumentObject*, std::__1::__hash_value_type<App::DocumentObject*, std::__1::set<App::DocumentObject*, std::__1::less<App::DocumentObject*>, std::__1::allocator<App::DocumentObject*>>>, std::__1::equal_to<App::DocumentObject*>, std::__1::hash<App::DocumentObject*>, true>, std::__1::allocator<std::__1::__hash_value_type<App::DocumentObject*, std::__1::set<App::DocumentObject*, std::__1::less<App::DocumentObject*>, std::__1::allocator<App::DocumentObject*>>>>>::find<App::DocumentObject*>(App::DocumentObject* const&) + 180
1   libFreeCADGui.dylib           	       0x1021abf64 Gui::WrapperManager::destroyed(QObject*) + 64
2   libQt5Core.5.15.8.dylib       	       0x1042e2800 0x10419c000 + 1337344
3   libFreeCADGui.dylib           	       0x101c7fcdc Py::Object::~Object() + 40
4   libFreeCADGui.dylib           	       0x1021abc28 Gui::WrapperManager::clear() + 212
5   libQt5Core.5.15.8.dylib       	       0x1042e2800 0x10419c000 + 1337344
6   freecad                       	       0x10034b030 main + 4932
7   dyld                          	       0x1878910e0 start + 2360
```

commit 51ccf610 (2024-01-22): Fix occasional segmentation fault when exiting

Fixes #11521.
closed 2024-01-22 by chennes