FreeCAD/FreeCAD #11522: Path: ToolBits Tool names as O/S file names seems to cause issues with unexpected characters

JohnOCFII, 2023-11-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It appears that adding tools to the ToolBits library actually creates O/S files.  If the tool name is not a valid O/S file name (which has different restrictions on different O/Ss) it appear a silent substitution takes place.  This seems to cause issues when a tool is deleted, but the O/S doesn't see it as deleted.

Example: Attempted to create a tool named `1/8" 2FL flat Endmill`. Name isn't rejected, and a file is created at the macOS level with the fact extension.

The Tool List shows that tool as `1:8" 2FL Flat Endmill`

If I delete that tool from the Library, the actual file (with the unmodified file name) isn't actually deleted. If I attempted to re-create that tool with the same name (as it is gone from the library) I can't recreate the file as the OS has a file with that name. (2nd screen shot shows an entry with 7FL instead of 2FL as I couldn't re-enter the exact name due to the above issue.

It seems risky to have the user-entered tool name be the actual underlying OS file name.

[image]

[image]

[Cylinder4Path.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/13462897/Cylinder4Path.FCStd.zip)

### Full version info

```shell
OS: macOS 14.1.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35081 (Git)
Build type: Release
Branch: main
Hash: 32db044caa92a2adc448850179c524b793a03c3e
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
```
```

### Subproject(s) affected?

None

### Anything else?

This information was in the Report View window:

```
20:35:53  /Users/osh/Library/Application Support/FreeCAD/Macro/Bit/1:8" 7FL Flat Endmill.fctb
20:36:31  Traceback (most recent call last):
20:36:31    File "/Applications/Other Apps/FreeCAD-35081.app/Contents/Resources/Mod/Path/Path/Tool/Gui/BitEdit.py", line 63, in setModelData
20:36:31      editor.setModelData(widget)
20:36:31    File "/Applications/Other Apps/FreeCAD-35081.app/Contents/Resources/Mod/Path/Path/Base/Gui/PropertyEditor.py", line 212, in setModelData
20:36:31      self.setProperty(widget.currentText())
20:36:31    File "/Applications/Other Apps/FreeCAD-35081.app/Contents/Resources/Mod/Path/Path/Base/Gui/PropertyEditor.py", line 69, in setProperty
20:36:31      setattr(self.obj, self.prop, value)
20:36:31  ReferenceError: Cannot access attribute 'Material' of deleted object
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-11-26:
@LarryWoestman any chance you can test this ?

LarryWoestman, 2023-11-26:
I will test it on Windows 11 and Ubuntu 22.04, but I don't have access to any Apple products.

LarryWoestman, 2023-11-29:
On Ubuntu 22.04.03 I saw similar results.

When I tried to create a toolbit named '1/8" 2FL Flat Endmill' a file named ' 2FL Flat Endmill' of 0 size and another file named ' 2FL Flat Endmill.fctb' were created.  The toolbit name in the GUI was also ' 2FL Flat Endmill'.

When I tried to remove that toolbit, it vanished from the GUI but both files were still present in the file system.

When I tried to re-create the same toobit name again, it claimed that a file named ' 2FL Flat Endmill.fctb' already existed (which is correct) and wanted to know if I wanted to open it.

So a toolbit file name that doesn't match the requested toolbit name was created and was not deleted when requested.

```
OS: Ubuntu 22.04.3 LTS
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35039 +206 (Git)
Build type: Unknown
Branch: staging
Hash: 9176fba7872558f1d847da9bd52f60dc1a53069a
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: C/Default (C)
```

LarryWoestman, 2023-11-29:
The build that I used is identical to the main branch from a couple of days ago with changes to the linuxcnc_post.py and TestLinuxCNCPost.py files.

LarryWoestman, 2023-11-29:
On Windows 11 I saw somewhat different results.

When I tried to create a toolbit named '1/8" 2FL Flat Endmill' the GUI claimed that it was an invalid file name and wouldn't let me continue until I removed both the '/' and the '"'.  It then created a file named '1_8_ 2FL Flat Endmill.fctb'.  So it appears that something is checking file names for validity somewhere.

When I tried to remove that toolbit, it vanished from the GUI but the file was still present in the file system.

When I tried to re-create the same toolbit name again, it claimed that a file named '1_8_ 2FL Flat Endmill.fctb' already existed (which is correct) and wanted to know if I wanted to overwrite it.

So on Windows 11 the attempted file name was identified as invalid (but with no information about what was invalid) and a valid file name was required for the toolbit before continuing.

```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35240 (Git)
Build type: Release
Branch: main
Hash: 27b58fd410e62d0bcb596de956b80efcc0d0bec5
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
```

LarryWoestman, 2023-11-30:
It appears that the code (in Path/Path/Tool/Gui/Bit.py) is using QtGui.QFileDialog.getSaveFileName to ask for a file name for a new toolbit.

It then attempts to open the file name returned for writing as the only verification that the file name is a valid file name.  If the open fails then the "invalid filename" dialog is issued.  If the open works then the file name is assumed to be valid.

Something in the QtGui.OFileDialog.getSaveFileName may be rewriting the submitted file name in some manner.  I will attempt to find that code.

LarryWoestman, 2023-11-30:
Well, that led down a deep hole :-)

The QFileDialog.getSaveFileName function uses the native file dialog on Windows and MacOS and uses a QFileDialog on linux, at least by default.

I got lost at that point and can't figure out what code, where, is possibly doing things to the file name.

JohnOCFII, 2023-12-25:
> Well, that led down a deep hole :-)
> 
> The QFileDialog.getSaveFileName function uses the native file dialog on Windows and MacOS and uses a QFileDialog on linux, at least by default.
> 
> I got lost at that point and can't figure out what code, where, is possibly doing things to the file name.

Thanks for checking!

JohnOCFII, 2025-01-29:
The issue still exists.

PR #21425 (2025-06-02): CAM: Integration of the tool and shape editors from BTL (Better Tool Library)
This very large PR essentially replaces CAM Tool Management completely, as part of the [Better Tool Library](https://github.com/knipknap/better-tool-library/) integration.

## Relevant Issues

Discussion: #9466

I also went through all open CAM issues to find any that may be resolved or otherwise affected:

- Resolves: #8201
- Resolves: #8526
- Resolves: #9625
- Resolves: #11522
- Resolves: #12712
- Likely resolves: #12823, but should be tested to be sure
- Possibly resolves #13603, to be tested
- Resolves: #17083
-  [... cut]
closed 2025-06-02 by sliptonic