FreeCAD/FreeCAD #11523: [Problem]  Fails to build with mingw-w64-gcc

MehdiChinoune, 2023-11-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Fails to build with GNU toolchain (gcc + binutils +libstdc++)
```
  [1620/4686] Linking CXX shared library bin\libFreeCADApp.dll
  FAILED: bin/libFreeCADApp.dll src/App/libFreeCADApp.dll.a 
  cmd.exe /C "cd . && D:\M\msys64\ucrt64\bin\g++.exe -Wall -Wextra -Wno-write-strings -march=nocona -msahf -mtune=generic -O2 -pipe -Wno-deprecated-declarations -Wno-unknown-pragmas -Wno-attributes -O3 -DNDEBUG  -pipe -shared -o bin\libFreeCADApp.dll -Wl,--out-implib,src\App\libFreeCADApp.dll.a -Wl,--major-image-version,0,--minor-image-version,0 src/App/CMakeFiles/FreeCADApp.dir/FreeCADApp_autogen/mocs_compilation.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/Annotation.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/Document.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/DocumentObject.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/Extension.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/ExtensionPyImp.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/DocumentObjectExtension.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/DocumentObjectExtensionPyImp.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/ExtensionContainer.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/ExtensionContainerPyImp.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/Graphviz.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/GroupExtension.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/GroupExtensionPyImp.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/DocumentObjectFileIncluded.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/DocumentObjectGroup.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/DocumentObjectGroupPyImp.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/GeoFeaturePyImp.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/DocumentObjectPyImp.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/DocumentObserver.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/DocumentObserverPython.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/DocumentPyImp.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/Expression.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/ExpressionTokenizer.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/FeaturePython.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/FeatureTest.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/GeoFeature.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/GeoFeatureGroupExtensionPyImp.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/GeoFeatureGroupExtension.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/ImagePlane.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/OriginGroupExtensionPyImp.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/OriginGroupExtension.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/PartPyImp.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/Part.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/Origin.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/Path.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/InventorObject.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/MeasureDistance.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/Placement.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/OriginFeature.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/Range.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/Transactions.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/TransactionalObject.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/VRMLObject.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/MaterialObject.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/MergeDocuments.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/TextDocument.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/Link.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/LinkBaseExtensionPyImp.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/DynamicProperty.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/ObjectIdentifier.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/Property.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/PropertyContainer.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/PropertyContainerPyImp.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/PropertyFile.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/PropertyGeo.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/PropertyLinks.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/PropertyPythonObject.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/PropertyStandard.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/PropertyUnits.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/PropertyExpressionEngine.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/Application.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/ApplicationPy.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/AutoTransaction.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/Branding.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/Color.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/ColorModel.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/ComplexGeoData.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/ComplexGeoDataPyImp.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/ElementMap.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/Enumeration.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/IndexedName.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/MappedElement.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/MappedName.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/Material.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/MaterialPyImp.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/Metadata.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/MetadataPyImp.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/ElementNamingUtils.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/StringHasher.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/StringHasherPyImp.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/StringIDPyImp.cpp.obj src/App/CMakeFiles/FreeCADApp.dir/PreCompiled.cpp.obj  -lgdi32  src/Base/libFreeCADBase.dll.a  D:/M/msys64/ucrt64/lib/libboost_filesystem-mt.dll.a  D:/M/msys64/ucrt64/lib/libboost_program_options-mt.dll.a  D:/M/msys64/ucrt64/lib/libboost_regex-mt.dll.a  D:/M/msys64/ucrt64/lib/libboost_system-mt.dll.a  D:/M/msys64/ucrt64/lib/libboost_thread-mt.dll.a  D:/M/msys64/ucrt64/lib/libboost_date_time-mt.dll.a  D:/M/msys64/ucrt64/lib/libfmt.dll.a  D:/M/msys64/ucrt64/lib/libQt5Xml.dll.a  -lgdi32  D:/M/msys64/ucrt64/lib/libboost_atomic-mt.dll.a  D:/M/msys64/ucrt64/lib/libboost_chrono-mt.dll.a  D:/M/msys64/ucrt64/lib/libxerces-c.dll.a  D:/M/msys64/ucrt64/lib/libz.dll.a  -lRpcrt4  D:/M/msys64/ucrt64/lib/libpython3.11.dll.a  D:/M/msys64/ucrt64/lib/libQt5Core.dll.a  -lkernel32 -luser32 -lgdi32 -lwinspool -lshell32 -lole32 -loleaut32 -luuid -lcomdlg32 -ladvapi32 && cd ."
  D:/M/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: src/App/CMakeFiles/FreeCADApp.dir/Link.cpp.obj:Link.cpp:(.rdata+0x1ce0): undefined reference to `Base::Observer<char const*>::OnDestroy(Base::Subject<char const*>&)'
  
  D:/M/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: src/App/CMakeFiles/FreeCADApp.dir/Link.cpp.obj:Link.cpp:(.rdata+0x1ce8): undefined reference to `Base::Observer<char const*>::Name()'
  
  collect2.exe: error: ld returned 1 exit status
```
There was an issue with salomesmesh which I fixed by a simple patch not accepted by LLVM toolchain (llvm + clang + lld + libc++)

### Full version info

```shell
0.21.2
```

### Subproject(s) affected?

None

### Anything else?

It builds with mingw-w64 LLVM toolchain.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wsteffe, 2023-11-25:
I have found the same error trying to build on MSYS2 Mingw64 the RT LinkMerge branch:

> undefined reference to `Base::Observer<char const*>::OnDestroy(...

But the first error (in a long sequence) was:

> C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: cannot export
> _ZTVN3App8DocumentE: symbol not defined
> C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/Fre
> eCADApp.dir/objects.a(Document.cpp.obj):Document.cpp:(.text+0x4cd9): **undefined reference to `vtable for App::Document'**
> 

As already reported at  https://github.com/msys2/MINGW-pakages/issues/19175#issuecomment-1826304666. Nit sure but I suspect that the main problem is related with missing implementation of some virtual function. I got this idea reading at https://stackoverflow.com/questions/7720205/linking-error-undefined-reference-to-vtable-for-xxx

luzpaz, 2023-11-26:
@wsteffe github URL you mentioned above is broken

MehdiChinoune, 2023-11-26:
> @wsteffe github URL you mentioned above is broken

https://github.com/msys2/MINGW-packages/issues/19175#issuecomment-1826304666

wsteffe, 2023-11-26:
Corrected. Thanks [MehdiChinoune](https://github.com/MehdiChinoune).

wsteffe, 2023-11-26:
I have just looked for a definition of App::Document. 
In LinkMerge it is declared at line 47 of src/App/Application.h but I can not find anywhere a definition of this class.

wsteffe, 2023-11-27:
Apologize. App::Document is defined at line 57 of src/App/Document.h

> class AppExport Document : public App::PropertyContainer
> { ....
> 

Perhaps the problem is a wrong definition of AppExport. As said by @wmayer said at https://forum.freecad.org/viewtopic.php?style=4&t=750 AppExport is required on windows to export symbols as "__declspec(dllimport)"

luzpaz, 2023-11-27:
it's @wwmayer not wmayer (that's someone else :blush:)

wsteffe, 2023-11-27:
Hi all,
 I made a small progress by adding following lines in main CmakeList.txt :

> if(MINGW)
>  add_definitions(-DWNT)
>  add_definitions(-DOS_MINGW)
> endif()

Lines was added just after the line
`list(APPEND CMAKE_MODULE_PATH "${CMAKE_CURRENT_SOURCE_DIR}/cMake"`
but yoy may choose a different position.

I have also changed  src/FCGlobal.h

```
 #if defined(WIN64) || defined(_WIN64) || defined(__WIN64__) || 
 defined(__CYGWIN__)
 #  define FREECAD_DECL_EXPORT __declspec(dllexport)
 #  define FREECAD_DECL_IMPORT __declspec(dllimport)
 ...
 #endif

```
becomes:

```
#if (defined(WIN64) || defined(_WIN64) || defined(__WIN64__) || defined(__CYGWIN__)) && !defined(OS_MINGW)
#  define FREECAD_DECL_EXPORT __declspec(dllexport)
#  define FREECAD_DECL_IMPORT __declspec(dllimport)
 ...
#endif
```

 

Now it creates: 

-    libFreeCADApp.dll.a  inside build/src/App
-    libFreeCADBase.dll.a inside build/src/Base

I would have them placed in build/lib as it happens with libSMDS.dll.a but I don't know how to make it happen.

The next problem comes in CXX with a lot of errors like:

> C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/FreeCADApp.dir/objects.a(MetadataPyImp.cpp.obj):MetadataPyImp.:(.data+0x98): undefined reference to `Base::PyObjectBase::__s
> etattro(_object*, _object*, _object*)'
> C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/FreeCADApp.dir/objects.a(MetadataPyImp.cpp.obj):MetadataPyImp.:(.data+0x100): undefined reference to `Base::PyObjectBase::Ty
> pe'
> collect2.exe: error: ld returned 1 exit status

wsteffe, 2023-11-27:
No sorry the problem is still in src/App

wwmayer, 2023-11-27:
Try using the clang compiler instead of gcc. This way I am able to build FreeCAD.

See also: https://github.com/msys2/MINGW-packages/issues/19175#issuecomment-1827914275

wsteffe, 2023-11-27:
Do not like very much but If I remove shared from 

> add_library(FreeCADApp SHARED ${FreeCADApp_SRCS})

it creates a static library without errors:

> 
> [  7%] Building CXX object src/App/CMakeFiles/FreeCADApp.dir/Branding.cpp.obj
> [  7%] Building CXX object src/App/CMakeFiles/FreeCADApp.dir/Color.cpp.obj
> [  7%] Building CXX object src/App/CMakeFiles/FreeCADApp.dir/ColorModel.cpp.obj
> [  7%] Building CXX object src/App/CMakeFiles/FreeCADApp.dir/MappedElement.cpp.obj
> [  7%] Building CXX object src/App/CMakeFiles/FreeCADApp.dir/ComplexGeoData.cpp.obj
> [  7%] Building CXX object src/App/CMakeFiles/FreeCADApp.dir/ComplexGeoDataPyImp.cpp.obj
> [  7%] Building CXX object src/App/CMakeFiles/FreeCADApp.dir/Enumeration.cpp.obj
> [  7%] Building CXX object src/App/CMakeFiles/FreeCADApp.dir/Material.cpp.obj
> [  8%] Building CXX object src/App/CMakeFiles/FreeCADApp.dir/MaterialPyImp.cpp.obj
> [  8%] Building CXX object src/App/CMakeFiles/FreeCADApp.dir/Metadata.cpp.obj
> [  8%] Building CXX object src/App/CMakeFiles/FreeCADApp.dir/MetadataPyImp.cpp.obj
> [  8%] Building CXX object src/App/CMakeFiles/FreeCADApp.dir/PreCompiled.cpp.obj
> [  8%] Linking CXX static library libFreeCADApp.a
> [  8%] **Built target FreeCADApp**
>

wwmayer, 2023-11-27:
> it creates a static library without errors:

Of course, but you get linking errors as soon as you try to build the executable. A static library basically is just an archive containing all object files.

wwmayer, 2023-11-27:
I gave the mingw-gcc another try today. When running the CMake GUI I realized that the field for CMAKE_BUILD_TYPE was empty which I guess then tried to create a debug version and that's possibly the reason why linking FreeCADApp was hanging for me.

After setting its value to `Release` the linking of FreeCADApp failed immediately telling me that for Base::Observer two symbols were missing. After trying out a few things I figured out that this change 7dc14634cd is the culprit. After I have tmp. disabled the new code I was able to build FreeCADApp.

After making a few more changes in the SMESH code and the Import module I was able to build the entire application including the FEM module.

What's weird is that with mingw-clang I still get linking errors with vtk but with the mingw-gcc build I don't.

An annoying thing with the gcc build is that https://github.com/FreeCAD/FreeCAD/blob/main/src/App/StringHasher.h#L791 (and a few more functions) generate a warning each time the header is included (i.e. a few thousand times):
> 'bool App::StringID::isBinary() const' redeclared without dllimport attribute after being referenced with dll linkage 

The mingw-clang build doesn't show this warning and I wonder how this could be fixed.

wsteffe, 2023-11-28:
I also have solved, but in a different way, the linkage problem of libFreeCADBase.dll.a.

To me the problem was that with cmake+mingw64 it is not possible to link one shared library against an import library previously created in the same project.
But it is instead possible to link an object target agains the shared library and then make a shared library from the object target. So inside the src/App/CmakeList.txt now I have:

> if(MINGW)
>  add_library(FreeCADAppObject OBJECT ${FreeCADApp_SRCS})
>  target_link_libraries(FreeCADAppObject ${FreeCADApp_LIBS})
>  target_link_options(FreeCADAppObject PUBLIC "-Wl,--enable-auto-import -Wl,--export-all-symbols")
>  add_library(FreeCADApp SHARED $<TARGET_OBJECTS:FreeCADAppObject>)
> else(MINGW)
>  add_library(FreeCADApp SHARED ${FreeCADApp_SRCS})
>  target_link_libraries(FreeCADApp ${FreeCADApp_LIBS})
> endif(MINGW)
> 

I am not sure that the link option added to the target FreeCADAppObject are really necessary but in this way it works. 

Remeber that I have also removed the __declspec(dllexport) and __declspec(dllimport) for MINGW build. I think that it is not required. I have another project usimg OpenCadcade and ather similarity built with MinGW without this declaration and it works well on windows.  

But now I am facing a problem with a missing QDomDocument header (see below).
What have I to install in MSYS2/Mingw64 to have it ?

```
  9%] Building CXX object src/Gui/CMakeFiles/FreeCADGuiObject.dir/CommandTest.cpp.obj
[  9%] Building CXX object src/Gui/CMakeFiles/FreeCADGuiObject.dir/CommandView.cpp.obj
C:/msys64/home/walter/Software/FreeCad/FreeCAD/src/Gui/CommandView.cpp:35:11: fatal error: QDomDocument: No such file or directory
   35 | # include <QDomDocument>
      |           ^~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [src/Gui/CMakeFiles/FreeCADGuiObject.dir/build.make:1638: src/Gui/CMakeFiles/FreeCADGuiObject.dir/CommandView.cpp.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:5158: src/Gui/CMakeFiles/FreeCADGuiObject.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

```

wwmayer, 2023-11-28:
https://packages.msys2.org/package/mingw-w64-x86_64-qt5-base

wsteffe, 2023-11-28:
Hi wwmayer  mingw-w64-x86_64-qt5-base was already indstalled.

Anyway I would also add that, while now src/App/CmakeList.txt seems to work for me I think there is still a problem because the first time I executed make there was a long sequence of undefined references. I have executed make again to see the errors but the second time no error was reported (in the App dir).

Now I am thinking that it was because in the first run 
 `add_library(FreeCADApp SHARED $<TARGET_OBJECTS:FreeCADAppObject>)`
was using  older object files without waiting that target FreeCADAppObject is completed. Probably it is necessary to add some statement to assure that FreeCADApp is made only when FreeCADAppObject  is completed.

wsteffe, 2023-11-28:
Now I have added an add_dependencies. I think it should do the work.

```
if(MINGW)
 add_library(FreeCADAppObject OBJECT ${FreeCADApp_SRCS})
 target_link_libraries(FreeCADAppObject ${FreeCADApp_LIBS})
 target_link_options(FreeCADAppObject PUBLIC "-Wl,--enable-auto-import -Wl,--export-all-symbols")
 add_library(FreeCADApp SHARED $<TARGET_OBJECTS:FreeCADAppObject>)
 add_dependencies(FreeCADApp FreeCADAppObject)
else(MINGW)
 add_library(FreeCADApp SHARED ${FreeCADApp_SRCS})
 target_link_libraries(FreeCADApp ${FreeCADApp_LIBS})
endif(MINGW)

```

wwmayer, 2023-11-28:
From the file list you can see that it provides the file /mingw64/include/QtXml/QDomDocument
Now you may have to edit src/Gui/CMakeLists.txt and add `${QtXml_INCLUDE_DIRS}` to the `include_directories`

wsteffe, 2023-11-28:
Yes in fact I had also seen that ${QtXml_INCLUDE_DIRS} was missing in src/Gui/CmakeList.txt. But I was building the LinkMerge branch so may it is properly set in master.

wsteffe, 2023-11-28:
Forgot to say that I added similar cmake code also for shared library FreeCADGui in src/Gui/CmakeList.txt:

```
if(MINGW)
 add_library(FreeCADGuiObject OBJECT ${FreeCADGui_SRCS})
 target_link_libraries(FreeCADGuiObject ${FreeCADGui_LIBS})
 target_link_options(FreeCADGuiObject PUBLIC "-Wl,--enable-auto-import -Wl,--export-all-symbols")
 add_library(FreeCADGui SHARED $<TARGET_OBJECTS:FreeCADGuiObject>)
 add_dependencies(FreeCADGui FreeCADGuiObject)
else(MINGW)
 add_library(FreeCADGui SHARED ${FreeCADGui_SRCS})
 target_link_libraries(FreeCADGui ${FreeCADGui_LIBS})
endif(MINGW)
```

But I am still waiting to see if it works

wwmayer, 2023-11-28:
> Yes in fact I had also seen that ${QtXml_INCLUDE_DIRS} was missing in src/Gui/CmakeList.txt

Somehow the QtXml directory was pulled in implicitly and thus QtXml_INCLUDE_DIRS was never needed. But the correct way is to add it explicitly.

wwmayer, 2023-11-28:
Here some background information about `-Wl,--export-all-symbols`:
https://sourceware.org/binutils/docs-2.18/ld/WIN32.html

Using this option for FreeCADGui may lead to a build failure, something like this: C:/msys64/mingw64/bin/ld: error: export ordinal too large: 67248

wsteffe, 2023-11-28:
Ok then probably it is better to omit -Wl,--export-all-symbols.

wwmayer, 2023-11-28:
> But are Widgets a considered a Component ? If so it is not set in SetupQt.cmake.

Yes, Widgets is a component and it's added here:
https://github.com/FreeCAD/FreeCAD/blob/main/cMake/FreeCAD_Helpers/SetupQt.cmake#L18

You may have to modify Main/CMakeLists.txt by adding `QtWidgets_INCLUDE_DIRS` to the include directories.

wsteffe, 2023-11-28:
Hi wwmayer,
 I have judsy removed 
`target_link_options(..."-Wl,--enable-auto-import -Wl,--export-all-symbols") `

Nothing changed and I still have the strange problem I mentioned before:
In a first run I get a long sequence of undefined reference when linking the shared library. I send an interrupt and give again the make cmd. This time the shared library is built and the process goes on. 

The line:
`add_dependencies(FreeCADGui FreeCADGuiObject)`
was supposed to fix this problem but it didn't work.

wsteffe, 2023-11-28:
No sorry without: 
`target_link_options(..."-Wl,--enable-auto-import -Wl,--export-all-symbols") `
it doesn't link anymore

Now I have restored src/FCglobal.h to have  __declspec(dllexport)  __declspec(dllimport) in MINGW build. But I get a lot of warnings such ad:

`warning: 'virtual void App::DocumentObject::onSettingDocument()' redeclared without dllimport attribute after being reed with dll linkage.
`

And also a fatal error:

```
 DocumentObjectExecReturn *DocumentObject::StdReturn = nullptr;
      |                           ^~~~~~~~~~~~~~
make[2]: *** [src/App/CMakeFiles/FreeCADAppObject.dir/build.make:378: src/App/CMakeFiles/FreeCADAppObject.dir/DocumentObject.cpp.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:2117: src/App/CMakeFiles/FreeCADAppObject.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

```

wsteffe, 2023-11-28:
I have reverted back to the original cmake statement without using __declspec(dllexport) and __declspec(dllimport) (at list for the moment)

```
 add_library(FreeCADApp SHARED ${FreeCADApp_SRCS})
 target_link_libraries(FreeCADApp ${FreeCADApp_LIBS})
```
Now I have a lot of undefined references in the linkage of  FreeCADApp but there is a reason for most of them. In fact most undefineds are related with boost::signal2 or boost::bimap or other boost components that are not included in the Boost library of MSYS2/Mingw64 .

wwmayer, 2023-11-28:
With d88d076d51 I am now able to build FreeCAD with clang and gcc.

wwmayer, 2023-11-28:
> warning: 'virtual void App::DocumentObject::onSettingDocument()' redeclared without dllimport attribute after being reed with dll linkage.

I had a very similar warning. It was caused because for the implementation the inline keyword was used but not for the declaration. MinGW's gcc seems to be very sensitive about this

MehdiChinoune, 2023-11-28:
I have backported the patch https://github.com/msys2/MINGW-packages/commit/47cc7d49dedc05e8f108cdf4abc5b6611a3c65b8

closed 2023-11-28 by MehdiChinoune