FreeCAD/FreeCAD #11533: [Problem] Shiboken TypeError in: FreeCADGui.getMainWindow().windowStateChanged

Andrei-Pozolotin, 2023-11-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Shiboken TypeError in: `FreeCADGui.getMainWindow().windowStateChanged` signal connection

to reproduce:

1) build at revision 35203, e57f5c42cb902cb96098646997110f7f9c242a36  start the application

2) enter in `Python console`:
```python
main_window = FreeCADGui.getMainWindow()
main_window.windowStateChanged.connect(print)
```

3) change main window state, for example: `File -> New`, and observe the error (3 times in a row):
```python
14:59:18  TypeError: Can't call meta function because I have no idea how to handle Gui::MDIView*
14:59:18  TypeError: Can't call meta function because I have no idea how to handle Gui::MDIView*
14:59:18  TypeError: Can't call meta function because I have no idea how to handle Gui::MDIView*
```

### Full version info

```shell
OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35203 (Git)
Build type: None
Branch: makepkg
Hash: e57f5c42cb902cb96098646997110f7f9c242a36
Python 3.11.5, Qt 5.15.11, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.4
  * DynamicData 2.54.0 (Disabled)
  * WebTools 1.0.0 (Disabled)
  * AirPlaneDesign 0.4.0 (Disabled)
  * FEMbyGEN 1.2.0 (Disabled)
  * Assembly3 0.12.0 (Disabled)
  * Manipulator 1.5.0
  * Silk 0.1.5 (Disabled)
  * Ship 2022.4.11 (Disabled)
  * PieMenu
  * MeshRemodel 1.8919.0 (Disabled)
  * CurvedShapes 1.0.5 (Disabled)
  * A2plus 0.4.61
  * AutoMate1 2023.11.23
  * fasteners 0.4.70
  * Curves 0.6.14 (Disabled)
```
```

### Subproject(s) affected?

None

### Anything else?

similar problems were reported in the past:
https://forum.freecad.org/viewtopic.php?t=31776

shiboken code origin for the error message:
https://code.qt.io/cgit/pyside/pyside.git/tree/libpyside/signalmanager.cpp.in#n711

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-11-27:
Per bernd 
> 36029e9 fixed it
> and this caused the problem: 0c63ab4

@wwmayer JFYI, referencing your code changes

wwmayer, 2023-11-27:
One way is to implement a pair of converter functions to wrap and unwrap the C++ or Python objects. A much easier way is replace the MDIView arguments of the signal function with QWidget because then the wrapper class provided by PySide can be used.

#11552

closed 2023-11-27 by chennes

Andrei-Pozolotin, 2023-11-27:
Thank you. 

I confirm this works as of:
Version: 0.22.0dev.35240 (Git) 
Hash: 27b58fd410e62d0bcb596de956b80efcc0d0bec5

commit 3f7e4d42 (2023-11-27): fixes #11533: TypeError in: FreeCADGui.getMainWindow().windowStateChanged