FreeCAD/FreeCAD #11536: Techdraw unconsistency with 3D Drawing

PapouXav, 2023-11-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In Part Design I create a body then with sketch a rectangle on XY Plan reference then a Pad.
In Part Design I create a second body then with sketch a second rectangle on XY Plan reference and then a Pad.
I select the second body and translate it with Placement along Z. 
Both bodies appear correctly one over the other in 3D view

In TeckDraw, I ask for a projection of the 3D view (both bodies) on YZ axis. Both bodies appear on the same level. 
The translation along Z done with Placement is not taken into account in TechDraw. TechDraw is unconsistent. 

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: French/France (fr_FR)
[/code]
```

### Subproject(s) affected?

Techdraw

### Anything else?

3D view : [image]
TechDraw view : 
[image]

I have the very simple FCStd file (Freecad file) ready to be sent

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-11-26:
zip the fcstd file and then drag drop it in to github textbox

PapouXav, 2023-11-26:
[Test_Techdraw_Bug_T2.zip](https://github.com/FreeCAD/FreeCAD/files/13466848/Test_Techdraw_Bug_T2.zip)
Hello Luzpaz
Of course zip is the way to send the file !

I made some additional trials to check on FreeCad 019 revision if the bug was already present. And I found that it was already present.
But I found that **it appears only if the selection is performed in the 3D view by clicking with "left button" on the front face of the Body** I want to draw in TechDraw.

The TechDraw view is correct if the selection is a " whole Body selection" performed in the window "Combined view" where the hierachy appears.

In the zip I tested again with both ways to select : Face and Body.
and I performed the test on my 019 revision and 021 revision. Same behaviour. You'll find both files and four results in Techdraw : two are ok ; two are wrong.

I hope this report will be more accurate to help you to reproduce the behaviour of Freecad that I suspect beeing a bug.
Thanks for your quick answer
Best regards

Roy-043, 2023-11-26:
The issue does not occur in the current dev version. Please test with that version.
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35024 (Git)
Build type: Release
Branch: main
Hash: 4ff00ef99869e35295a5a17b8f486901504c655b
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

luzpaz, 2023-11-26:
@PapouXav see https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

luzpaz, 2023-11-26:
@WandererFan keep ticket open and refine or close ?

PapouXav, 2023-11-26:
I agree with Roy-043. I checked with current 022dev revision (ref hereunder).
The issue is fixed on 022dev.
As suggested by WandererFan I shall select the body and not the feature in my current projet 
Thanks to all of you 
Best Regards
OK to close

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35209 (Git)
Build type: Release
Branch: main
Hash: 1c0456654880e58e5d9e839f27df177c68980185
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: French/France (fr_FR)
```

closed 2023-11-26 by luzpaz