FreeCAD/FreeCAD #11541: Freecad V0.21.1. Problem with displaying restrictions in the viewport.

IgorPack, 2023-11-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When working with sketches in PartDesign, restrictions are displayed with black backlighting, which is a significant nuisance. 
I couldn't find a solution to this problem in the settings.

Any help is welcome.

Thanks

[image]

### Full version info

```shell
v0.21.1 33668 +26 (Git)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-11-26:
Full [About](https://wiki.freecad.org/About) info

IgorPack, 2023-11-27:
Maybe you are right. How does this solve the problem with highlighting restrictions? 

The restrictions highlighting almost completely disappears if you disable restrictions, do you know how to turn it off?

[image]

Roy-043, 2023-11-29:
I can reproduce this if I select the "Anti-Aliasing" option "Line Smoothing" in the preferences. Select a different option and restart FreeCAD.
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35209 (Git)
Build type: Release
Branch: main
Hash: 1c0456654880e58e5d9e839f27df177c68980185
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

IgorPack, 2024-01-08:
> Я могу воспроизвести это, если выберу в настройках опцию «Сглаживание» и «Сглаживание линий». Попробуйте выбрать другой вариант и перезапустить FreeCAD.
> 
> ```
> OS: Windows 8 build 9600
> Word size of FreeCAD: 64-bit
> Version: 0.22.0dev.35209 (Git)
> Build type: Release
> Branch: main
> Hash: 1c0456654880e58e5d9e839f27df177c68980185
> Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
> Locale: Dutch/Netherlands (nl_NL)
> Installed mods:
> ```

Спасибо мой друг, это сработало!

Успехов в Ваших делах.

furgo16, 2024-05-14:
This is still an issue with FreeCAD 0.22, it took me a while to figure out, but unsetting "Line smoothing" did solve the problem.

maxwxyz, 2024-07-03:
related: https://github.com/FreeCAD/FreeCAD/issues/11444

luzpaz, 2024-09-08:
This ticket wasn't auto-closed. Closing.

luzpaz, 2024-09-08:
My mistake, @xtemp09's solution wasn't implemented.

closed 2025-02-22 by github-actions[bot]

FEA-eng, 2025-03-07:
> I can reproduce this if I select the "Anti-Aliasing" option "Line Smoothing" in the preferences.

This is still relevant. @maxwxyz Can you reopen this issue?

[image]

There's also some issue with the appearance of the new origin planes when this Anti-Aliasing setting is used:

[image]

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/90.0633
```

furgo16, 2025-03-07:
Related:
- https://forum.freecad.org/viewtopic.php?t=47129
- https://forum.freecad.org/viewtopic.php?t=79594