FreeCAD/FreeCAD #11543: [Problem] high load/unusable UI after installation of Raytracing Workbench

x1125, 2023-11-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I've been working with this FreeCAD installation fine for month now. Then I installed the Raytracing Workbench, set the Povray binary path and started the tracacing. It used a lot of CPU, memory was steadily increasing (without OOM) and at some point, the window was gone.

Now, whenever I open freecad, I get a constant flood of the following error and the UI locks up:
`Failed to compute left/right minimum bearings for "cursor.pcf"`

Launching FreeCAD via terminal shows that the amount of messages is insane. I googled the error message, but there was nothing useful. Also I'm unable to disable the Raytracing Workbench without a working UI.

### Full version info

```shell
[code]
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: makepkg
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.11.6, Qt 5.15.11, Coin 4.0.1, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Render 2023.8.18
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-11-26:
Do you mean Render workbench instead of 'Raytracing' ?

luzpaz, 2023-11-26:
This is a duplicate of #10514

closed 2023-11-26 by luzpaz

x1125, 2023-11-26:
Yes, #10514 does the trick! Thanks and sorry for not looking thoroughly enough. Ticket can be closed.