FreeCAD/FreeCAD #11553: [Problem] Memory leak in: addonmanager_utilities.restart_freecad()

Andrei-Pozolotin, 2023-11-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

During plugin development, when using `addonmanager_utilities.restart_freecad()` for repeated application restart, the application crashes after a few restarts rather quickly with "out of memory" error. That seems to happen due to `FreeCAD` leaving multiple orphaned `QtWebEngineProcess` instances.

to reproduce:

1) build at revision 35216, 5a4a358f13fa6c01a8930d1792fe85f0caaaedcc

2) provide a custom "verify restart" workbench:

<details>
<summary><code>$HOME/.local/share/FreeCAD/Mod/VerifyRestart/InitGui.py</code></summary>

```python
print(f"### VERIFY RESTART ###")

import FreeCADGui

class VerifyRestartWorkbench(FreeCADGui.Workbench):

    @classmethod
    def std_ui_has_app_init(self) -> bool:
        main_window = FreeCADGui.getMainWindow()  #
        return main_window.property("eventLoop") is True

    @classmethod
    def std_ui_invoke_restart(self) -> None:
        import addonmanager_utilities
        addonmanager_utilities.restart_freecad()

    @classmethod
    def timer_invoke_after_delay(self,
            invoke_user_func:callable,
            invoke_delay_ms:int,
        ) -> any:
        from PySide2 import QtCore
        this_timer = QtCore.QTimer()
        this_timer.singleShot(invoke_delay_ms, invoke_user_func)
        return this_timer

    @classmethod
    def timer_invoke_when_ready(self,
            has_user_state:callable,
            invoke_user_func:callable,
            wakeup_period_ms:int,
        ) -> any:
        from PySide2 import QtCore
        this_timer = QtCore.QTimer()

        def timer_func() -> None:
            if has_user_state():
                this_timer.stop()
                this_timer.deleteLater()
                invoke_user_func()

        this_timer.timeout.connect(timer_func)
        this_timer.setInterval(wakeup_period_ms)
        this_timer.start()

        return this_timer

    @classmethod
    def report_process_state(self) -> None:
        import os
        import pathlib
        user_home_dir = pathlib.Path.home()
[... 112 lines cut ...]
  * DynamicData 2.54.0 (Disabled)
  * WebTools 1.0.0 (Disabled)
  * AirPlaneDesign 0.4.0 (Disabled)
  * FEMbyGEN 1.2.0 (Disabled)
  * Assembly3 0.12.0 (Disabled)
  * Manipulator 1.5.0 (Disabled)
  * Silk 0.1.5 (Disabled)
  * Ship 2022.4.11 (Disabled)
  * PieMenu
  * MeshRemodel 1.8919.0 (Disabled)
  * CurvedShapes 1.0.5 (Disabled)
  * A2plus 0.4.61 (Disabled)
  * AutoMate1 2023.11.23
  * fasteners 0.4.70
  * Curves 0.6.14 (Disabled)
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Andrei-Pozolotin, 2023-12-10:
more info:

when running `FreeCAD` built with extra debug output enabled, there is a QT console message on restart:
```
Release of profile requested but WebEnginePage still not deleted. Expect troubles !
```

that seems to be a known problem/solution:

https://stackoverflow.com/questions/57789171/i-get-release-of-profile-requested-but-webenginepage-still-not-deleted-expect

chennes, 2024-01-26:
It this leak was related to QtWebEngine, then I expect it to have been resolved by #12097 -- can you confirm?

Andrei-Pozolotin, 2024-01-26:
when `Start` workbench is present:
* as of `35740` the issue is still present:  number of `QtWebEngineProcess` keeps going up with every restart

 when `Start` workbench is missing:
 * however, when `Start` workbench removed from auto-load (both check boxes are off), then the issue is gone

[image]

```
OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35740 (Git)
Build type: None
Branch: makepkg
Hash: 441b1d278361020256d72754474d02314a9fafbc
Python 3.11.6, Qt 5.15.11, Coin 4.0.1, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
```

chennes, 2024-01-26:
Let's either close this issue and open a new one, or rename and retag this one. It isn't really Addon Manager at this point, it's Web, or possibly QtWebEngine itself.

Andrei-Pozolotin, 2024-01-27:
#### Code which uses `QProcess::startDetached()` :

https://github.com/FreeCAD/FreeCAD/blob/02c6a90c160a7feac269d8aa0fe1b490bfa5c898/src/Gui/DlgPreferencesImp.cpp#L711

https://github.com/FreeCAD/FreeCAD/blob/02c6a90c160a7feac269d8aa0fe1b490bfa5c898/src/Mod/AddonManager/addonmanager_utilities.py#L155

#### Code which uses `QWebEngineView()` :

https://github.com/FreeCAD/FreeCAD/blob/02c6a90c160a7feac269d8aa0fe1b490bfa5c898/src/Mod/Help/Help.py#L435

https://github.com/FreeCAD/FreeCAD/blob/02c6a90c160a7feac269d8aa0fe1b490bfa5c898/src/Mod/Web/Gui/BrowserView.cpp#L313

Andrei-Pozolotin, 2024-01-27:
#### Similar issues upstream:

* supposed to be resolved in `Qt-5`:
https://bugreports.qt.io/browse/QTBUG-69030

* supposed to be resolved in `Qt-6`:
https://bugreports.qt.io/browse/QTBUG-116856

Andrei-Pozolotin, 2025-01-30:
@chennes : 

RE: `addonmanager_utilities.restart_freecad()`
* the memory leak / hang / crash on restart is now gone
* the one remaining issue with  `restart_freecad()` is that :
* preferences changed before the restart are not present after the restart