FreeCAD/FreeCAD #11561: Edit / Preferences window off screen

gitnoel, 2023-11-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The Edit/Preferences window opens with lower half off the screen, it is not vertically resizable so the Accept, Cancel and OK buttons are unreachable.

### Full version info

```shell
0.22.dev 35209-2023-11-26
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

VM4Dim, 2023-11-28:
https://github.com/FreeCAD/FreeCAD/issues?q=sort:updated+Preferences+dialog
#11004
#11511
#11530

luzpaz, 2023-11-28:
Should be fixed by #11540  (https://github.com/FreeCAD/FreeCAD/commit/7f2175f8a0d3ece2d4b2ab2d4e1d9cecf4e9a05d) which was merged yesterday. @gitnoel please try the latest bleeding edge build and report back. Also please always add your full [about](https://wiki.freecad.org/About) info when reporting bugs. Thanks. Closing.

closed 2023-11-28 by luzpaz