FreeCAD/FreeCAD #11562: When the upper-level model is reduced, a red exclamation mark appears on the lower-level model.[Problem]

yueduz, 2023-11-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[录屏 2023-11-28 14-57-29.webm](https://github.com/FreeCAD/FreeCAD/assets/1448159/e6565ca3-ad11-4eb2-bcf7-3d021e4d6043)
After the lower-level model refers to the edge of the upper-level model, when the reference edge moves to the left, an exclamation mark appears on the lower-level model. When the reference edge moves to the right, there is no error.

### Full version info

```shell
[code]
OS: Arch Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: makepkg
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.11.6, Qt 5.15.11, Coin 4.0.1, Vtk 9.3.0, OCC 7.7.2
Locale: Chinese/China (zh_CN)
Installed mods: 
  * kicadStepUpMod 10.21.5
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2023-11-28:
This isn't really a bug in the code, and there isn't any action to be taken by developers here (short of the work we're doing to mitigate the topological naming problem, which is in progress) -- I suggest you reach out to one of the FreeCAD user groups (the forums, Reddit, Facebook, etc,) and ask for assistance with your model there.

closed 2023-11-28 by chennes

yueduz, 2023-11-28:
I think this is because the object cannot distinguish whether it is on the left or right of the reference object, so after the x value of the reference object is reduced, the object that is originally on the right moves to the right.  You can use positive and negative values ​​to distinguish whether it is on the left or right.

luzpaz, 2023-11-28:
See https://wiki.freecad.org/Topological_naming_problem  
It's being worked on https://github.com/orgs/FreeCAD/projects/2