FreeCAD/FreeCAD #11568: [Wiki] Add command to launch recently compiled FreeCAD

superbonaci, 2023-11-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Add at the bottom of the building section:
https://wiki.freecad.org/index.php?title=Compile_on_Linux&oldid=1330049#:~:text=used%20for%20compilation.-,Building,-To%20build%20simply

How to launch the recently built FreeCAD from `freecad-source/build` folder:
```
bin/FreeCAD
```

Since it's not clear where the binary is placed after successful compilation.

### Full version info

```shell
not needed.
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-11-29:
Please read more carefully:

> The FreeCAD executable will eventually appear in the build/bin

closed 2023-11-29 by Roy-043

superbonaci, 2023-11-29:
That's true didn't notice.