FreeCAD/FreeCAD #11571: FreeCAD 21.1 Can't Specify a Plane Using MultiTransform Tool

GITMuniac, 2023-11-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

ChrisB suggested I make a ticket on this.  The details on the situation are on the FreeCAD forum linked below:
( https://forum.freecad.org/viewtopic.php?t=83053 )
Hope this helps the developers.  Enjoy your day!

### Full version info

```shell
OS: Linux Mint 20.3 (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods:
* Assembly3 0.12.0
* sheetmetal 0.3.12
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

bgbsww, 2023-12-03:
At step 7, I used the plane selection menu to select a plane ( XY ). 

 I also tried XZ, XZ, and specifying a plane.  All of this seems to work in terms of showing the preview.  If I click the 'OK' at the bottom and then the 'OK' at the top, everything is fine.

However, if I right-click on the resulting 'Mirrored' transformation and delete it, the entire application crashes.

If save it however, from that point I cannot make it crash, no matter how I edit, or delete and try to recreate it.

There are lots of "MultiTransform: No mirror plane reference specified" whenever I have a mirror and have not yet specified a plane.  But the crash only happens the very first time through.

So, overall I suspect we need to go for a default option in this case, and preselect "XY plane" instead of nothing in the plane selection button, and track down the crash if it still happens then.

GITMuniac, 2023-12-04:
Hi - I did play around with trying to crash the program via deleting transformations.  I couldn't get it to crash so that isn't a problem on this.  Glad you found this underlying issue and fixing it means adding minimal code.  I do a fair amount of coding myself in C so I try to be sensitive to others trying to find and fix problems.  The devil is always in the details.  Many hands make light work.  Enjoy your day and thanks for the development efforts.

Best - Scott

closed 2023-12-11 by sliptonic

GITMuniac, 2023-12-12:
Hi - I did check the weekly build 35370-2023-12-11 and it works as expected with a default plane filled in.  I donated some bread to your development efforts.  Thanks for the fast fix and making FreeCAD available for those in need.  Enjoy your day!

Best - Scott

PR #11638 (2023-12-11): Add missing initialization of MirrorPlane on non sketch objects (fixes #11571)
Fix #11571