FreeCAD/FreeCAD #11573: [Problem] Sketcher-Deactivated constraints don't show their value in the 3D-view

M4xTheMan, 2023-11-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I've noticed that deactivated constraints don't show their values in the 3D-view anymore. This is how it should look like (from the wiki - [Sketcher_ToggleActiveConstraint](https://wiki.freecad.org/Sketcher_ToggleActiveConstraint)):
[image]
_deactivated constraint (angle); the sketch is no longer fully constrained_

This is how it looks like now. The value of the deactivated angle constraint is missing in the 3D-view but can be seen in the constraints list:
[image]
_value of the deactivated constraint is missing_

I've tested this using the following versions:
0.20.2, 0.21.1 and 0.22.0dev

The full FreeCAD version information:
```
OS: Ubuntu 22.04.3 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * LCInterlocking
  * kicadStepUpMod 10.21.4
  * Lithophane
  * lattice2 1.0.0
  * Defeaturing 1.2.1
  * Manipulator
  * fasteners 0.4.70
  * Assembly3 0.12.0
  * sheetmetal 0.3.12
  * InventorLoader 1.4.0
  * Assembly4
  * Curves 0.6.14
  * POV-Ray-Rendering
  * OpenDark 2023.10.30
  * 3DfindIT 1.2.0
  * Freecad-Built-in-themes-beta 1.2.2
 ```
```
OS: Ubuntu 22.04.3 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * LCInterlocking
  * kicadStepUpMod 10.21.4
  * Lithophane
  * lattice2 1.0.0
  * Defeaturing 1.2.1
  * Manipulator
  * fasteners 0.4.70
  * Assembly3 0.12.0
  * sheetmetal 0.3.12
  * InventorLoader 1.4.0
  * Assembly4
  * Curves 0.6.14
  * POV-Ray-Rendering
  * OpenDark 2023.10.30
  * 3DfindIT 1.2.0
  * Freecad-Built-in-themes-beta 1.2.2
```
```
OS: Ubuntu 22.04.3 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34367 (Git) AppImage
Build type: Release
Branch: master
Hash: e36e16de59dece2d67caf478d25e3a6b825102c6
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * LCInterlocking
  * kicadStepUpMod 10.21.4
  * Lithophane
  * lattice2 1.0.0
  * Defeaturing 1.2.1
  * Manipulator
  * fasteners 0.4.70
  * Assembly3 0.12.0
  * sheetmetal 0.3.12
  * InventorLoader 1.4.0
  * Assembly4
  * Curves 0.6.14
  * POV-Ray-Rendering
  * OpenDark 2023.10.30
  * 3DfindIT 1.2.0
  * Freecad-Built-in-themes-beta 1.2.2
```

chrisb (FreeCAD-forum) confirmed this behaviour using 0.21.0
```
OS: macOS 13.0
Word size of FreeCAD: 64-bit
Version: 0.21.0.33357 (Git)
Build type: Release
Branch: master
Hash: f0560ac2078ad71b4c97b74281990bb916431da9
Python 3.11.3, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.2, OCC 7.7.1
Locale: C/Default (C)
Installed mods: 
  * FC_SU
  * freecad.gears 1.0.0
  * FeedsAndSpeeds 0.5.0
  * dxf-library
  * fcgear 1.0.0
  * fasteners 0.4.59
  * sheetmetal 0.2.62
  * ExplodedAssembly
  * Curves 0.6.8
```

### Full version info

```shell
OS: Ubuntu 22.04.3 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34367 (Git) AppImage
Build type: Release
Branch: master
Hash: e36e16de59dece2d67caf478d25e3a6b825102c6
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * LCInterlocking
  * kicadStepUpMod 10.21.4
  * Lithophane
  * lattice2 1.0.0
  * Defeaturing 1.2.1
  * Manipulator
  * fasteners 0.4.70
  * Assembly3 0.12.0
  * sheetmetal 0.3.12
  * InventorLoader 1.4.0
  * Assembly4
  * Curves 0.6.14
  * POV-Ray-Rendering
  * OpenDark 2023.10.30
  * 3DfindIT 1.2.0
  * Freecad-Built-in-themes-beta 1.2.2
```

### Subproject(s) affected?

Sketcher

### Anything else?

The corresponding thread at the FreeCAD-forum: https://forum.freecad.org/viewtopic.php?p=721092#p721092

FreeCAD-example file (changed the ending from FCStd to zip):
[Sketch_Test_deactivate_constraints.zip](https://github.com/FreeCAD/FreeCAD/files/13493011/Sketch_Test_deactivate_constraints.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FlachyJoe, 2023-11-29:
ping @PaddleStroke #11507 could fix this too.

PaddleStroke, 2023-12-04:
I do not know but this might have been done on purpose to avoid confusion. Because for distance constraints they then act as driven constraint but showing a wrong value obviously.

M4xTheMan, 2023-12-04:
I don't think so. The color marks it as deactivated. The value shows the user which value the driving constraint is going to have as soon as it's getting activated again. Of you need the current value in addition, you've to add a reference constraint. Then you'd be able to see the current and the currently deactivated driving constraint.

oursland, 2023-12-04:
https://github.com/FreeCAD/FreeCAD/pull/11507 was just merged.

PaddleStroke, 2023-12-04:
Yes but it wasn't related to this issue so it is not a problem.

chrisb-github, 2024-05-02:
Another argument, why it is not on purpose: To select such a constraint you cannot click on the existing line, you must click in the void gap.

pierreporte, 2024-05-02:
#12036 has a proposition for displaying deactivated dimension: the value would be strike through.

PaddleStroke, 2025-01-30:
Still relevant

pierreporte, 2025-06-08:
The issue is still relevant.

closed 2025-11-02 by kadet1090