FreeCAD/FreeCAD #11574: [Problem] Polar Link array not exported correctly

PLChris, 2023-11-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When exporting a polar link array from Draft workbench to STL, then the positions are calculated wrong. The equivalent non link array works.
Link to the [forum file](https://forum.freecad.org/download/file.php?id=244083)

### Full version info

```shell
OS: macOS 13.0
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35081 (Git)
Build type: Release
Branch: main
Hash: 32db044caa92a2adc448850179c524b793a03c3e
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * FC_SU
  * freecad.gears 1.1.0
  * FeedsAndSpeeds 0.5.0
  * dxf-library
  * fasteners 0.4.71
  * sheetmetal 0.3.12
  * ExplodedAssembly
  * Curves 0.6.15
```

### Subproject(s) affected?

Draft

### Anything else?

[Corresponding forum post](https://forum.freecad.org/viewtopic.php?p=721770#p721602)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-11-29:
I think this is a duplicate: #11200. Please test with the next weekly.

bgbsww, 2023-12-02:
If it isn't a duplicate, I'll pick it up since I understand the area.  Waiting to hear.

Roy-043, 2023-12-04:
This has been fixed now.
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35274 (Git)
Build type: Release
Branch: main
Hash: dc063eabec61bffc72c92595c66a4649f25fe381
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods: 
```

chrisb-github, 2024-03-05:
Fixed in ```
OS: macOS 13.0
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36228 (Git)
Build type: Release
Branch: main
Hash: 6c8ee5964521a99e01c8d33cbf9aac8a19273ef3
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * FC_SU
  * freecad.gears 1.2.0
  * FeedsAndSpeeds 0.5.0
  * dxf-library
  * fasteners 0.5.13
  * sheetmetal 0.4.2
  * ExplodedAssembly
  * Curves 0.6.23
```