FreeCAD/FreeCAD #11576: [Multiple ToolLibrary buttons in toolbar]

gitnoel, 2023-11-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

multiple ToolLibrary buttons appear on the toolbar when using path workbench.
It appears to add a new button to the toolbar for every face selected when creating a new operation.
[image]

### Full version info

```shell
OS: Linux Mint 21 (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35209 (Git) AppImage
Build type: Release
Branch: main
Hash: 1c0456654880e58e5d9e839f27df177c68980185
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/New Zealand (en_NZ)
Installed mods: 
  * Help 1.0.3
  * BIM 2021.12.0
  * fcgear 1.0.0
  * ExplodedAssembly
  * Curves 0.6.14
  * lattice2 1.0.0
  * fasteners 0.4.24
  * FeedsAndSpeeds 0.5.0
  * ThreadProfile 1.89.0
  * sheetmetal 0.2.59
  * Assembly4 0.12.4
  * btl 0.9.9
  * dxf-library
  * A2plus 0.4.60e
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-11-29:
@JohnOCFII can you confirm on latest dev version ? TIA!

gitnoel, 2023-11-30:
Latest version issue still present - screenshot below
```
OS: Linux Mint 21 (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35249 (Git) AppImage
Build type: Release
Branch: main
Hash: 94ca51b7799b2ac60a2a56d7e6e753bb97a73671
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/New Zealand (en_NZ)
```

[image]
[image]

luzpaz, 2023-12-05:
@gitnoel can you test with Appimage to see if it's not a downstream packaging issue ?

gitnoel, 2023-12-05:
Tested with yesterday's appimage.
A much earlier version doesn't have this issue.

AFK untill tonight. No FreeCAD on the phone :(  in getting withdrawal symptoms!

luzpaz, 2023-12-13:
@gitnoel please add full [About](https://wiki.freecad.org/About) info of the appimage you used

JohnOCFII, 2023-12-25:
I'm not seeing the issue with either the macOS or Linux (Debian Bookworm) weekly versions.  I wonder if some older information in the `user.cfg` might be causing this issue?

```
OS: macOS 14.2.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35414 (Git)
Build type: Release
Branch: main
Hash: 759930e3aab711355d2f0895f641a8657897af48
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
```
and

```
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35414 (Git) AppImage
Build type: Release
Branch: main
Hash: 759930e3aab711355d2f0895f641a8657897af48
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
```

JohnOCFII, 2023-12-25:
I also note the OP has a number of Add-ons installed.  Possibly one of those might be the cause.

gitnoel, 2023-12-25:
Thanks, I'll rename usr.config and report back.  Noel.

JohnOCFII, 2023-12-25:
I see one your your add-ons @gitnoel is btl -- Better Tool Library.  I wonder if that is the culprit.  (I'm tempted to try that, as I'm having Tool Library issues myself)

gitnoel, 2023-12-25:
Ok, I'll check it out. I don't use that one and s couple of others. Time
for a cull.

luzpaz, 2025-01-31:
@gitnoel can you reproduce on v1.0 ?

gitnoel, 2025-02-02:
I am pleased to say that 
Version: 1.1.0dev.39980 (Git) Conda AppImage with
Installed mods: 
  * Help 1.0.3
  * fcgear 1.0.0
  * ExplodedAssembly
  * Curves 0.6.53
  * lattice2 1.0.0
  * FeedsAndSpeeds 0.6.0
  * sheetmetal 0.2.59
  * InventorLoader 1.5.1

no longer has the multiple ToolLibrary button issue.

Thank you Dev team!

luzpaz, 2025-02-03:
Thanks for reporting back.

closed 2025-02-03 by luzpaz