FreeCAD/FreeCAD #11600: [Problem] BIM/ArchView invert door hinge on Section view

rostskadat, 2023-12-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have a wall with a door set with OPERATION_TYPE to SINGLE_SWING_RIGHT.
In the BIM workbench the door open correctly on the right hand side:
[image]
I also have a TechDraw::DrawViewArch (Added through Workbench Techdraw / TechDraw >> View From Other Workbenches >> Insert Arch Workbench Object)
In this view the door opens in the wrong direction:
[image]

I also upload the model in case it is model related issue.

[ViaVerde.zip](https://github.com/FreeCAD/FreeCAD/files/13524357/ViaVerde.zip)

### Full version info

```shell
OS: Ubuntu Core 20 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git) Snap 759
Build type: Release
Branch: tag: 0.21.1
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.7, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Reinforcement
  * Help 1.0.3
  * A2plus 0.4.61
  * Reporting
  * ArchTextures
  * parts_library
  * dodo 1.0.0
  * WebTools 1.0.0
  * BIM 2021.12.0
  * fasteners 0.4.71
  * woodworking 0.21.30486
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-12-01:
Can reproduce on 
```
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35249 (Git) AppImage
Build type: Release
Branch: main
Hash: 94ca51b7799b2ac60a2a56d7e6e753bb97a73671
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * OpenDark 2023.11.27
```

Roy-043, 2023-12-01:
Arch views are generated by the Draft and Arch Workbenches. TD just shows the SVG data that is provides.

In this case a full recompute will update the view. IMO this can be closed.

yorikvanhavre, 2023-12-01:
Upon recomputing I get something really weird :)
[image]
But the whole document gets a red warning, I think there are problems in the model

Roy-043, 2023-12-01:
@yorikvanhavre that has to do with the known issue that an Arch_Building shows a ghost of its content. If you mark all objects in the document for recompute, and then recompute that effect goes away (but I suspect that everything in the SVG is still double). Note that before you recompute in V0.22 you should switch to the 3D view otherwise there is a problem with the calculation of normals. I have to look at the get_3d_view function again, some recent mods by me are probably to blame for that.

The best thing to do here is to section the garage buildingpart and delete the building itself.

Roy-043, 2023-12-04:
The fact that the user is not informed that the document needs to be recomputed for the Arch view to update is an issue. Let's leave this open for now.

marcuspollio, 2024-04-20:
Hi @yorikvanhavre @Roy-043

Could you please let us know the status of this issue : should it be tagged to be fixed for version 1.0 or is it for later ?

Thank you !

Roy-043, 2024-04-20:
This issue is caused by the hosting approach of the Arch WB. Fixing it would require some major changes. So definitely not for the 1.0 release.