FreeCAD/FreeCAD #11603: [Problem] Sketcher: geometry is masked by dimensions

pierreporte, 2023-12-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In a sketch, the dimensions are drawn above the geometry. In particular, when linear dimensions are perpendicular to a line (for instance, when setting the distance of an element from a line, or the distance between two parallel lines), they can completely mask the geometry. Because of the line width, the geometry type (construction or regular) can be invisible. The picture below shows an example, not extreme at all.

The rendering order should change so that geometry is above dimensions.

### Full version info

```shell
OS: Ubuntu 23.04 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35249 (Git) AppImage
Build type: Release
Branch: main
Hash: 94ca51b7799b2ac60a2a56d7e6e753bb97a73671
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * Render 2023.8.18
  * OpenDark 2023.10.30
```

### Subproject(s) affected?

Sketcher

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2023-12-04:
@M4xTheMan I didn’t know you could chose the rendering order. I don’t find it in the settings or while in sketch mode.

It appears that this is a matter of preference. I remember a maintainer writing about this specific issue but I couldn’t find it anymore (in the issues, PR, discussions, and forum). There could be an alternative way: highlight the constrained geometry when the constraint is selected (applicable for every kind of constraint). It would be very useful actually.

M4xTheMan, 2023-12-04:
It's always a good idea to check the wiki. Looking at the side regarding the rendering order, it only handles geometry. I'm not at a computer right now to check it: https://wiki.freecad.org/Sketcher_RenderingOrder

Nevertheless, I still think it would be a good idea to click on such constraints to see where they belong to exactly.

Usually it would be enough to place the.constraint on the other side or use the filter: https://wiki.freecad.org/Sketcher_Dialog

pierreporte, 2023-12-04:
I knew about this but as we both see it doesn’t allow to modify at what level the dimensions should be, hence the question.

M4xTheMan, 2023-12-04:
To filter the constraints isn't a solution for you?
[image]

If not I assume you'd like to make a feature request where the rendering order includes the dimensions, correct?

pierreporte, 2023-12-04:
The filter isn’t a solution IMO because it will make the dimension disappear. Adding dimensions to the rendering order should be part of this issue. A new one could be made for the highlighting of geometry when a constraint is clicked though. Feel free to create that one, I won’t do it tonight.

There are now 3 points:

* Default rendering order: part of #11603.
* Ability for the user to specify where in the rendering order the dimensions should be: part of #11603.
* Constrained geometry is highlighted when a constraint is selected (potentially optional): new feature request.

pierreporte, 2023-12-04:
This proposal would make the dimensions look similar to what was introduced in #11507.

PaddleStroke, 2023-12-05:
Line to line distance coin rendering could be improved as I improved the angle one as to not cover the lines.
Though it can't solve the problem entirely because line to line distance is actually line to point distance.
[image]
So just improving the rendering would improve only on one side

To really improve on both side, we would have to enable real 'line to line' distance.

Hmm after thinking some more about it, I think we could fix the rendering even with current situation.

M4xTheMan, 2023-12-05:
I don't think that what's shown in the screenshot is an improvement. I'd like to know between which points I'm constraining. I can't do that now.

PaddleStroke, 2023-12-05:
The screenshot is the current situation not a proposal. It is the distance between a line and the start point of another line.

PaddleStroke, 2023-12-05:
Below is 2 actual proposals of how we can improve the rendering : 
[image]

Proposal 1 : Fix the rendering on the line side of the constraint only. On the point side we keep the red line going to the actual point constrained.

Proposal 2 : Fix the rendering on both sides. In this the point to which the constraint applies can't be seen. This would be done like this only if lines are parallel obviously.

M4xTheMan, 2023-12-05:
Proposal 1 is what I had in mind. Proposal 2 could be misleading.

I haven't seen the previously commented behaviour (I'm going to check it). Thanks for pointing it out!

pierreporte, 2023-12-05:
Proposal 2 is what I wish. The other still masks the type of the other line.

@PaddleStroke Do you think it would be best to open a new issue to implement line to line distance and close this one?

I understand that @M4xTheMan wants to know which entities are constrained so I will open a new issue about this.

M4xTheMan, 2023-12-05:
I don't get why you'd like to create another issue. This one is about the drawing order, isn't it?

I think the only viable solution is to add the possibility to add dimensions to the drawing order and let the user choose.

pierreporte, 2023-12-05:
I want to create new issues precisely because this one is about rendering order. There would be a one for the implementation of line to line distance, and one for the highlighting.

maxwxyz, 2023-12-15:
also when dimensioning a line with an axis, the leading line of the dimension disappears. Maybe this could be tackled how @PaddleStroke improved the angle and arc constraints?
[image]

PaddleStroke, 2023-12-15:
@maxwxyz this is the same problem as we described before. What happend is that the 'point' of the constraint here is the origin. And the rendering draws the red lines up to the point. If you look at other screenshots you will see it is the same, the line stops on empty air : 

[image]

maxwxyz, 2024-03-24:
Moved this comment over from https://github.com/FreeCAD/FreeCAD/issues/12036

To add from https://github.com/FreeCAD/FreeCAD/issues/12756#issuecomment-1981648341
The render order should be improved to improve visibility and legibility.

1. geometry on top (would solve the overlay issues of dimension lines); dotted geometry (construction, external over normal one to see dashes)
2. constraints / dimensions
3. origin and axis cross

M4xTheMan, 2024-10-31:
I've to admit that I've just experienced that the dimension on the top is very annoying if elements are parallel to an axis and you're using the mid point to constraint something + moving the dimension to one side of the sketch. 
[image]

pierreporte, 2024-12-28:
@M4xTheMan After redesigning an existing assembly in FreeCAD, I can say that this is definitely really annoying.

Ping @PaddleStroke

maxwxyz, 2024-12-28:
@FlachyJoe @AjinkyaDahale is it possible to render geometry always on top?

AjinkyaDahale, 2024-12-29:
> @FlachyJoe @AjinkyaDahale is it possible to render geometry always on top?

There may be a way to give a small negative z-coordinate (wrt the sketch) to make it happen. Though in that case looking at it from the other side would make for the same issue.

pierreporte, 2024-12-29:
@AjinkyaDahale Can the sign be changed dynamically depending on the view angle?

PaddleStroke, 2024-12-29:
It's possible to change the rendering order. But it was on purpose that constraints are on top. As else constraints can be hidden by geos.
I think this is not the best solution.
IMO the best solution is to fix this special case.

pierreporte, 2024-12-29:
If you fix the points where the dimensions start, then there can be a problem if cases like the following. Take the example of two parallel lines illustrated bellow. One of the line is made of 3 segments. If you constrain the distance between line A and line C, the dimension can be above line B or D even though it doesn't mask A or C.

[image]

Even if only the dimensional constrain were put in the background (leaving the other constrains on the foreground), there could be a problem when the dimensions lines are co-linear to construction lines because they are dashed and thus partly transparent.

FlachyJoe, 2024-12-29:
Couldn't we act on the thickness to let the geometry appear below the constraint line?

pierreporte, 2024-12-29:
It appears that Catia and Alibre have this problem too: you can see the dimension behind the dashes. In Albire, construction lines are as thick as normal lines, and thicker than dimensions, so you can distinguish the two. Catia, however, uses the same thickness for dimensions and construction lines so the problem is worse.

My new sketcher style proposal recommends thinner construction lines, and dimensions with the same color as fully-constrained geometry (Alibre, Catia, and Solidworks do this) so FreeCAD could be in the worst situation. So it would be better to drop these thinner lines for the solution to work best.

PaddleStroke, 2025-01-11:
I'm currently making some actual CAD work. And I must say that this is indeed very annoying.

M4xTheMan, 2025-01-11:
Yesterday, I've realized that I'm thinking about where to place dimensions and adding construction geometry only for the purpose of not getting the displayed dimensions in my way too much.

closed 2025-06-18 by kadet1090