FreeCAD/FreeCAD #11604: [Problem] PartDesign: can’t create a sketch on a face when `PartDesign_NewSketch` is active

pierreporte, 2023-12-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

To create a sketch on the face of a part, it is necessary to:

1. click the face,
2. click on `PartDesign_NewSketch`.

The tool should allow the opposite as well:

1. click on `PartDesign_NewSketch`,
2. click on the face.

### Full version info

```shell
OS: Ubuntu 23.04 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35249 (Git) AppImage
Build type: Release
Branch: main
Hash: 94ca51b7799b2ac60a2a56d7e6e753bb97a73671
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * Render 2023.8.18
  * OpenDark 2023.10.30
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

M4xTheMan, 2023-12-04:
Like it is now you're able to easily select the origin planes. Which is good practice by the way. I don't see that your proposal is helping here.

pierreporte, 2023-12-04:
@M4xTheMan My proposal is about consistency. You can select planes before or after activating the tool but faces need to be selected first. This is why I consider this a problem.

M4xTheMan, 2023-12-04:
Like I've said, how should you be able to follow the regular workflow if the model is constantly in the way, preventing the user to select the planes?

pierreporte, 2023-12-04:
Why are you talking about planes? I want to be able to select a face of the part. Putting a sketch on a face is perfectly valid. If we want to force users to put sketches on planes, FreeCAD should not accept to create a sketch on a face in the first place, so when selecting a face then clicking on `PartDesign_NewSketch` should raise an error.

M4xTheMan, 2023-12-04:
I've to apologize, I thought FreeCAD would behave differently. This is how the 3D-view looks when adding a new Sketch:
[image]

I've selected Face14 (green in the 3D-view). When I click in OK, nothing happens.

Your feature request is that the list I've marked on the right hand side stays like it's now but in addition, clicking OK should lead to a new Sketch, mapped to Face14. Correct?

```
OS: Ubuntu 22.04.3 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34367 (Git) AppImage
Build type: Release
Branch: master
Hash: e36e16de59dece2d67caf478d25e3a6b825102c6
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * LCInterlocking
  * kicadStepUpMod 10.21.7
  * Lithophane
  * lattice2 1.0.0
  * Defeaturing 1.2.1
  * Manipulator
  * fasteners 0.4.71
  * Assembly3 0.12.0
  * sheetmetal 0.3.12
  * InventorLoader 1.4.0
  * Assembly4
  * Curves 0.6.15
  * POV-Ray-Rendering
  * OpenDark 2023.11.27
  * 3DfindIT 1.2.0
  * Freecad-Built-in-themes-beta 1.2.2
```

pierreporte, 2023-12-04:
@M4xTheMan you are correct.

FlachyJoe, 2023-12-05:
Hi,
#11133 propose to open the Attachment panel. It would solve this and more.

pierreporte, 2023-12-05:
@FlachyJoe does your proposal would create a plane on the face?

FlachyJoe, 2023-12-05:
@pierreporte no. A plane is not required.

pierreporte, 2023-12-05:
@FlachyJoe I have trouble understanding how it would solve this issue. Can you describe the process?

FlachyJoe, 2023-12-05:
@pierreporte when you create a [primitive](https://wiki.freecad.org/PartDesign_CompPrimitiveAdditive) the  [Attachment](https://wiki.freecad.org/Part_EditAttachment) panel is shown and you can select a plane or a face or something else and a mode for the primitive to be attached.
I propose to have the same behavior for Sketches so if you select a face or a plane it attaches with the FlatFace mode but it also allow to select an edge and attach with NormalToEdge mode, and so else.

pierreporte, 2023-12-05:
@FlachyJoe OK I see now. Pretty slick. With more options it could enable positioned sketches (where you define the location of origin and axis orientations), which is a feature I wanted to request. This issue is then clearly a duplicate. Would you mind editing #11133 to explain more clearly that it will allow user to select a plane for sketches once the tool is launched? It will reduce the risk of other duplicates. I will close this one afterwards.

maxwxyz, 2024-03-13:
@pierreporte should we close this in favor of https://github.com/FreeCAD/FreeCAD/issues/11133 ?
It would be possible if the attachment dialog is used, as implemented in the RT link branch.

pierreporte, 2024-03-13:
Yes you can. Just add the missing details to #11133.

maxwxyz, 2024-03-13:
Duplicate of #11133

closed 2024-03-13 by maxwxyz