FreeCAD/FreeCAD #11607: [Problem] Assembly: creating and using datums is not practical

pierreporte, 2023-12-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The assembly container accepts datum elements (planes, lines, points) and they can be used. However, there are serious issues.

First, `PartDesign_Point`, `PartDesign_Line`, and `PartDesign_Plane` are not available from the Assembly worbench.

Second, it is necessary to create a body to create datum elements. Only then they can be dragged in the assembly container.

Third, when editing a datum plane at the assembly level to position it relative to base planes, its size becomes infinite and it is impossible to determine if it worked.

Four, when creating a joint, selecting a datum inside a part or body shows that the JCS symbol is always at the center of the part or body and not on the plane or along the line. Surprisingly, it works for sketches (see #11606).

### Full version info

```shell
OS: Ubuntu 23.04 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35249 (Git) AppImage
Build type: Release
Branch: main
Hash: 94ca51b7799b2ac60a2a56d7e6e753bb97a73671
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * Render 2023.8.18
  * OpenDark 2023.12.1
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-01-26:
The problem reported in #12106 also exist for `PartDesign_Point`, `PartDesign_Line`, `PartDesign_Plane`, and `PartDesign_CoordinateSystem`. They could be more generic and available in the assembly workbench.

Syres916, 2024-01-26:
> it is necessary to create a body to create datum elements. Only then they can be dragged in the assembly container.

I genuinely don't know where this is coming from:

```
App.ActiveDocument.addObject('PartDesign::Point','PartDatumPoint')

App.ActiveDocument.addObject('PartDesign::Line','PartDatumLine')

App.ActiveDocument.addObject('PartDesign::Plane','PartDatumPlane')
```

all the above works without any body existing.

pierreporte, 2024-01-26:
@Syres916 Try to do it with the GUI. Like sketches created with `PartDesign_NewSketch`, a body appears to contain datum elements. It may be because the PartDesign workbench needs to be opened. They can be dragged in an assembly container and re-attached to assembly origin planes but you can’t exactly call this a streamlined workflow.

pierreporte, 2024-02-23:
Interesting forum post about this: https://forum.freecad.org/viewtopic.php?p=742525#p742525

Zolko-123, 2024-02-27:
> First, `PartDesign_Point`, `PartDesign_Line`, and `PartDesign_Plane` are not available from the Assembly worbench

Don't worry, Ondsel will copy that from Assembly4 too. Alternatively, you could simply use Assembly4 : select a container and click on "_Insert Datum_" and HOP! the datum is created inside the container (Body, Part or Assembly)

maxwxyz, 2024-03-02:
ad 4: https://github.com/FreeCAD/FreeCAD/issues/12446

pierreporte, 2024-03-10:
@maxwxyz this is related to the following complain at Brussels this year:  “Reference objects such as datum planes, axes lines and axis systems should be available globally to all workbenches.”

maxwxyz, 2024-03-10:
Could this be merged into: https://github.com/FreeCAD/FreeCAD/issues/12584

pierreporte, 2024-03-10:
I don’t think so. #12584 is about appearance of planes and interaction with them. This one is about creating them outside of PartDesign. What can be done though is extending #11607 to take other workbenches into account.

pierreporte, 2024-03-15:
@maxwxyz @PaddleStroke

closed 2024-12-13 by chennes