FreeCAD/FreeCAD #11616: [Problem] Application::runApplication() does not apply FileDialog::restoreLocation() on Linux

Andrei-Pozolotin, 2023-12-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

@yorikvanhavre : 

PROBLEM:

the result of this change:
https://tracker.freecad.org/view.php?id=1588

https://github.com/FreeCAD/FreeCAD/blob/dc063eabec61bffc72c92595c66a4649f25fe381/src/Gui/Application.cpp#L2135

```cpp
#if defined(FC_OS_LINUX)
    // See #0001588
    QString path = FileDialog::restoreLocation();
    FileDialog::setWorkingDirectory(QDir::currentPath());
    FileDialog::saveLocation(path);
#else
    FileDialog::setWorkingDirectory(FileDialog::restoreLocation());
#endif
```

is that currently,  on `Linux`, under `KDE`, when `FreeCAD` is started from `KDE Launcher`, 
`QDir::currentPath()` == `$HOME` always, 
with result being `File -> Open` dialog always showing user `$HOME`
and ignoring last saved location from `FileDialog::restoreLocation()`

SOLUTION:

restore the original snippet:
```cpp
    FileDialog::setWorkingDirectory(FileDialog::restoreLocation());
```

to verify the problem:
* on Linux:
* start `FreeCAD`, make new file, save on non-user-home location
* restart `FreeCAD`, try `File -> Open`, observe default open folder is user `$HOME`

to verify the solution:
* on Linux:
* build  with correction applied
* repeat steps above, observe default open folder is last-saved-folder and not user `$HOME`

### Full version info

```shell
OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35270 (Git)
Build type: None
Branch: makepkg
Hash: 1245c3f3b2c18edf505d01036ecdae3f22d349b2
Python 3.11.6, Qt 5.15.11, Coin 4.0.1, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.4
  * DynamicData 2.54.0 (Disabled)
  * WebTools 1.0.0 (Disabled)
  * AirPlaneDesign 0.4.0 (Disabled)
  * FEMbyGEN 1.2.0 (Disabled)
  * Assembly3 0.12.0 (Disabled)
  * Manipulator 1.5.0 (Disabled)
  * Silk 0.1.5 (Disabled)
  * Ship 2022.4.11 (Disabled)
  * PieMenu
  * MeshRemodel 1.8919.0 (Disabled)
  * CurvedShapes 1.0.5 (Disabled)
  * A2plus 0.4.61 (Disabled)
  * AutoMate1 2023.12.2
  * fasteners 0.4.70
  * Curves 0.6.14 (Disabled)
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-08-27:
@Andrei-Pozolotin is this still relevant ?

@wwmayer b322668d and 13298573 were your commits part of https://tracker.freecad.org/view.php?id=1588 Can you weigh-in ?

wwmayer, 2024-08-28:
According to https://tracker.freecad.org/view.php?id=1588 the default directory should be the cwd. So, using the directory of the last FreeCAD session is incorrect behaviour on Linux.

closed 2025-02-04 by Andrei-Pozolotin