FreeCAD/FreeCAD #11617: [Problem] displaced sphere offset error

adlfdlf, 2023-12-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

You create a sphere in the Part workbench and then move it (for example 5 mm on the x and y axis). After you apply an offset to the sphere, the following error message appears. Offset: BRepOffsetAPI_MakeOffsetShape not done

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35249 (Git)
Build type: Release
Branch: main
Hash: 94ca51b7799b2ac60a2a56d7e6e753bb97a73671
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: Spanish/Spain (es_ES)
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2023-12-05:
I've reproduced the problem on the above platforms mentioned by me. The error comes from these lines:

https://github.com/FreeCAD/FreeCAD/blob/a1c08dbf43d4f887a1ac4682360264cb390f83ad/src/Mod/Part/App/TopoShape.cpp#L2622-L2629

luzpaz, 2025-11-24:
> I've reproduced the problem on the above platforms mentioned by me. The error comes from these lines:

@xtemp09 any idea on a possible solution?