FreeCAD/FreeCAD #11619: [Problem] Fedora COPR Nightly Builds failing

ilf-S, 2023-12-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Fedora nightly builds on COPR fail. After examining the log files, the first thing that becomes obvious is that the spec file and rpkg.macros in the FreeCAD git repo still refer to the "master" branch, not the "main" branch. However even after quick fix and attempting to build the package in copr myself in a private project, there still seem to be problems. The F38 build still fails, I need to examine further why, and the F39 build fails because of missing dependencies, probably because python-pyside2 is still non-existent on F39/F40 (that package fails for unrelated reasons to FreeCAD). It would be great if at least the references to "master" can be fixed, so COPR builds don't fail for trivial reasons.

### Full version info

```shell
N/A
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-12-16:
Can someone ping przemof on gitlab ?

luzpaz, 2023-12-16:
@ilf-S is this still an issue ?

luzpaz, 2023-12-16:
CC @DeflateAwning

DeflateAwning, 2023-12-17:
Hmm I'd be happy to help investigate this issue. Wondering if you can help advise where I should begin to look though? Not familiar at all with FreeCAD's automated builds

luzpaz, 2023-12-17:
Oh i though you were since you contributed to to the specs before in 8de6db3e97db97021773caf3188b0b11f390774e - Now I see that it was just for changing the freecad url. Sorry, for the noise. 

@adrianinsaval any chance you can reach out to @PrzemoF on this  ? I don't want to bushwack this by submitting a PR and not knowing how it will effect downstream?

ilf-S, 2023-12-17:
Sorry for the late reply. Yes, the issue still persists. I did not have enough time this week to poke further the build process and on top of that I'm feeling a bit sick, so will probably not be able to investigate further in my own copr.

I did try to change the "master" to "main" and the F38 build progresses, but ultimately fails. The F39 build is a different matter, as F39 ships with Python 3.12 so python-pyside2 is non-existent on F39 (and the future F40). Ultimately from what I gathered, FreeCAD would have to be compiled against Qt6 in Fedora39+. The last successful build for FreeCAD (0.20.2) in the official Fedora repos for F39 was from February (I would assume the F39 dev branch had python 3.11 and the qt bindings for Qt5 were still available). In other words the F39 and future Fedora releases might need a bit more work for the nightly (and even stable) to compile successfully. That is assuming FreeCAD can compile against Qt6 (I know a port was started, I don't know how far it has progressed). Also I don't think pyside6 is in Fedora yet. 

For reference this is the Fedora retirement request for python-pyside2 and FreeCAD https://pagure.io/fesco/issue/3080

Either way, as F38 is still supported for another 6 months, I still would say the nightly should be able to be build for it.

luzpaz, 2023-12-19:
@wwmayer can you ping @PrzemoF please ?

luzpaz, 2023-12-23:
Can a COPR user post to this page please ?
https://discussion.fedoraproject.org/t/freecad-nightly/3857

ilf-S, 2023-12-23:
The last successful build was 2 months ago. What you are seeing as "succeded" is the last build that completed succesfully for all version of Fedora (at the time F37, F38). After that the builds have a status of failed, but that is because F39 was branched out and there was never a successful build for F39. The last successful build for F38 was this one:
https://copr.fedorainfracloud.org/coprs/g/freecad/nightly/build/6506406/
[image]

After that even F38 builds started failing because the spec file is still referring to "master" instead of "main". Again, as I already mentioned, a nightly F39 build at this point isn't going to happen against the stock F39 repos, as there are no python qt5 bindings. For a FreeCAD nightly to be compiled for F39, python qt6 binding need to be in compiled (there is already a build in copr for python-pyside6). This would require copr to include python-pyside6 in it's virtual build environment for FreeCAD nightly. When (and if) FreeCAD nightly is compiled successfully, any user who enables the FreeCAD nightly repo and wants to use it, they need to install python-pyside6 too - either will be provided from another copr repo, or ideally shipped in the FreeCAD copr repo (until Fedora officially includes the python-pyside6 in their official repos). Can someone just fix the references in the spec file from master to main, so we can get at least an attempt for a build for F38. It might finish successfully, or it might fail, but at least we can get the logs.

luzpaz, 2023-12-24:
@ilf-S heads up, 2fe1926

luzpaz, 2023-12-25:
@ilf-S Note:

> Are they using Py_LIMITED_API in their pyside build? Pyside6 uses it by default I think, I had to disable it to be able to build freecad on flatpak because freecad doesn't support that.

ilf-S, 2023-12-25:
> @ilf-S Note:
> 
> > Are they using Py_LIMITED_API in their pyside build? Pyside6 uses it by default I think, I had to disable it to be able to build freecad on flatpak because freecad doesn't support that.

I have absolutely no idea. I randomly found a copr repo with a successfully built python-pyside6. The person to whom it belongs has rights to submit packages in koji, and seems to be a Fedora packager (however don't quote on me that, might be pure conjecture on my side). I was looking for a working Cura for F39 - another package that seems to be in constant limbo in Fedora repos - I stumbled on the python-pyside6 build by pure luck. I'll post a link to it here. I'll poke around copr a bit more during the holidays. I can probably put together a proper build environment, but copr is terra incognita to me still, so even if it works, at best it would be hacky.

By the way, the last build from few hours ago still fails... still seems to refer to master. My understanding was that the FreeCAD nightly repo in copr is set up in a way that it pulls the spec file from the github repo. It could be that it uses a cached file, or it just did not pull the latest version. Unfortunately @PrzemoF  might be the only person who can tell us how the copr repo has been set up at this point. I'll wait for today's build to start with the hope that it was just a timing issue.

[The buildlog from the latest attempt of a build in copr](https://download.copr.fedorainfracloud.org/results/@freecad/nightly/srpm-builds/06813236/builder-live.log.gz)

[The copr repo with python-pyside6](https://copr.fedorainfracloud.org/coprs/smani/FreeCad/packages/) - as you can see the repo is called "FreeCad", so it seems I'm not the only one who needs and uses FreeCAD on Fedora.

Once upon a time I escaped Arch because I was sick of the constant rebuilds of KiCAD and FreeCAD and came back to Fedora mostly because I didn't have to edit and rebuild PKGBUILD and spec files constantly just to end up here again after few years 😁 . Circle of life - or more of the "Linux Desktop" curse I've been living for the past 25+ years (not complaining, I'm just getting old and I'm starting to appreciate my time away from the monitors more).

mr-miky, 2023-12-27:
If it helps, you can use the spec file that I use to compile freecad on copr.
It has been working for a couple of months and obviously has the pyside6 problems of fedora 39 but 37 and 38 work

[mr-miky](https://copr.fedorainfracloud.org/coprs/mr-miky/)/[freeCadDev](https://copr.fedorainfracloud.org/coprs/mr-miky/freeCadDev/)

https://download.copr.fedorainfracloud.org/results/mr-miky/freeCadDev/

ilf-S, 2024-02-22:
> If it helps, you can use the spec file that I use to compile freecad on copr. It has been working for a couple of months and obviously has the pyside6 problems of fedora 39 but 37 and 38 work [...]

I should have checked the issue before I spent the weekend trying to make this work 😄 . I also have a working F38 build of FreeCAD on copr currently, although to me at least, your spec file looks much, much better than mine. Mine has some hard-coded sources and few hacks to get the build running. Either way, at least we have a kicad working. Running it through systemd-nspawn currently, which in itself is not ideal, but it works.

I see that in your spec file you have provisions for pyside6 for F39. Any reason why you don't do a build for F39 yet? I hope you don't mind me plagiarising parts of your spec to test a F39 build later this week.

Sidenote1: Here is my (I bet non-conformant to Fedora rules) spec: https://github.com/ilf-S/FreeCAD/tree/main/package/fedora

Sidenote2: The new assembly workbench is nice 😃

mr-miky, 2024-02-26:
> I see that in your spec file you have provisions for pyside6 for F39. Any reason why you don't do a build for F39 yet?

Leaving out the fact that there is currently no Pyside6/shiboken6 package in fedora 39, even if you use the pyside6 package that I prepared on copr, compiling the package with rpmbuild fails because it does not find any pyside2/shiboken2 includes that should not be necessary :(
The interesting thing is that if I compile with qt creator everything goes well and generates freecad that runs smoothly on fedora 39.

For some reason rpmbuild defines HAVE_PYSIDE2 and HAVE_SHIBOKEN2 but I don’t understand where it happens since the spec file DOESN’t include pyside2/shiboken2 for versions of fedora > 38

ilf-S, 2024-03-05:
@mr-miky I was thinking about Sandro Mani's pyside6 package in copr because I found his, but yours is just as valid an option.

2 quick questions/thoughts (without testing anything, just a hunch). First could it be that some of the qt5 deps in the BuildRequires that are defined but not "if"-ed for Fedora version. Could they be pulling another package from where PYSIDE2/SHIBOKEN2 gets in the build env somehow? Also, if those qt5 deps aren't needed for a F39 build, maybe just make them version dependent. Also, I'm assuming your build fails in copr, not locally? If it is locally and you are like me, doing rolling updates of Fedora, could there be some orphaned files left from previous versions that taint the build environment? If it happens in copr, maybe save some of the logs, I would be happy to comb through them. I'm currently using FreeCAD in a systemd-nspawn chroot, and it works OK, but when doing a sketch that is not that complicated, it progressively becomes more and more sluggish when constraints pile up, so I would love nothing more than a F39 build, just to compare, because it could very well be time for a hardware upgrade for me, instead of blaming the software 🤣

mr-miky, 2024-03-09:
> @mr-miky I was thinking about Sandro Mani's pyside6 package in copr because I found his, but yours is just as valid an 

If I remember correctly my repo comes from that and I modified something in the specs file because otherwise something conflicted with other rpm during the installation of rpm.
 
> 2 quick questions/thoughts (without testing anything, just a hunch). First could it be that some of the qt5 deps in the 

Normally I use mock that creates a fedora39 x86_64 environment completely isolated from my system and install it inside Pyside6 and then have it build the rpms.
In this condition, for some unknown reason, those two define are added to the gcc command line that makes you include code that requires pyside2 and shiboken2.
I have another native Fedora 39 PC and trying to compile with qtcreator does not produce such a problem and generates the freecad executable.
If also here I try to use rpmbuild -bb freecad.spec I run into the problem of defines that enable code that requires pyside2, shiboken2.

Normally I use mock that creates a fedora39 x86_64 environment completely isolated from my system and install it inside Pyside6 and then have it build the rpms.
In this condition, for some unknown reason, those two define are added to the gcc command line that makes you include code that requires pyside2 and shiboken2.
I have another native Fedora 39 PC and trying to compile with qtcreator does not produce such a problem and generates the freecad executable.
If also here I try to use rpmbuild -bb freecad.spec I run into the problem of define that enable code that requires pyside2, shiboken2.

Lately I have no time or desire to follow this and I continue to use FC38 in all my PCs even at work.
Already in the past I skipped even 2-3 releases of fedora because things happened not very dissimilar

luzpaz, 2024-08-27:
@adrianinsaval can you contact @PrzemoF (not sure he is active on github) on the forum and bump this COPR build still failing ?

reneas, 2024-12-01:
Is there any news on this?
Or is my only bet to use the AppImage on Fedora?

tarman3, 2024-12-01:
AppImage better for any Linux because this is the only way to use the latest version.

reneas, 2024-12-02:
I could get the 1.0 release of freecad from `apt` sources though?

Would just be great to have access to all the subdirectories of Freecad (unlike in the AppImage) and also to not have it running as a flatpak which I personally don't like that much...

So a `copr` repo would be great if you're asking me!

ljo, 2025-02-28:
I have a PR #19565 which probably solve this build fail (syntax error) as well.

ilf-S, 2025-02-28:
There are few people building the nightly in copr (my builds are private, they don't get rebuild on automatically on new commits, and deviate a bit from the old spec file, as I also build netgen ang gmsh). Out of curiosity, I can probably make a copr repo with nightly builds, but only tracking the current Fedora release (i.e. 41). Is there a need for other arch than x86_64 and aarch64?