FreeCAD/FreeCAD #11620: [Problem] Part::MultiFuse fails to merge cylinders and spheres

csirmaz, 2023-12-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have a computer-generated Python script that creates a number of spheres and cylinders in FreeCAD and then attempts to merge them using Part::MultiFuse to get a single object. This operation frequently fails without any error messages, but with the resulting object clearly missing a large number of facets. I found that the following strategies help to make this process successful more often, but the results are unreliable at best:

- Randomize the placement and angle of objects to avoid coplanar surfaces
- Create a tree of fusions merging a small number of objects at a time instead of all

As an example, see the attached series of Python commands that I paste into the Python console. The problem can also be seen if the individual objects created are selected in FreeCAD (Ctrl-A) and then Part:Fuse is selected from the OpenSCAD toolbar - often no facets are shown.

I also attached two screenshots. The individual cylinders and spheres are displayed correctly (notfused.png) but after fusing, facets disappear (fused.png).

My aim is to export a STL file describing the object that is the union of these individual shapes. If there is a better way to do this, do let me know.

[image]
[image]
[script.txt](https://github.com/FreeCAD/FreeCAD/files/13537703/script.txt)

PS The custom fuse function in the script is based on https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Part/BOPTools/BOPFeatures.py#L85 as the implementation there does not support providing a custom name for the resulting object.

### Full version info

```shell
[code]
OS: Debian GNU/Linux 11 (bullseye) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

josephSummerhays17, 2023-12-03:
I'm having this exact same problem. I guess there's a good work around, but it is annoying.

ElementW, 2026-09-14:
Well, OCC (here version 8.1.0.dev1-3d097a0328) can create degenerate geometry even when fusing two cylinders, O94 and O95 from the script for example, so we're not off to a good start, and this kind of stuff causes all sorts of failures down the line including entire swathes of geometry disappearing during boolean operations.

[image]

It apparently thinks it's a perfectly fine flat face with no errors whatsoever

[image]

You can make a binder of it juuust fine (no)

[image]

nothing is wrong with the bind either (it is very wrong)

[image]

Mind you this is probably just *one* of multiple OCCT failure modes exhibited by your script.

@maxwxyz Can you add to the OCC Liaison board with "still unresolved" version please?
@pieterhijma FYI