FreeCAD/FreeCAD #11622: [Problem] Creating a fillet crashes

carlosbaraza, 2023-12-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When trying to create a fillet for this edge, FreeCAD would crash.

[image]

### Full version info

```shell
[code]
OS: macOS 13.2
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * Freecad-Built-in-themes-beta 1.2.2
  * Behave-Dark-Colors 0.1.1
  * freecad.gears 1.0.0
  * Dracula 0.0.5
  * Render 2023.8.18
  * fasteners 0.4.70
  * CurvedShapes 1.0.5
  * ExtremeProDark 2.9.0
  * ExtremeProLight 1.6.0
  * Curves 0.6.14
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-12-04:
@carlosbaraza Please post your file.

@drcowsley Your issue seems unrelated. Please start a [forum](url=https://forum.freecad.org/) topic to address your problems. Note that there will probably be some criticism regarding your modeling strategy (which is a bit inefficient).

drcowsley, 2023-12-04:
> Please start a forum topic to address your problems. Note that there will

I will do as you suggest, and nobody should be embarrassed by helping me
with my modelling skills.  I am completely self taught, and a teacher who
is learning at the same time, and is not even one step ahead, is
embarrassment enough.

Thank you for spending time looking at my (our?) problem.

Cheers.

Chris.

xtemp09, 2023-12-05:
@drcowsley, FreeCAD is known to crash when doing fillet. I export models as STL and do fillet (and chamfer) in Blender.

carlosbaraza, 2023-12-12:
[FreeCAD-issue-11622-reproduction.zip](https://github.com/FreeCAD/FreeCAD/files/13649695/FreeCAD-issue-11622-reproduction.zip)

@Roy-043 I have uploaded a reproduction file.

# How to reproduce in FreeCAD 21.1.0 OSX
[image]

xtemp09, 2023-12-13:
I've reproduced on Ubuntu. The backtrace is somehow similar to all other crashes involving the fillet.

<details><summary>Backtrace</summary>

```ruby
* thread #1, name = 'FreeCAD', stop reason = signal SIGSEGV: address not mapped to object (fault address: 0x0)
    frame #0: 0x00007fff5c3056bb libTKFillet.so.7`___lldb_unnamed_symbol5508 + 139
libTKFillet.so.7`___lldb_unnamed_symbol5508:
->  0x7fff5c3056bb <+139>: movq   (%rdi), %rax
    0x7fff5c3056be <+142>: movsd  0x8(%r13), %xmm0          ; xmm0 = mem[0],zero 
    0x7fff5c3056c4 <+148>: movq   $0x0, -0x50(%rbp)
    0x7fff5c3056cc <+156>: movq   %rdx, -0x90(%rbp)
(lldb) bt
* thread #1, name = 'FreeCAD', stop reason = signal SIGSEGV: address not mapped to object (fault address: 0x0)
  * frame #0: 0x00007fff5c3056bb libTKFillet.so.7`___lldb_unnamed_symbol5508 + 139
    frame #1: 0x00007fff5c3077f1 libTKFillet.so.7`ChFi3d_Builder::PerformOneCorner(int, bool) + 2769
    frame #2: 0x00007fff5c2c14f1 libTKFillet.so.7`ChFi3d_Builder::PerformFilletOnVertex(int) + 1169
    frame #3: 0x00007fff5c2c466e libTKFillet.so.7`ChFi3d_Builder::Compute() + 3198
    frame #4: 0x00007fff5c421e0b libTKFillet.so.7`BRepFilletAPI_MakeFillet::Build() + 43
    frame #5: 0x00007fff803034a2 _PartDesign.so`PartDesign::Fillet::execute(this=0x000055555cf9c4a0) at FeatureFillet.cpp:114:23
    frame #6: 0x00007ffff60cf1d7 libFreeCADApp.so`App::DocumentObject::recompute(this=0x000055555cf9c4a0) at DocumentObject.cpp:161:29
    frame #7: 0x00007fff462959e4 Part.so`Part::Feature::recompute(this=<unavailable>) at PartFeature.cpp:92:42
    frame #8: 0x00007ffff605f6e2 libFreeCADApp.so`App::Document::_recomputeFeature(this=0x000055555917a000, Feat=0x000055555cf9c4a0) at Document.cpp:3129:41
    frame #9: 0x00007ffff60677e6 libFreeCADApp.so`App::Document::recompute(this=0x000055555917a000, objs=<unavailable>, force=<unavailable>, hasError=0x0000000000000000, options=<unavailable>) at Document.cpp:2856:48
    frame #10: 0x00007ffff6133d17 libFreeCADApp.so`App::DocumentPy::recompute(this=0x00005555590a0d90, args=<unavailable>) at DocumentPyImp.cpp:584:54
    frame #11: 0x00007ffff6133fb9 libFreeCADApp.so`App::DocumentPy::staticCallback_recompute(self=0x00005555590a0d98, args=<unavailable>) at DocumentPy.cpp:1949:66
    frame #12: 0x00007ffff5328008 libpython3.10.so.1.0`___lldb_unnamed_symbol3713 + 72
    frame #13: 0x00007ffff52dfadc libpython3.10.so.1.0`_PyObject_MakeTpCall + 140
    frame #14: 0x00007ffff527ba1c libpython3.10.so.1.0`_PyEval_EvalFrameDefault + 40380
    frame #15: 0x00007ffff53c23af libpython3.10.so.1.0`___lldb_unnamed_symbol4418 + 1647
    frame #16: 0x00007ffff53bd3de libpython3.10.so.1.0`PyEval_EvalCode + 190
    frame #17: 0x00007ffff540c4cd libpython3.10.so.1.0`___lldb_unnamed_symbol4668 + 157
    frame #18: 0x00007ffff540d4f9 libpython3.10.so.1.0`PyRun_StringFlags + 121
    frame #19: 0x00007ffff598c77d libFreeCADBase.so`Base::InterpreterSingleton::runString[abi:cxx11](this=<unavailable>, sCmd="App.ActiveDocument.recompute()") at Interpreter.cpp:249:15
    frame #20: 0x00007ffff6f2b9a4 libFreeCADGui.so`Gui::Command::_runCommand(file="/home/xtemp09/Downloads/freecad-source/src/Gui/Command.cpp", line=850, eType=App, sCmd="App.ActiveDocument.recompute()") at Command.cpp:725:38
    frame #21: 0x00007ffff6f2bb62 libFreeCADGui.so`Gui::Command::_doCommand(file="/home/xtemp09/Downloads/freecad-source/src/Gui/Command.cpp", line=850, eType=App, sCmd="") at Command.cpp:677:16
    frame #22: 0x00007ffff6f2bc4f libFreeCADGui.so`Gui::Command::updateActive() at Command.cpp:850:5
    frame #23: 0x00007fff441a0c77 PartDesignGui.so`finishFeature(cmd=0x00005555598afe80, Feat=0x000055555cf9c4a0, prevSolidFeature=0x0000555559dd9380, hidePrevSolid=true, updateDocument=true) at Command.cpp:548:26
    frame #24: 0x00007fff441a5431 PartDesignGui.so`finishDressupFeature(cmd=0x00005555598afe80, which="Fillet", base=0x0000555559dd9380, SubNames=size=1, useAllEdges=false) at Command.cpp:1655:18
    frame #25: 0x00007fff441ac347 PartDesignGui.so`makeChamferOrFillet(cmd=0x00005555598afe80, which="Fillet") at Command.cpp:1685:26
    frame #26: 0x00007fff441ac44f PartDesignGui.so`CmdPartDesignFillet::activated(this=0x00005555598afe80, iMsg=<unavailable>) at Command.cpp:1708:24
    frame #27: 0x00007ffff6f311bc libFreeCADGui.so`Gui::Command::_invoke(this=0x00005555598afe80, id=0, disablelog=<unavailable>) at Command.cpp:444:26
    frame #28: 0x00007ffff6f3182c libFreeCADGui.so`Gui::Command::invoke(this=0x00005555598afe80, i=0, trigger=TriggerAction) at Command.cpp:408:12
    frame #29: 0x00007ffff6f1c15f libFreeCADGui.so`Gui::Action::onActivated(this=<unavailable>) at Action.cpp:112:22
    frame #30: 0x00007ffff6f261fa libFreeCADGui.so`QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (Gui::Action::*)()>::call(f=<unavailable>, o=<unavailable>, arg=<unavailable>) at qobjectdefs_impl.h:152:20
    frame #31: 0x00007ffff6f26238 libFreeCADGui.so`QtPrivate::QSlotObject<void (Gui::Action::*)(), QtPrivate::List<>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) [inlined] void QtPrivate::FunctionPointer<void (Gui::Action::*)()>::call<QtPrivate::List<>, void>(arg=<unavailable>, o=<unavailable>, f=<unavailable>) at qobjectdefs_impl.h:185:95
    frame #32: 0x00007ffff6f26230 libFreeCADGui.so`QtPrivate::QSlotObject<void (Gui::Action::*)(), QtPrivate::List<>, void>::impl(which=<unavailable>, this_=<unavailable>, r=<unavailable>, a=<unavailable>, ret=<unavailabl [... cut ...]
    frame #62: 0x00007ffff3429e40 libc.so.6`__libc_start_main_impl(main=(FreeCAD`main at MainGui.cpp:105:1), argc=1, argv=0x00007fffffffdde8, init=<unavailable>, fini=<unavailable>, rtld_fini=<unavailable>, stack_end=0x00007fffffffddd8) at libc-start.c:392:3
    frame #63: 0x0000555555569c05 FreeCAD`_start + 37
(lldb)

```

</details>

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35335 (Git)
Build type: Debug
Branch: main
Hash: 0e5d2c46f0d2661d9f8b6692ae0bfad6c5504762
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
```

FlachyJoe, 2023-12-13:
The bug is in OCC. We can't do anything on our side.
Exception are caught in `ChFi3d_Builder::Compute()` but none are thrown below i.e. in `ChFi3d_Builder::PerformFilletOnVertex` nor in `ChFi3d_Builder::PerformOneCorner`. It's certainly lacks a validity test somewhere.
ref : https://github.com/Open-Cascade-SAS/OCCT/blob/6dbfade69200b6029b8e375cee8f829293fc185a/src/ChFi3d/ChFi3d_Builder.cxx#L266-L279

maxwxyz, 2024-10-07:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/8179

closed 2024-10-07 by maxwxyz