FreeCAD/FreeCAD #11635: [Feature Request] TechDraw: Show/Hide Grid in Taskbar / Right Click Menu

maxwxyz, 2023-12-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

@WandererFan Currently, the grid can only be activated in the preferences dialog. I often use the right mouse menu or the button in the toolbar to temporarily disable the frames but wish that the grid would be hidden as well. Is it possible to add the show/hide grid in the right click menu and maybe also in the toolbar? The icon from draft plane could be used here.
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35274 (Git)
Build type: Release
Branch: main
Hash: dc063eabec61bffc72c92595c66a4649f25fe381
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.24.8
  * CurvedShapes 1.0.5
  * Curves 0.6.15
  * Defeaturing 1.2.1
  * fasteners 0.4.71
  * FEMbyGEN 2.0.0
  * freecad.gears 1.0.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.11.27
  * sheetmetal 0.3.12
  * woodworking 0.21.30486
```
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2023-12-06:
The grid can also be toggled in the Page's View tab in the property editor.    That said, I think the 3 export options in the context menu are a bad use of space as there are tool bar buttons that do the same things.

I don't toggle the grid very often, but I have no objection to adding that to the menu.   Maybe "recompute view" would be another more useful item.

Adding a toolbar button becomes mostly trivial once we add it to the context menu.

luzpaz, 2024-12-04:
Does this just entail the straightforward task of adding the function to the context menu ?