FreeCAD/FreeCAD #11636: [Feature Request] TechDraw: Snap Dimensions Values / Align Dimensions / Move without changing relative text position on dimension

maxwxyz, 2023-12-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

@WandererFan is it somehow possible to snap the dimensions in TechDraw:

- Create them with an equal (defined) spacing to the model
- Snap the text on the dimension to the center of the dimension arrows
- Align the dimension (arrows) while moving to other dimensions (snapping)
- Move the dimension without changing the local (relative) position of the dimension text (so it stays in the center)

Quick demo, creating a new dimension I have to carefully place them in line with the existing one but while dragging, the value won't stay centered:

https://github.com/FreeCAD/FreeCAD/assets/6246609/246a2806-40d3-4bf4-a274-7225d4234c9e

I know that the text can be centered with *position chain dimension* but every time I have to move or adjust a dimension, the value is not centered. Maybe there could be an option to center it, or it snaps centered while dragging, or I can hold a key while dragging to restrict the value text from moving relative.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35274 (Git)
Build type: Release
Branch: main
Hash: dc063eabec61bffc72c92595c66a4649f25fe381
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.24.8
  * CurvedShapes 1.0.5
  * Curves 0.6.15
  * Defeaturing 1.2.1
  * fasteners 0.4.71
  * FEMbyGEN 2.0.0
  * freecad.gears 1.0.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.11.27
  * sheetmetal 0.3.12
  * woodworking 0.21.30486
```
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

ab0v3g4me, 2023-12-04:
This would be a pretty good feature in my opinion.

For example in Creo you create something called `Snap lines` which can be offset either from a geometric entity or view bounding box. These can be used for snapping dimensions, symbols, annotations. Another usage of this is that you can cut extension lines that cover another dimension with the snap lines.

But there is very little functionality for organizing the drawing in TechDraw currently. It would be awesome to see some of it added.

pierreporte, 2023-12-04:
This is a must for good quality drawings.

Something very important for centering the text of dimensions: it must take into account everything like prefixes, suffixes, added text, whatever. In some programs like Alibre or Catia only the base value is centered. You have to move the text by hand for it to be visually in the middle of the arrow.

WandererFan, 2023-12-06:
Snap lines sound like "guides" in Inkscape.   Easy to add the lines, harder to use them for alignment.  Will take some thinking.

maxwxyz, 2023-12-06:
@WandererFan numbers referencing the demo below.

- 1-2: The idea was that the dimensions snap to other existing dimensions, so that they can be aligned easily by snapping to each other (multiple vertical or horizontal dimensions, no chains). Even on different views, that the page looks ordered and clean.
- 3: Same behavior when dragging the text/value of the dimension on the arrow/dimension line: it could snap centered if you move it near the center of the arrow/dimension line.
- 4-5: That is meant by the first point *Create them with an equal (defined) spacing to the model*: New dimensions should have a default distance to the model geometry/lines and also parallel dimension lines above existing ones should have a defined distance. Maybe that can be achieved by snapping to those distances as well I think there is a standard which defined that: 

> Dimension lines are drawn at a distance of 10 mm from the body edges as a fine solid line. If several dimension lines are required, they should be drawn parallel at a distance of 7 mm - under no circumstances should they cross. In addition, the dimension lines should extend about 1 to 3 mm beyond the dimension line boundary.

[image]

PaddleStroke, 2024-05-29:
Can you please detail what you mean by the remaining point?

> Create them with an equal (defined) spacing to the model

Currently it is possible to position them in a 'cascading' disposition. Which sounds like what this point is meaning.

https://github.com/FreeCAD/FreeCAD/assets/19984177/a7014552-f93d-46bf-b844-4350176a4a5c

Is this not what you meant?

maxwxyz, 2024-05-29:
@PaddleStroke no. What is meant: 
1. Snap also to a distance of the model edge which is the feature the dimension is measuring:
I usually should be 10 mm (in comparison to 7mm between cascading dimensions). This should snap to the distance on both sides of the model line.
[image]
2. When you currently create a new dimension (not with the smart tool) the dimension s usually directly placed on the model line which is annoying as it is not legible and you have to move all dimensions. This should be utomatically created 10mm outside of the model line (where snapping in point 1 should also snap to)
[image]
[image]

Is this understandable?

PaddleStroke, 2024-05-29:
Ah yes this is not possible yet. I thought I had done this but actually not yet.

PaddleStroke, 2024-05-29:
Why 1.5x cascadespacing ? I would rather use cascadespacing directly so that it looks evenly paced?

maxwxyz, 2024-05-29:
> Why 1.5x cascadespacing ? I would rather use cascadespacing directly so that it looks evenly paced?
I think it is recommended to use minimum 10 mm to the model and minimum 7 between cascading. Not sure why, maybe that it does not stick too close to the model itself.

pierreporte, 2024-05-29:
The spacing values should be a document parameter (similar to template or projection).

PaddleStroke, 2024-05-29:
Well I have just spent my whole afternoon on this which should have been simple, but I couldn't get it to work properly. I'm dropping it for now.