FreeCAD/FreeCAD #11637: [Problem] Docked Overlay Task Dialog not Showing on all Features

maxwxyz, 2023-12-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The new docked overlay mode for the task view is great, I have set it to *Auto show dialog when editing* which seems ok in most cases, but on some dialogs it won't show by default. Not sure if this is a thing of the individual dialog / WB or if this could be fixed in the auto show setting somehow.
WB / Dialogs where this behavior sometimes occur:
- TechDraw
- FEM
- new Assembly
- Part

Quick demo, first dialog opens correctly when double clicking on a dimension in TechDraw, second operation the task view does not open automatically and has to be invoked by hovering some time at the edge of the screen, not a great behavior for new persons:

https://github.com/FreeCAD/FreeCAD/assets/6246609/bb745f2e-5f1d-4770-9f8d-ddada4ec93cf

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35274 (Git)
Build type: Release
Branch: main
Hash: dc063eabec61bffc72c92595c66a4649f25fe381
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.24.8
  * CurvedShapes 1.0.5
  * Curves 0.6.15
  * Defeaturing 1.2.1
  * fasteners 0.4.71
  * FEMbyGEN 2.0.0
  * freecad.gears 1.0.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.11.27
  * sheetmetal 0.3.12
  * woodworking 0.21.30486
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2023-12-04:
This also occur in PartDesign. When creating a standard part (with the new start page), the task panel is invisible. I’ve encountered the problem elsewhere so I’ll try to contribute to the list.

For the new assembly workbench it’s no surprise since it’s not finished.

pierreporte, 2023-12-04:
It would be less of a problem if the panel hint was more obvious, say something like this:

[image]

The thing would appear only when there are available tasks.

pierreporte, 2023-12-04:
In Part Design, the appearance panel of a body does not appear automatically.

maxwxyz, 2023-12-12:
Sketcher_ValidateSketch is also not opening the task dialog

adrianinsaval, 2023-12-12:
@realthunder

pierreporte, 2023-12-13:
I think that my idea of making the hind more obvious, like a vertical tab-bar, could allow more very useful features and could solve #11072 by the way. This wouldn’t solve task panels not showing when they should though.

pierreporte, 2023-12-13:
There seems to be a workaround to this issue. When set to “Auto task”, the panel is always shown when there is a task and hidden when there isn’t. However it is not possible to just hide it temporarily when unwanted (when it shows possible actions, not when it is used as an editor).

----------------

By the way I found something strange with “Auto task”:

1. Open TechDraw workbench.
2. `TechDraw_PageDefault`.
3. The panel is completely black.
4. The panels turns transparent as expected when hovered.

[image]

realthunder, 2023-12-14:
The reason of the behavior in OP is because some featurse do not implement `Editing` task as a standard FreeCAD editing operation, such as many `TechDraw` features. Standard FreeCAD editing operation allows only one object being edited at any time and will auto close any current editing task before launch new editing task. Many features do not use FreeCAD editing service for whatever reason. They just insert a dialog into the task view. Secondly, there are task dialogs that do not correspond to any object editing, such as the task list in `PartDesign`. This is the reason of having overlay `Auto Task` mode.

I understand that having `Show on Edit` and `Auto Task` may confuse newcomers. We can probably remove `Show on Edit` option in the future. Regardless, the editing behaviors of various FreeCAD features are kind of messy. It is very difficult to implement an `Auto` option that covers every corner. There is another not so obvious `Manual` mode, which is more user friendly for some, especially seasoned users. Turn off any `Auto` mode. Click the X button on the panel to hide it. You can press shortcut (`Ctrl + Left/Right/Up/Down`) to toggle the panel at any time when you need it. This works for all panels. The shortcut can be customized through `Tools -> Customize -> Keyboard`.

maxwxyz, 2023-12-14:
@WandererFan how difficult is it to implement the editing task?

WandererFan, 2023-12-14:
I've never heard of the "standard FreeCAD editing operation".  

I think it might be referring to the "getInEdit" method which doesn't support TechDraw, Spreadsheet or Dependency Graph.  see #9626.

realthunder, 2023-12-16:
> It would be less of a problem if the panel hint was more obvious, say something like this:

@pierreporte Forgot to mention this. The `Tab bar` hint feature is already there. You can enabled in with `Preference -> Display -> UI -> Hint show tab bar`

[image]

realthunder, 2023-12-16:
> I've never heard of the "standard FreeCAD editing operation".

@WandererFan Here is an example of lanuching a `standard FreeCAD editing operation` in a command,

```c++
doCommand(Gui, "Gui.ActiveDocument.setEdit('feature_name', edit_mode)");
```
For this to work, the feature view provider method `setEdit(int mode)` needs to handle the requested mode, e.g., by creating a task dialog.

This second operation shown in the OP screencast is invoking command `CmdTechDrawExtensionSelectLineAttributes` which simply creates the task dialog by itself, and that's why it's not recognized as a `editing operation`.

pierreporte, 2023-12-16:
@realthunder thank you for the hint! However the tab is displayed only when the mouse is near the edge of the screen. It could be always visible. See what I propose here: https://github.com/FreeCAD/FreeCAD/issues/11072#issuecomment-1854444129

maxwxyz, 2023-12-16:
> I understand that having `Show on Edit` and `Auto Task` may confuse newcomers. We can probably remove `Show on Edit` option in the future. Regardless, the editing behaviors of various FreeCAD features are kind of messy. It is very difficult to implement an `Auto` option that covers every corner. 

I see, then just remove the one which is not working :D

WandererFan, 2023-12-16:
>>> This second operation shown in the OP screencast is invoking command CmdTechDrawExtensionSelectLineAttributes which simply creates the task dialog by itself, and that's why it's not recognized as a editing operation.  <<<

Well, it isn't an editing operation.  It isn't modifying an existing object, just collecting options to be applied later.  

Many TaskDialogs are not edit operations.  Dialogs that create a new feature objects can't be started from the ViewProvider since the ViewProvider doesn't exist until after the feature object has been created and added to the document.

maxwxyz, 2024-03-03:
Related: https://github.com/FreeCAD/FreeCAD/issues/11013