FreeCAD/FreeCAD #11639: [Feature Request] Multiple steps (multistep analyses) in FEM

FEA-eng, 2023-12-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently, CalculiX analysis defined in the FEM workbench can have only one step while multistep analyses are often used. The steps can be even of different types (e.g. static preload followed by frequency or buckling).

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35274 (Git)
Build type: Release
Branch: main
Hash: dc063eabec61bffc72c92595c66a4649f25fe381
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2023-12-05:
Also mentioned here https://forum.freecad.org/viewtopic.php?t=81769

FEA-eng, 2023-12-05:
> Also mentioned here https://forum.freecad.org/viewtopic.php?t=81769

Actually, that one would require a different approach. Frequency or buckling analysis steps can be preloaded by preceding static steps but to use mode shapes from buckling step in a subsequent static analysis, one would have to export the deformed mesh and perform a separate simulation on it. This is related to the following issue: #11301

maxwxyz, 2023-12-05:
I thought this preloading is also meant by this multi step analysis - coulnd't this be implemented as well?

FEA-eng, 2023-12-05:
> I thought this preloading is also meant by this multi step analysis - coulnd't this be implemented as well?

Those are 2 different approaches:
1) static --> frequency/buckling - **preload** from static step to frequency/buckling step, can be done with a multistep analysis
2) buckling --> static - **imperfection** from linear buckling step to nonlinear static analysis, requires separate analyses with deformed mesh export and import

Multistep analyses have many more applications. They are necessary for some dynamic procedures (#11647) but can also be used to perform e.g. nonlinear static simulations with multiple stages (loading then unloading and so on).